betaflight/betaflight #6214: moving TPA to PID profile from rate profile and.. changing stock threshold value ###feature request

theSkonk, 2018-06-25:
so my self and @flint723 have been testing and advocating the TPA threshold set down to 1050 for reasons listed in   #6175. 
as its a feature really useful when used to go from 4s to 5 or even 6s id like to make the request to have it moved from rate profile over to PID profile ( and the lowering of the default threshold setting from 1650 to 1050 ) 
as its is right now to use the TPA to help move up the cell count you'd need to change 2 profiles instead of just one and the second, rate profile, the only different setting you need is the TPA %..

thanks

SadisticLeprechaun, 2018-07-16:
This would be great, I alter my PIDs in different profiles for 4S or 5S or 6S but just changing some TPA would rock :)

TheMrJahn, 2019-06-16:
Is there any chances that devs will move TPA and throttle limiting to a PID Profile?
In BF 4.x we have great feature auto_profile_cell_count, so it would be great to auto change TPA setting. And one more - there is another parameter, that I'm changing for different count of cells - throttle limiting. I know there is alternative parameter "motor output limit" that locks to PID profile, but I like more throttle limiting.
IMHO - TPA and throttle limiting has nothing common with rates to be in a rateprofile...

theSkonk, 2019-06-16:
it was done for switching between race and acro rates.. remember this happened before trying to run  6s on 4s motors became a thing remember.....

Prajwal-Koirala, 2023-12-27:
**Here's a summary of the GitHub issue #6214, incorporating insights from the discussion:**

**Issue Title:** Moving TPA to PID Profile from Rate Profile and Changing Stock Threshold Value (Feature Request)

**Status:** Open

**Summary:**

- Users request moving TPA (Throttle PID Attenuation) settings from Rate Profiles to PID Profiles.
- They also suggest lowering the default TPA threshold from 1650 to 1050.
- The goal is to streamline profile switching for different cell counts and leverage TPA effectively for higher cell counts.

**Key Points:**

- **Current State:** TPA is tied to Rate Profiles, requiring adjustments in both PID and Rate Profiles for cell count changes.
- **Proposed Change:** Integrate TPA into PID Profiles for a cohesive experience.
- **Threshold Adjustment:** Lower default TPA threshold to 1050 for better performance with higher cell counts.
- **Auto Profile Switching:** Users highlight the potential synergy with the `auto_profile_cell_count` feature to automate TPA adjustments based on cell count.
- **Throttle Limiting:** Some users also request moving throttle limiting to PID Profiles for a similar streamlined experience.

**Discussion Highlights:**

- Users express support for the proposed changes, citing convenience and compatibility with `auto_profile_cell_count`.
- TheSkonk notes that the original placement of TPA in Rate Profiles was likely intended for easy switching between race and acro rates, before the prevalence of high cell counts.

**Next Steps:**

- Developers will evaluate the feasibility and implications of the suggested changes.
- Community feedback will be weighed to determine priority and potential implementation.

**Additional Insights:**

- Consider potential impacts on other features or settings before making adjustments.
- Prioritize changes that align with common usage patterns and enhance user experience.
- Explore alternative approaches, such as adding TPA to a separate profile category, if integration into PID Profiles is not feasible.

haslinghuis, 2023-12-27:
Fixed in #11779

closed 2023-12-27 by haslinghuis