betaflight/betaflight #6216: Betaflight f4 buzzer bepping all time

jjdiazgarcia, 2018-06-25:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

When i power on my betaflight f4, hellgate buzzer starts to beep all time battery is connected. I have armed quad and spin up motors and it is still beeping. 

I have updated fc firmware to betaflight 3.3.3.

I have soldered buzzer - pin from buzzer to buzzer - pin from fc, 5v pin from buzzer to 5v pin from fc and gnd pin from buzzer to gnd pin from fc.

```
# diff

# version
# Betaflight / BETAFLIGHTF4 (BFF4) 3.3.3 May 31 2018 / 12:55:00 (3565182d3) MSP API: 1.38

# name
name jjdg quad

# resources

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature ANTI_GRAVITY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1950 2100 0
aux 1 2 1 1800 2100 0
aux 2 13 2 1400 1600 0
aux 3 28 1 1350 1600 0
aux 4 36 3 1950 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT600
set failsafe_kill_switch = ON
set vbat_warning_cell_voltage = 36
set battery_meter = ADC
set beeper_inversion = OFF
set beeper_dshot_beacon_tone = 1
set small_angle = 180
set osd_warnings = 63
set osd_rssi_alarm = 50
set osd_vbat_pos = 2470
set osd_rssi_pos = 2476
set osd_flymode_pos = 2483
set osd_warnings_pos = 2409
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set osd_stat_tim_2 = OFF

# profile
profile 0

# rateprofile
rateprofile 0

set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
```

P.S. if you need it, i can post a video

etracer65, 2018-06-25:
Well you do have the modes set up to control the beeper with aux3. What kind of beeping are you getting?  Continuous tone or a beeping pattern?

Also, why do you have this in your config `set beeper_inversion = OFF`?

jjdiazgarcia, 2018-06-25:
If aux3 is out of range, beeping tone is continuous. If I activate aux3, beeping is a pattern.

I have set beeper_inversion=off because if I turn it on, motors beeping isn't correct.

etracer65, 2018-06-25:
That setting has nothing to do with motor beeping (known as beacon).  It only applies to the physical beeper you connect to the buzzer pins on the board.  For that board it should be `set beeper_inversion = ON` (which is the default).

Wait a minute, your description of how you have the buzzer connected sounds strange.  How many pins does your buzzer have?  It should only have 2 pins, + and -.  You connect these to the bz+ and bz- pins on the flight controller.

jjdiazgarcia, 2018-06-25:
Ok, i have done it trying differents configurations. If inversion is on, buzzer doesn't beep.

I have hellgate buzzer. It has 3 cables. It has buzzer - , 5v and GND.

etracer65, 2018-06-25:
I'm not sure that buzzer with work with that flight controller.

jjdiazgarcia, 2018-06-25:
I think it was due to a bug, but I don't know about It.
I have bought this buzzer because it has its own lipo to avoid not beeping if lipo gets disconnected when quad crashes.

mikeller, 2018-06-25:
@jjdiazgarcia: When you say

> If aux3 is out of range, beeping tone is continuous. If I activate aux3, beeping is a pattern.

Do you mean 'continuos without any interruption' or 'with short interrruptions'? It could be that your max battery voltage / voltage calibration is off, and your battery is recognised as 1 S higher than it is, but critically low.

jflyper, 2018-06-25:
@jjdiazgarcia 
Can you do some basic test to know how the hellgate buzzer works for us?

1. Connect 5V and GND.
2. Connect Buzzer to 5V.
3. Connect Buzzer to GND.

dustinb, 2018-06-26:
Not sure if related but another continuous buzzer issue.  On OMNIBUSF4 (Frsky XSRF40) my buzzer stays on continuously after disarming.  It doesn't start out on, only after arming and then disarm, buzzer comes on continuously and will not stop until I pull the battery.

I found these boards do this when put in bootloader mode which I had to deal with https://github.com/iNavFlight/inav/issues/2361 but this disarming thing is new for me, previously using 3.3 without this issue.

If I should start a new issue I can do that.

```test
# version
# Betaflight / OMNIBUSF4 (OBF4) 3.4.0 Jun 17 2018 / 08:44:35 (2fe9d7990) MSP API: 1.39

board_name OMNIBUSF4
manufacturer_id AIRB

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature RSSI_ADC
feature AIRMODE

# serial
serial 0 64 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0
aux 1 1 2 900 1300 0
aux 2 2 2 1300 1700 0
aux 3 13 1 1700 2100 0

# master
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT300
set align_board_yaw = 90
set ibata_scale = 179
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2455
set osd_rssi_pos = 2433
set osd_tim_1_pos = 2439
set osd_current_pos = 2446

# profile
profile 0

# rateprofile
rateprofile 0
```

jjdiazgarcia, 2018-06-26:
@mikeller when it is out of range, beeping is a continuous tone without interruptions. How can I test if it is what you tell me?

@jflyper what do you mean? solder cables to test those situations?

@dustinb i think my problem is a bit different because even if quad is armed or not, buzzer is beeping continuous without any interruption. Although quad is armed and motors are spinning up, buzzer is beeping.

jflyper, 2018-06-26:
@jjdiazgarcia
> @jflyper what do you mean? solder cables to test those situations?

Yes. We don't know when and how the hellgate buzzes, do we?
I'm concerned especially when the "Buzzer" is kept open or grounded, as the buzzer should be beeping when signal is absent.

jjdiazgarcia, 2018-06-26:
@jflyper I don't know if it is a good idea to connect 5V and GND because buzzer has its own lipo battery so i think result will be a short circuit. What do you think?

mikeller, 2018-06-26:
@jjdiazgarcia:

If the beeping is continuos without interruptions it's not the 'battery critical' beep.

I think what @jflyper meant is 'connect buzzer 5V to 5V source, connect buzzer GND to external GND'.

AndersHoglund, 2018-06-26:
You can use a standard buzzer with external battery, no need for any special kind of buzzer. Provided the FC has a circuitry similar to the schematics. Most FCs have that. I use this method with a 3S BFF3 setup, to increase sound level. Not for redundancy.
[image]

mikeller, 2018-06-26:
@AndersHoglund: I think one of the selling points of the Hellgate device is that it's got all the components built into one device - this way, even if you smash up your craft badly, there is a good chance that it will stay intact enough to be able to make noise...

supiiik, 2018-06-26:
@jjdiazgarcia as @jflyper wrote. Disconnect BUZ- wire from your FC and try to temporarily connect that wire to +5V, buzzer shouldn't beep. If you connect that wire to GND, than the buzzer should beep. With disconnected BUZ- wire is buzzer passthrought inactive as it has internal pull-up. 
HellGate buzzer works only on the boards, where the buzzer- pin is switched and buzzer+ has always +5V

AndersHoglund, 2018-06-26:
Yes, I know. My connection does that too, but maybe not under every possibly condition after a crash. Only battery redundancy. Open circuit on the buzzer minus (-) input side will not make any sound.

closed 2018-08-02 by stale[bot]