betaflight/betaflight #6222: Irc tramp, luminier (matek) vtx signal cutout

trintcole, 2018-06-26:
Video will momentarily blackout. It will come  right back after a second.
Worked fine with 3.3.  Loaded 3.4 and started fresh.  Did not copy diff or dump from 3.3.
Switched back to 3.3 and the problem is gone.

mikeller, 2018-06-26:
So it's actually the video feed from the camera and the OSD overlay that are both blacking out, and not just the OSD overlay?

trintcole, 2018-06-26:
Yes. Everything goes out.  Seems very similar to the vtx lower power on disarm issue in a previous version. I have since not used that feature either. Initially thought it was a vtx issue.  Then remembered i had just put the latest rc candidate on this frame for comparison. Flashed back and problem gone. 

When i upgrade I always do a fresh start by the way. I have 5 identical builds, so i build one from ground up the use the diff copy for the other 4 once i have the new one performing accordingly.

RustyToad, 2018-06-27:
I have this same issue. Relevant details:
Matek VTx, running latest firmware (1.9), connected to UART 1
Omnibus F4 SD FC

Channel is selectable, but power continually resets to 25 mW whether power level is set via the OSD or with physical buttons on the VTx. No higher power level is selectable.

VTx reboots sporadically, approx every 15-20 seconds in flight. Video drops for about 1.5 seconds while this occurs.

VTx is running the latest firmware (1.9).

This does not occur on a different build, using the same VTx but different FC.

I do not know if this is 3.4 related or not as the build it is happening on is new, and I have only used it with 3.4 RC2. As above though, it seems specific to this one build, my other using the same VTx, also on 3.4 (RC3), does not do it.

I initially thought this was related to the issue below, but now I suspect it may be down to a resource mapping clash on the FC, perhaps the UART address is clashing with something else - LEDs or similar. I have not got to the bottom of it but simply disabled the VTx control until I can sort it out. It is different enough from the issue below that I am not convinced it is related (despite apparent similarity).

https://github.com/betaflight/betaflight/issues/4917

mikeller, 2018-06-27:
@mateksys: Do you have an idea what's going on here?

MATEKSYS, 2018-06-27:
@RustyToad I remember Uart1 of F4 series don't support tramp protocol.  Try to connect it to other Uart.

MATEKSYS, 2018-06-27:
@trintcole  Did the blackout happen after powering on , or during flight?
once, endless circle or occasional?
What FC are you using

DieHertz, 2018-06-27:
I think we solved this issue, it had to do with USB Vsense or something.
If it doesn't work, try a 5-10KOhm pull-up to 5V

mikeller, 2018-06-27:
@MATEKSYS: I thought this problem was fixed in #5675, which went into #5675. Can you confirm that this change has not fixed the problem for the MATEK VTX?

MATEKSYS, 2018-06-27:
@mikeller  I have not seen #5675 before. Tested it on Uart1 of F405-STD just now.  fixed.

RustyToad, 2018-06-27:
@MATEKSYS thanks, will try a different uart tonight. At work now, will update in 8 hours or so.

jeffystockes, 2018-06-27:
Good morning, for information i have the same pb on my MatekF722 and TrampHV (immersionRC). It appends in flight. I lost video. The transmitter is on the UART5, protocol IRC Tramp. But after a first bug, i flash again my board + i desactivate UART5. Video come back, but after a moment (may be 3 mins of flight) , i lost again video... I don't have this problem with BF 3.4 RC 1, and i think with RC 2 (but with RC2 i don't do a lot of flights). I wonder if this is not a OSD problem, because with UART5 desactivate, the problem persist. Thanks for all. 

 diff

 version
 Betaflight / MATEKF722 (MKF7) 3.4.0 Jun 24 2018 / 09:56:39 (1c081ae8c) MSP API: 1.39

board_name MATEKF722
manufacturer_id 

 name

 resources

 mixer

 servo

 servo mix

 feature
feature GPS
feature RSSI_ADC

 beeper
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

 beacon

 map

 serial
serial 2 32 115200 57600 0 115200
serial 3 2 115200 57600 0 115200

 led

 color

 mode_color

 aux
aux 0 0 1 1900 2100 0
aux 1 1 2 1300 1700 0
aux 2 3 3 900 1175 0
aux 3 11 4 900 1175 0
aux 4 27 1 900 1175 0
aux 5 46 4 1875 2100 0
aux 6 13 1 900 1100 0
aux 7 28 2 1800 2100 0

 adjrange

 rxrange

 vtx

 rxfail

 master
set gyro_use_32khz = ON
set acc_trim_pitch = 2
set acc_trim_roll = -4
set acc_calibration = 8,-18,-1
set mag_hardware = QMC5883
set mag_calibration = -2373,14224,8399
set baro_hardware = BMP280
set min_check = 1000
set max_check = 2000
set fpv_mix_degrees = 25
set blackbox_p_ratio = 16
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT300
set failsafe_switch_mode = KILL
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_angle = 15
set gps_rescue_initial_alt = 20
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 278
set gps_rescue_min_sats = 4
set deadband = 4
set yaw_deadband = 2
set pid_process_denom = 3
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 49
set osd_cap_alarm = 1550
set osd_alt_alarm = 150
set osd_vbat_pos = 2145
set osd_rssi_pos = 2113
set osd_tim_2_pos = 2337
set osd_remaining_time_estimate_pos = 2389
set osd_flymode_pos = 2081
set osd_throttle_pos = 2358
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2242
set osd_mah_drawn_pos = 2371
set osd_gps_speed_pos = 2344
set osd_gps_lon_pos = 2065
set osd_gps_lat_pos = 2053
set osd_gps_sats_pos = 2049
set osd_home_dir_pos = 2093
set osd_home_dist_pos = 2095
set osd_compass_bar_pos = 2122
set osd_altitude_pos = 2262
set osd_power_pos = 2274
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2177
set osd_battery_usage_pos = 2376
set osd_disarmed_pos = 2442
set osd_nheading_pos = 2132
set osd_nvario_pos = 2294
set osd_core_temp_pos = 2421
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_battery = ON

 profile
profile 0

set dterm_notch_cutoff = 0
set p_pitch = 113
set i_pitch = 55
set d_pitch = 37
set p_roll = 75
set i_roll = 55
set d_roll = 20
set p_yaw = 188
set d_yaw = 15
set level_limit = 87

 rateprofile
rateprofile 0

set thr_mid = 36
set thr_expo = 65
set tpa_rate = 25

jeffystockes, 2018-06-27:
I do some tests. The problem is the VTX change band of frequency. I find the video by changing the settings manually with the micro switch. It's not a OSD problem. Sorry for my first post. :-(

MATEKSYS, 2018-06-27:
@trintcole Try to set the VTX power level to 100 or 600mW in BFOSD with BF3.3.  if it works normally.  the VTX should has V1.9 preloaded.  if it reboot endless each 10 seconds. that is the VTX firmware issue.

FrenzyIncarnate, 2018-06-29:
#6222 Same issue happened to me once.. This quad have 3.4 Rc1 installed.  Fc is Matek F405-CTR  and Matek VTX-Hv.. At close proximity to myself once the same blackout happened . Almost for a second and video came back. Landed immediately and not sure if I should try again.. Will it get fixed at offical 3.4 release? 3.4 RC1 is really good. Filters and PIDs are great.

RustyToad, 2018-07-01:
@MATEKSYS here is a follow on from my post above.

I have moved to UART3 and the problem persists.

I can select band and channel correctly through Betaflight. But power level refuses to go to anything except 25mW. If I select a higher level, either through BF or the buttons, it immediately resets to 25mW.

Additionally, the vTx seems to reboot sporadically. I cannot see a pattern. Sometimes it is frequently (every 20 seconds or so), sometimes it is very rare. It seems to happen more often in flight than on the bench.

As above I am running BF 3.4 RC2, and have the v1.9 firmware on the VTx. I currently have it disabled in BF and control using the buttons. Usable in my installation, but hoping for a fix.

My other quad, with a different FC and with the VTx on whatever firmware it was supplied with 6 months ago, on BF 3.4 RC1, works fine.

Full dump is [here](https://pastebin.com/RySKHmhy).

redneonrt, 2018-07-02:
@RustyToad 
the problem is with your vtx settings in BF, 
change these lines and get back to us
`set vtx_power = 4`
`set vtx_low_power_disarm = OFF`
`save`

when you first power on the quad you should see the vtx leds blinking just like when you manually push them, but then they should go solid within 10-15 seconds.

and dont forget to re-enable the tramp protocol on whatever port your connected to

Reverb2400, 2018-07-03:
Hi i have got a FPVMBFF7 board and a Tramp HV and having the same issue with all the 3.4 RC's. Today i found out that the issue starts when the tramp is powering down due to temperature. I also used vtx_low_power_disarm = ON and vtx_power = 4. May be the tramp is sending a lower powerlevel to BF but BF tries to crank it back up to 4. then i am losing signal from tramp but the led is still illuminating.
After some cooling with no battery connected back to normal.
Maybe this information is a little helpful.

FrenzyIncarnate, 2018-07-12:
As said I ve made the settings below at CLI. But now that I notice 4-5 secods after plugging battery vtx kinda restarts itself once.. Like losing connection screen goes weird and comes back. I m using Runcam split 2 with Matek VTX hv. and runcam is not getting the power from vtx

set vtx_power = 4
set vtx_low_power_disarm = OFF
save

closed 2018-08-18 by stale[bot]