betaflight/betaflight #6223: 能不能使用f3飞控制作双发固定翼垂起啊

miao19980215, 2018-06-26:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

mikeller, 2018-06-26:
Yes.

closed 2018-06-26 by mikeller

jflyper, 2018-06-26:
@miao19980215 I believe you are asking:

Can you use F3 fly control for dual-prop fixed wings?

And the answer is, as already given, yes.

However, I would recommend you take a look at iNav, too.