betaflight/betaflight #6226: 3 different quad update on 3.4rc3 wont arm

secretman28, 2018-06-27:
on new update, complete erase chip 3 quad canot arm, easy fix arming angle was set to 0 deg
i input the right value  and its' good 

just look if it was set default , i dont know how look at it but can me messing with starting people that not use to bf.

and the arming safety reporte it correctly in the configurator

secretman28, 2018-06-28:
i saud it on first post .. complete erase chip i always erase

mikeller, 2018-06-29:
@secretman28: In this case something with your backed up configuration is wrong - the default for small angle is 25, like already explained in https://github.com/betaflight/betaflight/issues/6226#issuecomment-400670376.

secretman28, 2018-07-02:
i dont remember if i was using the 3.0 or 3.1 configurator can be same as the gyro bug anyways.
 just trying  to hlep

avr39-ripe, 2018-07-03:
I solve this on my matek f405-std by 

set gyro_calib_noise_limit = 150

in CLI. try this and report back please!

closed 2018-08-09 by stale[bot]