betaflight/betaflight #6228: OSD not showing up on screen

Tbeal79, 2018-06-27:
ok try this again, sorry for the first one.

ok all i did with the JBF4 FC was update to the 10.3 for the new stuff coming out as every one has to do. then i updated to the new firmware to 3.3.3, did my pids and all, went out side to try it, all worked fine but i have nothing showing up for the OSD, no battery of anything shows up.
i know its there bc i tryed to go into the osd butwhen you do that you cant get out of the menu unless you disconect the battery and restart the quad.
but as i said i didnt do anything else but update.
***[BTFL_cli_pokeball_20180627_121619.txt](https://github.com/betaflight/betaflight/files/2142088/BTFL_cli_pokeball_20180627_121619.txt)***

if that didnt work, here is the visual dump

***# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.3.3 May 31 2018 / 12:56:13 (3565182d3) MSP API: 1.38

# name
name pokeball

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# map

# serial
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 27 3 1700 2100 0
aux 2 13 2 1700 2100 0

# adjrange
adjrange 0 0 1 900 1200 12 1
adjrange 1 1 1 1350 1650 12 1
adjrange 2 2 1 1800 2100 12 1

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 0
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set gyro_stage2_lowpass_hz = 200
set gyro_overflow_detect = YAW
set acc_hardware = NONE
set min_check = 1006
set max_check = 1996
set rssi_scale = 26
set fpv_mix_degrees = 25
set serialrx_provider = IBUS
set dshot_idle_value = 550
set motor_pwm_protocol = DSHOT600
set failsafe_throttle = 1300
set failsafe_kill_switch = ON
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1300
set deadband = 5
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set osd_warnings = 21
set osd_vbat_pos = 2456
set osd_rssi_pos = 2373
set osd_throttle_pos = 2405
set osd_warnings_pos = 2064
set osd_avg_cell_voltage_pos = 2418
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_tim_2 = OFF
set debug_mode = NOTCH

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass = 110
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set anti_gravity_gain = 3000
set setpoint_relax_ratio = 80
set dterm_setpoint_weight = 100
set d_pitch = 30

# rateprofile
rateprofile 0

set thr_mid = 40
set thr_expo = 35
set roll_rc_rate = 78
set pitch_rc_rate = 78
set yaw_rc_rate = 75
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 5
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 81
set tpa_rate = 20
set tpa_breakpoint = 1600 ***

mikeller, 2018-06-28:
Looks like your update went wrong. Try flashing again, this time with 'Full chip erase' selected. Or try updating to 3.4.0-RC3.

Tbeal79, 2018-06-28:
I'll try to reflash it, or I'll just do the 3.4.0.... funny thing is it was a full chip erase lol. I haven't flown this quad in like 3 months so when I got it out I said hey, I'm just going to update it to 3.3.3 like all the others and retune it with out using a diff paste.All was good till I didnt see any osd at all, could do the menu but couldn't see anything I was in lol.But thanks I'll go try that now

Tbeal79, 2018-06-28:
ill check it, but i didnt touch anything in the modes tab but the arm, failsafe, beeper.

Tbeal79, 2018-06-28:
wasent enabled, i never use that anyway, i never set it to a switch

Tbeal79, 2018-06-28:
well mikeller i did the update, full chip erase and everything, but it didnt change a thing...... i have the video like always, but no OSD at all, not sure wth is up with this. as i said before its the JBF4 fc, yes im using the right target for it, no i havent changed hardly anything, i have the diff for you right here.
should i try and go back 3.3.2??? and see if it comes back on with that??? kinda at a lose right now about it..
***[BTFL_cli_pokeJb_20180628_131351.txt](https://github.com/betaflight/betaflight/files/2146255/BTFL_cli_pokeJb_20180628_131351.txt)***

Tbeal79, 2018-06-28:
ok so here is where i am at, i went back to 3.3.2 and guess what? i have OSD im the screen with that version.. i will put the diff in there for you to check it out also and try to compair them to the others to try and help you guys fuigure out what is going on with the new update to version 3.3.3 and 3.4.0 RC3... so for some reason it wont let me have OSD in these two new ones.
here is that diff of 3.3.2.
***[BTFL_cli_pokejb_3.3.2_20180628_143445.txt](https://github.com/betaflight/betaflight/files/2146509/BTFL_cli_pokejb_3.3.2_20180628_143445.txt)***

Tbeal79, 2018-06-28:
ok so you wont belive this, but i just reflashed 3.3.3 and i didnt go in and change nothing, pluged the battery in to see if the OSD is working, and it showed up..... im trying to recreat why and what is making it do it.... bc i also knoticed that the beeper wasent working also when i was haveing the OSD issue..
then i was like im going to try 3.4.0 rc3 again. and i flashed it, did the same as 3.3.3 and pluged the battery in and boom i had OSD come up.
so im going to try and set all my stuff up, im not pasting in nothing when i been doing all this just so you know.
but im going to go set 3.4.0 rc3 to all my stuff and see if the osd stops again

Tbeal79, 2018-06-28:
all is good now, dont know what happen or if you guys found the problem

etracer65, 2018-06-28:
There haven't been any changes. I'm beginning to suspect you have a hardware problem with that board.  Your problem seems to be intermittent and not dependent on a firmware version or settings.  Possibly a loose/intermittent component causing the OSD overlay to not work.  Also there have been no other reports of problems like this.

Tbeal79, 2018-06-28:
Wouldn't dought that it's the board its self, it just magically fixed its self some how.And the reason I'm saying it prob is, is bc this is the second board I have gotten, the first had to be sent back bc of the sd card not working right, and now I have this problem when it does this dumb crap, and the sd still dont work.

etracer65, 2018-06-28:
All it would take is a loose component, cracked solder joint, etc.  Just handling the board, plugging/unplugging USB and temperature changes (the MAX7456 OSD chips tend to run pretty hot) could contribute to an intermittent problem.  It might seem to be fixed now but could potentially start acting up again.  Try pressing on different components and flexing the board slightly to see if you can induce the problem.

Tbeal79, 2018-06-28:
Yeah I def will try all that you said, but in going to bed for now, bc I been up past 36 hrs working on this quad.

closed 2018-08-05 by stale[bot]