betaflight/betaflight #6234: GPS Rescue not compatible with 3D mode

etracer65, 2018-06-28:
@s0up @mikeller 

Seems like we should consider disabling GPS Rescue if the user has enabled 3D mode.  The primary problem is that throttle management is not 3D aware and is simply based on an offset +- from the "learned" hover throttle.  This could potentially cause the throttle to drop below the 3D transition value around 1500 and reverse the motors.

This is complicated by the fact that there are some 3D modes where GPS Rescue would be okay. For example it the 3D disable switch was active, or "3D on a switch" was configured but not active.

Also take the worst-case where the pilot was inverted in 3D flight and then GPS Rescue was enabled (either by mode or failsafe).  The quad would flip over to normal orientation because of self-level, but throttle would still be in the reversed thrust region.

etracer65, 2018-06-28:
But even those special cases where 3D mode is okay for GPS Rescue (3D disable switch and "3D on a switch") both depend mode switches that we can't really trust in a failsafe condition.  To be safe for GPS Rescue under failsafe the mode for 3D disable would have to be on, while the "3D on a switch" would need to be off.

s0up, 2018-06-28:
I 100% agree with this.  I don't think there is any need to worry about the 3d modes that it would work with.  If you're flying the kind of range where you want rescue mode, you aren't flying 3d :P

mikeller, 2018-06-29:
This should at least be mentioned in a warning in the instructions for GPS rescue **and** 3D - an unsuspecting user might set up a craft for occasional 3D and long range flying.

etracer65, 2018-06-30:
Created https://github.com/betaflight/betaflight/pull/6247

closed 2018-07-03 by etracer65