betaflight/betaflight #6236: 3.4 Rc2/Rc3 OmnibusF4 quad won't rearm

Mattawonga, 2018-06-29:
Spedix is100 flytower F4 stack
R-xsr

Upon arming, I can fly the quad pretty well, flies great, but if I land/disarm I can't rearm to takeoff again. Have to unplug battery and replug. when disarming buzzer will go and keep beeping also... cant access osd after disarming.

Flashed Rc3 on a JBF4 on another quad... no problem there!

Tried Flashing Rc2, same issue... went back to 3.3.3 will works fine, issue only on 3.4 so far

SGURLY, 2018-06-29:
Got the same problem with the Aribot Nox v2.17 on Bf 3.4 RC 3...

jflyper, 2018-06-29:
@Mattawonga @SGURLY 
How about CLI `diff`?

etracer65, 2018-06-29:
@Mattawonga If you're using target OMNIBUSF4, then I think you should be using OMNIBUSF4SD.

SGURLY, 2018-06-29:
Targeted the NOX :/ @etracer65

jflyper, 2018-06-30:
@SGURLY 
`diff`

SGURLY, 2018-06-30:
`# version
# Betaflight / NOX (NOX1) 3.4.0 Jun 24 2018 / 09:58:51 (1c081ae8c) MSP API: 1.39

board_name NOX
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature AIRMODE

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 90
set acc_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_type = FILTER
set motor_pwm_protocol = DSHOT1200
set pid_process_denom = 1
set osd_vbat_pos = 2082
set osd_rssi_pos = 2103
set osd_current_pos = 2091

# profile
profile 0

set dterm_lowpass_type = BIQUAD
set dterm_lowpass2_hz = 250
set dterm_notch_cutoff = 0
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 90
set iterm_relax = RP
set iterm_relax_cutoff = 10
set throttle_boost = 8
set p_pitch = 63
set i_pitch = 65
set d_pitch = 40
set p_roll = 55
set i_roll = 60
set d_roll = 35
set p_yaw = 70
set i_yaw = 60
set d_yaw = 25

# rateprofile
rateprofile 2

set roll_expo = 14
set pitch_expo = 14
set yaw_expo = 14
set roll_srate = 80
set pitch_srate = 80

# `

@jflyper  I just saw that it says NOX1. am using the v2.17, maybe thats part of the problem?

EklipZgit, 2018-07-02:
I have the same issue with RC2, CLRACING F4S. When the issue happens the OSD freezes, mah counter stops counting up, no more voltage fluctuations etc. Can't get back into the OSD menu or arm the quad. Unplugging and replugging in the quad fixes the issue.

Sometimes it would happen if I didn't release left on left stick after releasing up on right stick when trying to enter the OSD menu.

jflyper, 2018-07-02:
@SGURLY Can you try this?
[betaflight_3.4.0_NOX-fixnonrearming.hex.zip](https://github.com/betaflight/betaflight/files/2153405/betaflight_3.4.0_NOX-fixnonrearming.hex.zip)

jflyper, 2018-07-02:
@EklipZgit Can you try this?
[betaflight_3.4.0_CLRACINGF4-fixnonrearming.hex.zip](https://github.com/betaflight/betaflight/files/2153409/betaflight_3.4.0_CLRACINGF4-fixnonrearming.hex.zip)

closed 2018-07-02 by mikeller

EklipZgit, 2018-07-02:
will try this tonight and let you know, thanks

EklipZgit, 2018-07-03:
@jflyper Your hex works great, seems to have fixed the issue. No issues re-arming now after a disarm :)

PR #6261 (2018-07-02): Call flashfsIsSupported instead of flashfsIsReady to check flash chip existence
Fixes: #6236 

There are number of cases that a flash chip is non-existent even if it is configured in target definition:
- Flash chip is selected as blackbox device on a flash/sdcard dual storage configuration board, but the board actually does not have the flash chip.
- A flash is provisioned as an optional daughter board.
- Incorrect target firmware, almost identical, but differs in flash v.s. sdcard is flashed (This is #6207 case).

This PR replaces calls to `flashfsIsReady` with calls to `flashfsIsSupported` ( [... cut]