betaflight/betaflight #6239: 3.4 RC3 - RC filtering with Crossfire

ITSFPV, 2018-06-29:
Feature #6103  does not seem to be working, as I am getting  d-term spikes with 3.4 RC3 when Crossfire moves from RFMD2 (150hz) to RFMD1 (50hz).  
I have rc_smoothing_type set to FILTER and input_hz & derivative_hz  set to 0 to allow the code to detect \retrain the optimal cutoffs.
However, seem to be getting +/- deflection of RCcommand \RCrates, which is causing pid errors and then subsequent d-terms spikes.
Both UAVTech (Mark) and Flint Lackman have reviewed the BB logs and have identified the potential issue. Please see  following screen captures. 

[image]
[image]

Build details  - 7" Freestlye quad: 

FC: Holybro Kakute F4 v2
ESC: Holybro Tekko32 4-1
RX: TBS Nano Crsf
TX: TBS Crossfire Micro 

Nano Settings
mode - 8ch
Telemetry - on
Force RF @50hz - off
Failsafe mode - cut
Rc link by MA?link - on

Micro TX Settings
Region - Open
Max Power - 100mw
Dyn Power - off
Frequemcy - 868Mhz
Op.Mode - Normal

Extract of Master & PID profile dff settings below
'''
# master
set gyro_lowpass_hz = 100
set rssi_channel = 8
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 25
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set current_meter = ESC
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_descent_dist = 150
set gps_rescue_ground_speed = 1500
set pid_process_denom = 1
set debug_mode = GYRO_SCALED
set vtx_power = 5
set vcd_video_system = PAL

# profile
profile 0

set dterm_lowpass_hz = 100
set dterm_notch_cutoff = 0
set anti_gravity_gain = 3500
set setpoint_relax_ratio = 10
set dterm_setpoint_weight = 100
set throttle_boost = 7
set p_pitch = 56
set i_pitch = 51
set d_pitch = 32
set p_roll = 53
set i_roll = 49
set d_roll = 27

# rateprofile
rateprofile 0

set thr_mid = 47
set thr_expo = 30
set roll_rc_rate = 98
set pitch_rc_rate = 98
set yaw_rc_rate = 107
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 82
set tpa_rate = 15
set tpa_breakpoint = 1100
set throttle_limit_type = CLIP
set throttle_limit_percent = 95
'''

etracer65, 2018-06-29:
Provide the output of the `rc_smoothing_info` CLI command while your rx/tx are on and connected.

Capture a blackbox log with `set debug_mode = RC_SMOOTHING_RATE` with a case where the Crossfire receiver changes rate from 150hz to 50hz (and hopefully back).

ITSFPV, 2018-06-30:
Will try to get this done in the next couple of days.

flint723, 2018-06-30:
In the log pictured in OP's post, the header shows everything correct (I think) for auto detect crossfire.  My impression from the rc command traces is that the rc link is never or almost never in 150hz mode.

The rc_smoothing debug will hopefully tell the story.
[image]

ITSFPV, 2018-06-30:
rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: NO SIGNAL
# Input filter type: BIQUAD
# Active input cutoff: 0hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 0hz (auto)

DieHertz, 2018-06-30:
You seemingly didn't power up the radio, receiver, etc

ITSFPV, 2018-06-30:
@flint723   My understanding was that the rc_smoothing code should in affect re sample the average RX frame rate and set new cutoffs values accordingly.  So if my Crossfire mode is  primarily operating at RFMD 1 (50hz), then should the code not have dynamically set the cut offs to around 25 and thereby smoothing  RCcommand inputs? 

The following is a screen shot of RCcommand when i manually set the cut offs to 25. As you can see this has reduced the rapid changes in pid errors and dterm spikes. Obviously, this is not good when its in RFMD2  at 150hz.
[image]

etracer65, 2018-06-30:
@ITSFPV I need a blackbox log with `set debug_mode = RC_SMOOTHING_RATE` and with the cutoffs set to 0 to see what the smoothing is seeing from the receiver and how it's adapting to the frame rate.  If the cutoffs aren't set to 0 then the detection/adapting logic is disabled.

ITSFPV, 2018-06-30:
@etracer65 I appreciate that, and will get these to you shortly.

ITSFPV, 2018-07-01:
Hi All, 
As requested, please find attached a bb log with rc debuging enabled. The log shows significant rc command steping\negative deflection from about 27 seconds on wards.
[BTFL_BLACKBOX_LOG_Nick_Torora_20180701_112210.zip](https://github.com/betaflight/betaflight/files/2152727/BTFL_BLACKBOX_LOG_Nick_Torora_20180701_112210.zip)

etracer65, 2018-07-01:
@ITSFPV Ok, I figured out what's going on. The rc smoothing detection/change logic is working correctly, but the problem is being caused by a conflict in how rx input is handled by the firmware.  It's very complicated, but the short answer is that the firmware has a "maximum time" that it will wait for the next rx frame before it needs to continue processing other periodic tasks.  Normally the incoming rx frames drive this process so that the tasks are synchronized to the newly received data.  But it just so happens that the "maximum time" the firmware will wait is 20ms - which is 50Hz.  So what's happening is that due to jitter on both the rx signal and the FC processing loops there are times when the next actual rx frame might be slightly longer than 20ms.  So say for example the next frame comes in at 21ms.  The firmware will still process at 20ms because the limit was reached, but then process again 1ms later when the new rx frame is received.  This is what's causing the problem.

The rc smoothing retraining logic is seeing the new 20ms frames from the rx and calculating a new average, but then one of the 1ms delays gets in there and resets the training.  You can see that in this screenshot.  The red line in the bottom graph is showing the measured frame delays and the yellow graph is the current "locked" frame rate.  Note the "glitches" in the red graph - those are the 1ms frames resetting the retraining.  As a result the "locked" frame rate never gets adjusted and the filters don't change.

[image]

But look a few seconds later in the log and the "glitches" settle down and the retraining actually does lock onto the new 20ms rate and adjust the filters. Notice how the rcCommand graphs at the top smooth back out again.

[image]

Then a little bit later your rx switched back to 150hz mode and the retraining correctly detects it and adjusts the filters back to the higher cutoff. Note that it looks like your SD card was having trouble keeping up with the 8KHz logging at this point (all the little squares are signs of dropped log entries).

[image]

So again the good news is that the rc smoothing detection/retraining is working correctly. But there are issues with the rx processing that will effect any protocol that ends up using 20ms or slower rx frames.

So we'll have to discuss internally the issue with the 20ms maximum rx frame delay limit and how best to correct that.

etracer65, 2018-07-01:
@ITSFPV In the meantime, can you try this:

[betaflight_3.4.0_KAKUTEF4V2.hex.zip](https://github.com/betaflight/betaflight/files/2152901/betaflight_3.4.0_KAKUTEF4V2.hex.zip)

It's based on 3.4RC4 and has the max rx delay slightly adjusted to hopefully avoid the glitches.

Please treat this as experimental and test initially without props for safety.  You should be fine taking a `diff` from your current 3.4 config and applying it to this version.

If you could try it and produce a log with the same `set debug_mode = RC_SMOOTHING_RATE` it would be really helpful.  I don't have access to Crossfire hardware so any help is appreciated.  Need to talk TBS into supplying some for testing... :)

etracer65, 2018-07-01:
Also I forgot to mention that it's not necessary to log at 8KHz. 2KHz would be fine.

ITSFPV, 2018-07-01:
thanks @etracer65 for the great explanation and the experimental fix.  I shall send you a log without props, but moving the tx further away to induce frequency changes.  Do you think this will make 3.4?

ITSFPV, 2018-07-01:
Please find attached logs without props... motors got quite hot. I may have done it for too long. 
[BTFL_BLACKBOX_LOG_Nick_Torora_20180701_210914.zip](https://github.com/betaflight/betaflight/files/2153158/BTFL_BLACKBOX_LOG_Nick_Torora_20180701_210914.zip)

etracer65, 2018-07-01:
@ITSFPV Thanks for the test. While that looks better, it's still not perfect.  We are discussing the best way to fix it as the hack I gave you was primarily just to confirm the problem and not intended as a permanent fix.  I'll get back to you when we have more info or a fix.

ITSFPV, 2018-07-01:
Thanks, i look to forward to seeing the outcome.

etracer65, 2018-07-01:
@ITSFPV It looks like a fix will go in tonight or maybe early tomorrow pending some testing.  So if you see this issue closed it's only because it's linked to the fix.  The fix (assuming testing succeeds) will go into a RC5 release that should happen quickly.

PR #6259 (2018-07-02): Change TASK_RX frequency from 50hz (20ms) to 33hz (30ms)
Fixes #6257 
Fixes #6239 

Workaround to better handle RX protocols that have frame intervals >= 20ms.  Otherwise the task would run at the end of the task interval causing slower protocols to run again after the remaining delay. This interferes with rc interpolation and also the frame rate detection in rc smoothing.

We should review `processRx()` in fc/fc_core.c and the processing it performs to make sure there's no impact from the potentially slower minimum frequency.  The minimum will only happen during signal loss situations or when [... cut]
closed 2018-07-02 by mikeller