betaflight/betaflight #6243: Proshot1000 cpu usage much higher than in the past

flint723, 2018-06-30:
Noticed this a bit ago.  Is it expected that on f7, Proshot1000 has significantly more cpu usage than Dshot1200? Board is spracing f7 dual (operating on single gyro).  Seems this was always the case on this board (at least from 3.4 build 958 back to build 860).

32k32k multishot: 27%
32k32k Dshot1200: 34%
32k32k Proshot 1000 45%

But in previous versions of 3.4 on another ICM board, Matek f405, Proshot1000 cpu was much less than Dshot (similar to Multishot).  Now on matek f405 FW 3.4.0 958, 32k16k Proshot and Dshot1200/600 are pretty much equal.  When I get time I will flash this board back in 3.4 builds and see when it changed.

Additionally HESP 32k32k Proshot1000 has similar cpu usage as multishot, with Dshot 1200 higher.

DieHertz, 2018-06-30:
It's hard to compare CPU usage across different targets, let alone different architectures (F4/F7). I'm not sure it was even better on your particular board (SPRF7DUAL). It was (maybe is) better on Matek F722 though.

flint723, 2018-06-30:
I agree, f7 dual probably isn't a good example.  I'm not trying to compare across boards, only against multishot and Dshot on each board.

Matek f405 32k/16k:

Build 700 to current: MS = 46%, Dshot600 =52%, Proshot 1000 = 58% (not flying just left overclock off to show difference better).

I can only conclude that the Configurator is reading cpu differently than it used to.  I know for a fact I ran proshot on this board because it had less cpu than dshot.

etracer65, 2018-06-30:
I don't see any difference from before. Just tested with current master on board with ICM20608G in 32K and it's:

multishot -> proshot -> dshot

Small increase in load between each step (like 1% at most).  DSHOT600/1200 are the same.

flint723, 2018-06-30:
Matek f405 32k16k:
3.4 build rc2-962, Configurator 10.3.1

- MS = 49%
- Dshot600 = 52%
- Proshot 1000 = 61%

Not really an issue I guess, just that Proshot used to be lower than Dshot (that was always one of the selling points of the protocol) on this board.

DieHertz, 2018-06-30:
I think this selling point was pretty much taken out of thin air, as there's no reason for ProShot to be less CPU intensive.
The complexity of forming a packet for DSHOT and ProShot is roughly the same, while DMA load isn't a part of CPU load.

etracer65, 2018-06-30:
I think the "selling" point of ProShot is only lower utilization on the ESC side. More interrupts then multishot, but fewer than the serial DSHOT protocol.  But that trade-off is more chance for packet corruption since you now have a PWM pulse to measure that represents 16 possible values (a nibble).

DieHertz, 2018-06-30:
Yes, interrupt load in the receiving side will be lower

jflyper, 2018-06-30:
ProShot, by it's nature, is more CPU intensive than DShot, as it does extra symbol encoding (and decoding) which brings the lower clock at the same data rate.

closed 2018-07-01 by flint723

ledvinap, 2018-07-02:
Proshot CPU usage may be lower than DShot - processing is done 4-bits at time, so there are less interactions in encoding loop with slightly more processing per iteration (1 multiplication + 1 addition).