betaflight/betaflight #6245: 3.4 RC3 OSD stuck at bootup

Jensi, 2018-06-30:
After powering up (VBat or USB) the FC, the OSD get stuck on the boot screen (BF-Logo + "MENU:  THR MID....)

The problem occurs when i enable "RunCam Device" in the Ports tab (UART5 is the camera tab)

My FC is the "Fortini F4 OSD" made by FuriousFPV
as Camera i use the RunCam Swift2 
i wired everything up like in the manual page 7 and 13 but i use a 5V stepdown to power the VTx (5V version - unify pro v3)
(manual: http://furiousfpv.com/document/Manual_Furious_/User_manual_Fortini_F4_OSD.pdf)
other components: F40 pro v2, Aikon AK32 35A, r-xsr and Matek PDB

i tried "Betaflight 3.4 RC3" and "Betaflight 3.4 #3 (AKK patch)"

```
# diff

# version
# Betaflight / FF_FORTINIF4 (FORT) 3.4.0 Jun 24 2018 / 09:46:13 (1c081ae8c) MSP API: 1.39

board_name FF_FORTINIF4
manufacturer_id 

# name
name El_Burrito

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 1024 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0
aux 1 13 2 1700 2100 0
aux 2 28 0 1700 2100 0
aux 3 35 1 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_use_32khz = ON
set acc_hardware = NONE
set rssi_channel = 8
set dshot_idle_value = 350
set motor_pwm_protocol = DSHOT1200
set current_meter = ESC
set battery_meter = ESC
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set osd_vbat_pos = 2467
set osd_rssi_pos = 2105
set osd_tim_2_pos = 2486
set osd_current_pos = 2435
set osd_mah_drawn_pos = 2455
set osd_disarmed_pos = 2346

# profile
profile 0

# rateprofile
rateprofile 0
```

mikeller, 2018-06-30:
@azolyoung: Can you help here?

azolyoung, 2018-07-01:
@mikeller Yes, I'm looking into

azolyoung, 2018-07-01:
```
# serial
serial 0 1024 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 5 32 115200 57600 0 115200
```

@Jensi Hi, first look in your diff result, I didn't see you have enable `RunCam Device` , and you are enable SMARTPORT in uart6, is that right? And as I know, SmartPort are not compatiable with RunCam Device, so you can't  enable  them in same uart

Jensi, 2018-07-01:
`SmartPort` is UART6 and `RunCam Device` is UART5
and yes I have i postet the wrong diff with RunCam Device disabled 😅
therefore here:
```
# serial
serial 0 1024 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 4 16384 115200 57600 0 115200
serial 5 32 115200 57600 0 115200
```

Jensi, 2018-07-01:
I tried some things out:
- flashed BF 3.4 RC1 RC2 and RC3 only enabling `RunCam Device` at UART5 and boom - the ODS stuck
- enabling `RunCam Device` on other UART's (tried only with RC3) and ther was no problem, but the dedicated dedicated `Camera control` pin is UART5 so....
- flashed `BF 3.3.3` and everythig works so i assume it's not a hardware failure

azolyoung, 2018-07-01:
@Jensi 
But I can't find UART5 in the manual you post,  please take a look with the following image, are you wiring that pin to osd pin of swift 2?
https://s3.amazonaws.com/rctmpfiles/屏幕快照+2018-07-01+下午3.23.47.png

and more info, Are you using a runcam part adapter??

Jensi, 2018-07-01:
yes i wired it up like in the image
and in the manual there is nothing about UART5 that's right, but the `Camera OSD control / LED Strip` pin is UART5

azolyoung, 2018-07-01:
oh, may be I know something, I get into the target defines of FF_FORTINF4, and I see the pin with arrow in the image is a cam control pin(PB7), it's not a uart 5(the TX and RX of uart5 is PC12 and PD2), it's a pin using with LED & cam control. this cam control pin can work the feature developed by DieHertz, so if you using this pin for cam control, you no need to enable RunCam Device on uart5.

and about uart5,  there is no available solder pads for it in the manual, I think uart5 is for the FC that has not selling yet  or other model of FC, so it's not available for you board. 

available |  
-- | --

Jensi, 2018-07-01:
that's a bit confusing
in the Ports tab the Identifier of that "thing" i enabled `RunCam Device` is caled `UART5` so that is why i asumed it IS UART5
so the pin is not uart5 but i have to enable `RunCam Device` on uart5?
but that is what i did 😵

azolyoung, 2018-07-01:
haha,it is a bit confusing, in the target defines, it define uart5, so you can see it in the BF Configurator, but in the FC board, there is no uart5 in it, I think the uart5 issue you can get answer from FuriousFPV

Your finally target is make cam control on your fc, is that right? Your wiring is correct, and you just turn on your transmitter, and pull yaw to right and keep other three stick to center, then the OSD menu of camera will showing, the detail is in the `How to open camera setting by Transmitter` section of the manual.

azolyoung, 2018-07-01:
The osd will stuck, even you don't show the camera osd menu?
Is the fc hung when this issue happen?

azolyoung, 2018-07-01:
I only have rcsplit and fc in my hand, for now can't reproduce this problem. may be there a conflict between cam control and runcam device protocol, I'll test it tomorrow when I arrive office

And I just wonder, is the fc hung when this issue happen?

Jensi, 2018-07-01:
the FC is working normaly
only the boot up screen problem occurs (so fare i can tell)

azolyoung, 2018-07-02:
@Jensi 
I test in my quad this morning, the osd does not stuck, camera control work fine, and I have enabled runcam device at one of uart in the FC.  I was test in BF3.4 RC3 and BF3.4 RC4.

Could you try to enable runcam device in other uart for test(not uart5, I suspect that this UART5 is a bit problematic because there is no UART5 interface in the manual and the flight control hardware.)， and It's better to test it with BF3.4 RC4([here](https://github.com/betaflight/betaflight/releases),
Thanks for your help)

Jensi, 2018-07-03:
> enabling RunCam Device on other UART's (tried only with RC3) and ther was no problem, but the dedicated dedicated Camera control pin is UART5 so....
i'll try RC4 tomorrow

azolyoung, 2018-07-05:
Thanks! I'm wonder how do you know the uart5 is a camera control pin? I get into the target defines of Fortini F4, and the camera control pin is PB7, it also can used as a led signal pin, and uart5 tx and rx are pc12 and pd2, they are different,

Jensi, 2018-07-05:
how i know? in the Ports tab there was only one UART that was not mentioned in the manual. So i tried it out. Enabled RunCam Device and it worked so...

And i just tried RC5 and the problem still occurs
enabled = camera control works but the OSD is stuck
disabled = OSD works but of course no camera controll

azolyoung, 2018-07-06:
OK, from the situation you tested, UART5 may be related to camera control. for now I have no idea about this issue. I have found a local store that selling Fortini F4 flight control (the only one on taobao). I will buy one piece today to test this issues. Thank you for always helping to test this issue.

azolyoung, 2018-07-06:
Magical courier, I have received my fortini f4 osd, I will test this issue later)

azolyoung, 2018-07-06:
Hi @Jensi , I've test it, I wiring my x8r, tx200u, micro swift to the board, and flash 3.4RC5 into the board, by default, everything work fine, the camera control work fine as well. but when I select `RunCam Device` on uart5, and tap `save and reboot`, the osd is stuck, it's same with you met. then I try to disabled it, and `save and reboot`, bf osd work correctly. 

And I see the manual of the FC, there is no  tx5  pad on the board, only rx5. so I guess maybe we can't send data to uart5 with tx5, so I choose the other peripheals, e.g IRC tramp, smart audio, the bf osd is not work correctly:
1. when choose IRC Tramp on UART5 , the BF OSD does not appear
2. when choose smart audio on UART5, the BF OSD was stuck like `runcam device`
3. when choose blackbox logging on UART5, the bf logo on osd is correct, but after logo disapear, the BF osd was freezed, BF CMS menu、camera control and bf osd elements don't work, buf fc was not hung.
4. when choose gps on UART5, it's same with blackbox logging, the cmsmenu, camera control, osd elements did not worked. and fc was not hung
5. when choose esc on UART5, it's same with blackbox logging...
6. when turn on RX on UART5, and disable RX on UART3, click save and reboot, it's same with blackbox logging, but as I know, if turn on RX on UART5, the FC won't send data to the UART, it's only receive data from UART, so why it would freeze when turn on RX on UART5? no idea~~

So, I think the problem is not about the runcam device protocol, it's the BF firmware not compatiable with fortini f4, maybe something wrong in their target defines.

azolyoung, 2018-07-06:
@Jensi the cam control pin on fortini f4 is using the `Camera control` feature that originally designed by @DieHertz , this feature is not need  to turn on `runcam device` or other feature on any UART port,  the uart5 is not relate with camera control, and the camera contorl pin is not uart5, it just a normal IO PIN that used for camera control or led strip, if your osd menu can't display when you turn off `runcam device` on uart5, maybe it's something wrong of your camera control setting, the camera control setting can fit most camera by default, not all camera, might be you need adjust the setting of cam control.

azolyoung, 2018-07-06:
Hi @mikeller, What do you think about that?

mikeller, 2018-07-07:
@azolyoung: Thank you so much for testing this.

@azolyoung, @Jensi: Taking @azolyoung's finding into consideration, and looking at the target definition for the FF_FORTINIF4, it turns out that for this target, the PB12 pin are defined for both, UART5 TX:

https://github.com/betaflight/betaflight/blob/ad975d3948d0c5f66b59fe397f6cf9dea31bf411/src/main/target/FF_FORTINIF4/target.h#L137

and SPI3 MOSI (used for OSD):

https://github.com/betaflight/betaflight/blob/ad975d3948d0c5f66b59fe397f6cf9dea31bf411/src/main/target/FF_FORTINIF4/target.h#L188

This explains why the firmware will hang whenever OSD and something that uses UART5 TX are enabled at the same time. Setting `resource serial_tx 5 none` should probably fix this - but UART5 TX won't be useable in any case.

@MJ666 is the maintainer of this target, and should look into fixing it.

closed 2018-08-13 by stale[bot]