betaflight/betaflight #6255: Feature request: make RX_FRSKY_SPI_LED_PIN re-mappable/re-configurable on crazybeeF3

Nisker, 2018-07-01:
It would be nice to be able to control the function of the Frsky leds, its the 4 bright leds on the FC.

At the moment it blinks when its not receiving data from a transmitter, and constantly on when bound.
Right now it cannot be remapped to another function or at least it won't be able to control it.

The idea is to be able to set it as a buzzer so when flying LOS it can indicate low battery, or blink when a switch is toggled. or map it to PINIO to be able to turn it off when flying FPV in low light, the white leds right next to the frame is quite visible and annoying in the display (in low light)

i don't know for sure, but i think it's just some small changes in "src/main/rx/cc2500_frsky_shared.c" and maybe "src/main/target/CRAZYBEEF3FR/target.h". 2 other targets (MIDELICF3/MATEKF411RX)  also use RX_FRSKY_SPI_LED_PIN.

Antimatter88, 2018-07-05:
Seconded; that'd be very sweet; I fly low light a lot but those LEDs have really been an issue; was even thinking of desoldering them

mikeller, 2019-02-28:
@Nisker: I think this has been addressed now:

https://github.com/betaflight/betaflight/blob/1507cc880c493b13e89d61c821c054d5f78d0d2d/src/main/rx/rx_spi_common.c#L42

phobos-, 2019-03-05:
@mikeller is correct, this feature request has been addressed by #7109

closed 2019-03-05 by mikeller