betaflight/betaflight #6257: RX processing incorrectly handles protocols with frame rates >= 20ms

etracer65, 2018-07-01:
@mikeller 

I haven't yet figured out the exact cause/fix, but `rxUpdateCheck()` in rx/rx.c is not properly handling rx frame rates that exceed the `TASK_RX` desired period of 50Hz (20ms).

So take for example 22ms Spektrum. The effect it's causing is that at 20ms the downstream logic fires as if there was a new rx frame even though there wasn't.  So the `currentRxRefreshRate` variable gets set to 20ms.  Then 2ms later the actual rx frame comes in and this causes the downstream logic to run again this time with `currentRxRefreshRate` equal to about 2ms.

I don't think this is specific to 3.4 and believe the problem exists in 3.3 as well.

But it becomes a much bigger problem with RC Smoothing enabled.  Because the detected rx frame rate gets incorrectly calculated and the filters get set far too high causing noisy inputs.  Take the example of PPM with 27ms frames.  The 20/7 effect ends up causing the frame rate to be detected at around 13.5ms vs. the actual 27ms.  Making the filter cutoffs way too high.

The reason I found this was because of problems with CRSF properly detecting the rx frame rate changes.  In reality it's working fine but since CRSF switches to 50Hz it runs into the 20ms problem.  So slight jitter causes the logic to run at the fixed 20ms (before the frame comes in) and then again < 1ms later when the frame arrives.  This jitter is preventing reliable detection/retraining.  But end the end it's all a side effect of the 50Hz `TASK_RX` and firing based on the time limit rather than the actual rx frame.

I looked at the logic and on the surface it shouldn't be a problem because `needRxSignalMaxDelayUs = DELAY_10_HZ;` so it seems like the 100ms should be the limit - not 20ms.

etracer65, 2018-07-01:
Ok, I think I figured it out.  Not completely sure about the best approach to fix.  In `rxUpdateCheck()`:

```
    if ((signalReceived && useDataDrivenProcessing) || cmpTimeUs(currentTimeUs, rxNextUpdateAtUs) > 0) {
        rxDataProcessingRequired = true;
    }
```
The problem is that `rxNextUpdateAtUs` is defined as:
```
    rxNextUpdateAtUs = currentTimeUs + DELAY_50_HZ;
```
in `calculateRxChannelsAndUpdateFailsafe()`.  So that's driving all the downstream logic at the 20ms interval even if the rx frame hasn't come in yet.

etracer65, 2018-07-01:
The other aspect of this is that we're running the downstream rx processing unnecessarily multiple times for each rx frame in cases where the protocol is >= 20ms.

The overall problem seems to be the need for 50Hz processing of the ancillary tasks vs. the actual rx frame processing at >=20ms.

etracer65, 2018-07-01:
Also it's not just RC Smoothing that's affected.  This does a number on interpolation as well and makes the setpoint weight derivative a mess.

I'm wondering if as a workaround for the rx frame rate we can move the `currentRxRefreshRate` calculation and `isRXDataNew` flag setting up into rxUpdateCheck() and base them on the actual rx frame arrival.  This would fix the interpolation/RC Smoothing problems I think but wouldn't do anything for the extra calls to `taskUpdateRxMain()` so it's kind of a hack.

mikeller, 2018-07-01:
Hmm, would setting the forced update interval to something sure to be slower than any RX frame interval (100ms?) take care of this? As far as I can tell, the forced update is just there to keep all the different bits of status update that are part of the RX loop from starving when no RX signal is coming in.

etracer65, 2018-07-01:
Well, yes that would fix the rx frame problem but I'm not sure it's a good solution.  Things like `processRx()` in fc_core.c get called and it's kind of expecting a faster refresh.  For example it has the logic to deal with Runaway Takeoff deactivation, etc.  We'd have to take a look at the whole chain to see what might not like the much slower (potential refresh).  We could probably get by fine with changing to a 33Hz (30ms) refresh to at least cover us through PPM.  I'm not sure how PWM would go - probably just end up being the taxk frequency as it doesn't really have "data ready".  But even at 33Hz we'd have to check for potential problems in the `processRx()` chain.  I did a quick look and don't see any major problems.  The deactivate phase of Runaway Takeoff should be okay at 30ms.  I might be a little concerned with `calculateRxChannelsAndUpdateFailsafe()` in rx/rx.c.

BTW - I did try moving the calculation for `currentRxRefreshRate` and setting `isRXDataNew` up into `rxUpdateCheck()` and that solves the rx frame rate issues with interpolation and rc smoothing.  But of course it doesn't fix the problem with processRx() getting called irregularly and unnecessarily (burning a lot of cycles).

I hacked together a quick test for changing the TASK_RX to 33Hz and tested with SBUS (9ms) and PPM (27ms) and both worked as expected.  Do you want me to put together a PR for this?  Even so I'd appreciate another set of eyes on the `processRx()` chain to spot any potential problems with the slower cycle time in the event of signal loss.

etracer65, 2018-07-01:
Tested with PWM and it was fine at 30ms.

mikeller, 2018-07-01:
First of all, I think there are two different issues that we need to look into:
1. fixing what has to be fixed for 3.4, in a way that has the smallest possible risk of regression;
2. looking at how the whole rx loop can be overhauled, for after 3.4.

Happy to provide a second set of eyes for both of these, but 1. is more expedient than 2, obviously. If we get 2. done early after 3.4, we can get a lot of exposure, which will allow us to shake down eventual problems before releasing it.

With respect to the RX loop, I think there are two different types of functions that are currently lobbed together in the RX loop:
1. data driven functions that need new input data to work;
2. not data driven functions that consume RX input, but can work on stale data as well.

1. can probably be safely left in the RX loop even if we make it much slower - executing these without new data does not provide much value;
2. (into which runaway takeoff prevention probably falls) should probably be moved into a different loop where it is independent of (but potentially coupled to) the RX loop.

Also, making the minimum task frequency slower than the commonly expected minimum RX update frequency is probably a good candidate for the fix for 3.4.

etracer65, 2018-07-01:
Agreed on all.  Definitely a candidate for a major overhaul - but 3.4 just needs a quick fix.  I'll submit a small PR.  There's a one-line (remove a delay) fix also for rc smoothing retraining that I'll include.  It's just a delay that doesn't need to be there.  Doesn't technically hurt anything but could slow down latching onto the new rate if there rx frame rate has some instability (like with this rx frame rate issue which showed it).  The delay was only meant to be a buffer when the retraining completed and locks onto a new rate to prevent too rapid switching but it erroneously got set when the training is interrupted by an out of range packet.  All that should happen then is the training resets - the guard delay shouldn't be there.

closed 2018-07-02 by mikeller

ctzsnooze, 2018-07-02:
[image]
added this here just for wiki - thanks!

mikeller, 2018-07-03:
@ctzsnooze: An easier (maybe) and more portable way of doing this is to clone the wiki repository (https://github.com/betaflight/betaflight.wiki.git), and then stick your images into the `/images/` folder.

PR #6259 (2018-07-02): Change TASK_RX frequency from 50hz (20ms) to 33hz (30ms)
Fixes #6257 
Fixes #6239 

Workaround to better handle RX protocols that have frame intervals >= 20ms.  Otherwise the task would run at the end of the task interval causing slower protocols to run again after the remaining delay. This interferes with rc interpolation and also the frame rate detection in rc smoothing.

We should review `processRx()` in fc/fc_core.c and the processing it performs to make sure there's no impact from the potentially slower minimum frequency.  The minimum will only happen during signal loss situations or when [... cut]