betaflight/betaflight #6262: FC Bricked After Firmware Update

brunoleaomaia, 2018-07-02:
My FC Bricked After Try Update Firmware to 3.4.0 RC4. 

My FC is HGLRC F3 AIO V4 and was running 3.3.3 before this. 

Now I cant flash again, I've tried boot mode and st flash load demonstrator.

FC:
https://www.hglrc.com/products/hglrc-f3-v4-fc-aio-osd-bec-pdb-current-sensor-25-200-600mw-vtx-for-rc-racing-drone

mikeller, 2018-07-02:
@brunoleaomaia: It is almost impossible to permanently brick a flight controller with a firmware update, since the boot loader mode is part of the ROM of the MCU. If using the boot button does not result in the board being recognised by the PC, have a look to see if there are any conflicts in the device manager, and reinstall the drivers liste in the configurator start page.

DieHertz, 2018-07-02:
What was the target name on your FC before flashing and which target you flashed?

brunoleaomaia, 2018-07-02:
@mikeller using boot button the board is recognized as CP210x USB to UART Bridge (COM8).

@DieHertz the target before and flashed was SPRACINGF3.

mikeller, 2018-07-02:
@brunoleaomaia: Ok, in this case, bringing the board into boot loader mode with the boot button, and then flashing the latest version of the SPRACINGF3 target with 'Full chip erase' enabled should bring your board back to live.

brunoleaomaia, 2018-07-02:
@mikeller I've tried many combinations. This way like you said the answer is 'STM32 Communication failed, wrong response, expected: 121 (0x79) received: 0 (0x0)'

DieHertz, 2018-07-02:
Looks like you need STM Software to remove read/write protection.

jflyper, 2018-07-02:
@brunoleaomaia One simple and basic question. Is this the first time you flashed a firmware on to the board under question?

brunoleaomaia, 2018-07-02:
@jflyper, no! I was updating all stable versions since 3.2.1. First time to flash release candidate version. I was wating 3.4 release yesternday, because RC4 was released I tried this. Now I can't connect the FC, same in STM Software. I think that I need reflash the processor directly.

mikeller, 2018-07-03:
@brunoleaomaia: I suspect that in your attempts to flash the firmware with STM tools you might have locked the flash in your MCU. Flashing through DFU from configurator already uses the most basic way to flash, so if this does not work there probably isn't a way to save this board.

Ant73, 2018-07-03:
@brunoleaomaia 
Have you tried the Impulse RC driver fixer?
https://impulserc.com/pages/downloads

RipperDrone, 2018-07-03:
maybe try https://www.youtube.com/watch?v=9wwQZPGugoc

RipperDrone, 2018-07-03:
or https://www.youtube.com/watch?v=Om6teCbJM4U

brunoleaomaia, 2018-07-03:
I've tried all this things (less method 2 of above video link). I had a Naze32 Bricked before and I restored this with STM software. I know the process. My objective is to know if have some problem with new configurator, generating this issue. 

Thanks All!

TubeRider, 2018-07-24:
Same issue with the FuryF3osd flashed and now bricked.
Flashed countless FC's with no issues, can still access the DFU by boot button and re-flash with no result, even with older versions.
Have connected to 4 different computers and its still dead, it ain't the drivers.

closed 2018-08-30 by stale[bot]