betaflight/betaflight #6268: [Feature Request] UART MSP Setting from LUA or/and OSD

mickkn, 2018-07-02:
Hi Devs

My USB just broke off (with solder pads) on my XSRF4PO and that just nags me a lot.

I could really use a feature for changing one of my UART's to MSP and use a FTDI adapter to setup the FC and even update it to a few firmware. Don't know if the last request is a long shot.

The feature could be a way in the OSD setting or/and in a LUA script to change a UART to MSP. There should maybe, be a restriction from changing a Serial RX UART to MSP, since it will disable Stick commands, unless the LUA script is supported also :)

And another thing would be to have MSP on a UART as default (don't really know if this is the case at the moment, and I know one could make a custom build for this. But why not, just default?)

mikeller, 2018-07-03:
Hi @mickkn.

@AndersHoglund does maintain some builds with MSP enabled by default on an UART: http://andwho.sytes.net:8080/job/AndWho_Betaflight/ If your board isn't on there, asking him nicely might help. :wink: 

If you are set up to build the firmware yourself, you can always build with `make <target> OPTIONS=USE_MSP_UART` to get a target with MSP on UART1.

The reason this isn't done by default is that this causes problems if users connect an RX or another 'chatty' device to UART1, and then reset to defaults - the flight controller boots, and starts processing the incoming data on UART1 as MSP commands, typically completely wedging the board.

Good idea about adding a MSP bailout to CMS! This won't be able to solve cases where the firmware is reset to defaults, but it will work for users who still have a working configuration.

mickkn, 2018-07-03:
Yeah, the MSP as default makes sense :)

The problem on XSRF4PO is that it has a build in XSR receiver, meaning that it is only UART 3 that is free from hardware. Is there a #define for choosing the uart number with that build parameter?

mikeller, 2018-07-03:
@mickkn: No, there isn't. But if you're already set up to build from source, you can just edit https://github.com/betaflight/betaflight/blob/463d7aa1c1fa97a32246d283af82340bb2219280/src/main/io/serial.c#L143

and replace `SERIAL_PORT_USART1` with `SERIAL_PORT_USART3`, and then build with `make <target> OPTIONS=USE_MSP_UART`.

mickkn, 2018-07-03:
Thanks a lot. This totally solved my problem :)

I even managed to enable Soft Serial TX on Motor 6 output for Tramp control. That was a bit weird to accomplish, since I had to resource SERIAL_TX 11 to the A09 pin to make it show in the ports tab. Don't know it this how it used to be, when I used Soft Serial last. But it works :)

One question. Will I be able to update via the UART3 port? (Have to remember to use a custom firmware version of course)

jflyper, 2018-07-04:
@mickkn 
> One question. Will I be able to update via the UART3 port? (Have to remember to use a custom firmware version of course)

For F405, yes.

[image]

closed 2026-09-26 by blckmn