betaflight/betaflight #6269: OmniNXTF7 OSD Vario wont work

RoadyFPV, 2018-07-02:
I use the OSD Vario values and arrow.
In 3.4 RC4 it wont work. Value are frozen at 0.0 and arrow up.

The Altitude is working correct. Set to zero when ARM.

mikeller, 2018-07-03:
@RoadyFPV: Vario is not supported in 3.4. Sorry.

RoadyFPV, 2018-07-03:
Oh,
Is it possible to bringb it back?
Or why is it not supported?

mikeller, 2018-07-03:
The problem is that the code for vario (i.e. the altitude estimation) was old and unreliable. It was removed, and replaced with the new GPS rescue code.

We can look into generating a better vario signal from the altitude that is tracked by the GPS rescue code if there still is interest in getting vario data in Betaflight.

MrLAtt, 2018-07-06:
I have also just come across this issue, I guess i will have to revert back to 3.3 unless it is reimplemented. Thanks for the help, I thought itwas something i was doing wrong.

RoadyFPV, 2018-07-08:
@mikeller it would be nice to bring it back on the OSD.

mikeller, 2018-07-08:
@RoadyFPV: We can look into making vario work again - it just wasn't clear if users were actually using it, since Betaflight isn't that much focused on fixed wing.

MrLAtt, 2018-07-08:
Well, my most frequent use of betaflight (and the variometer) is as an OSD for an 'FPV Backpack', I have it mounted to a bit of plywood with a camera and transmitter with an omnibus f3 board, it's a very simple way to get a full OSD in a small form factor that is easy to program. However I now also tend to have the variometer enabled with my quads next to the altitude, just because it's often nice to know.

mikeller, 2018-07-08:
@MrLAtt: Ok, I don't quite get what benefit you get out of having Vario on your backpack, and certainly not on your quad, compared to the improved flight performance of 3.4. But that's up to you.

MrLAtt, 2018-07-08:
I mostly put my backpack on gliders, or at least power assist gliders, so it's quite helpful in those cases. And you have a good point, with the quads it's only really something interesting, I should probably just move over to 3.4 for the better flight performance

mikeller, 2018-07-08:
Ah, now I understand. I read 'backpack' as in 'something a human carries on his back'. :rofl:

MrLAtt, 2018-07-08:
Ah, yes, I can see the confusion. Probably not the best way to describe it.

bforbort, 2018-07-24:
I decided to work on this issue, but I have a question before I get too far into it.

Should the vario only work with a barometer, or should it also be able to work with GPS data?

I'm worried that the slow sample rate and relatively high drift in GPS altitude would require so much averaging, that the vario would have an extremely slow response.

MrLAtt, 2018-07-24:
Yes, i agree with your thinking that GPS altitude wouldn't be suitable to use, however, if it is simple to do it don't see any problem with letting the user choose if they want to. But yes, i think it would be mostly useless🤣

MrLAtt, 2018-07-24:
As a side note - for my particular usage (for a fpv glider) as little averaging as possible is best as quick reaction is more important, and i would prefer that i do the averaging when looking at it, but then again, i don't think that would be the typical use case. 

The vario in 3.2 (what is am currently using) seems to work well enough in terms of averaging

bforbort, 2018-07-24:
ok, for now I'll make my code only work with a baro.  I have a wing I could test out vario via GPS on, or I might just buy a small GPS module for testing on one of my quads.

I looked at the GPS rescue code, and it's vertical speed calculation is very heavily averaged...probably due to the relative inaccuracy of GPS.  For now I'll also use the old Complimentary Filter to keep that delay down.

RoadyFPV, 2018-07-24:
I prefer Vario from the Baro. Gps baro is to slow and inaccurate.

MrLAtt, 2018-07-24:
I don't currently have a GPS, but it would be happy to buy one if it would help

MrLAtt, 2018-07-25:
i would love to, but i have to admit i have embarresingly low knowledge of git and how 'trees' 'release' 'pull requests' and other such things mean (considering i have just finished a computer science A-level (uk)). im guessing i download the link you have provided as a zip and then flash it as local firmware?

bforbort, 2018-07-25:
@MrLAtt Unfortunitly it's a little complicated.  It has to be compiled first.  Let me know what target you have, and I'll post the .hex file for you here.

[HOW TO IDENTIFY BETAFLIGHT FIRMWARE TARGET OF FLIGHT CONTROLLER](https://oscarliang.com/identify-betaflight-firmware-target/)

MrLAtt, 2018-07-25:
i am using an omnibus f3 (OMNIBUS target) is there anyting else i need to know about going to 3.4, considering that i am only using it as an OSD?

bforbort, 2018-07-26:
@MrLAtt It should be fairly easy as you don't have to do any tuning.  I do highly recommend you BACKUP YOUR CURRENT SETTINGS. That way you can go back if this causes issues.

1) Download the latest [Betaflight Configurator](https://github.com/betaflight/betaflight-configurator/releases) to ensure that it's compatible.
2) Use the CLI to [backup your current settings](https://www.propwashed.com/betaflight-backup-restore/).
3) Flash the attached hex file using the "Load Firmware [Local]" option.  Be sure to also select "Full Chip Erase." If you don't do the erase, sometimes old settings and interfere with the new firmware.

[betaflight_3.5.0_OMNIBUS.zip](https://github.com/betaflight/betaflight/files/2230097/betaflight_3.5.0_OMNIBUS.zip)

4) Either set your OSD back up from scratch, or use the backup file you made.  If you want the G-Force meter (It's cool), it will be called "Unknown 2" in the configurator, because it hasn't been added in yet.

MrLAtt, 2018-07-26:
Ahh, well, the g force meter was the one other thing that I was interested in for my OSD, and was the main reason I didn't want the vario feature to be left out of future bf versions (I saw that there was a request for it and that g-force was likely to be in the next bf version) . This just got a lot more cool! 

I'm not sure when I will be able to test it as I am busy most of tomorrow but I will report back as soon as possible!

MrLAtt, 2018-07-26:
Thankyou for your hard work!

MrLAtt, 2018-08-26:
I'm sorry its been so long since my last reply, i have been really busy with applying for university. I finally got to fly this morning with my FPV backpack with the bf 3.5 code you wrote, it works great! The vario seems to be just as accurate as before and the G-Force meter is really cool as well.

MR444, 2018-09-02:
With Matek F405, still not working. Vario is always zero!

I have this problem since version 3.4.0. I though it was solved with 3.4.1 but it is still not. Also not in 3.5.0 as I thought after I read this thread.

Erasing variometer and altitude holding mode? Why all that?

[image]

PR #6447 (2018-09-18): Enable Vario on targets that support barometers
Fixes #6269 

The vario OSD and telemetry will provide climb rate, no longer always indicating '0.'  Targets that do not have USE_BARO defined will have the vario elements removed from the OSD and telemetry output.

This PR does not implement vario support via GPS data. I believe the primary users of vario prefer to use barometric readings.  GPS support can be added by implementing additional averaging to the signal if desired.
closed 2018-09-18 by mikeller