betaflight/betaflight #6270: CRAZYBEEF3FR RSSI & telemetry problems on BF3.4 RC4 & RC5

GeraldoGarcia, 2018-07-02:
A weird problem started as soon as I upgraded from RC3 to RC4 and still remains on RC5: The RSSI displayed on the OSD got way lower than the RSSI on the Taranis. At first I thought it could be something related to the RSSI scale or offset, but I started to experience failsafes on short distances and started to get "telemetry lost" warnings from the Taranis.

I am aware of the problems with this board's SPI and telemetry, so I disabled telemetry (diff below), but even with telemetry disabled the board still sends RSSI and voltage info to the radio.

I will also link a video showing the rssi on the OSD and the RSSI on the radio.

And yes, I checked the antenna on the board, resoldered it and the problem remains.

# diff
# Betaflight / CRAZYBEEF3FR (CBFR) 3.4.0 Jul  2 2018 / 12:10:00 (463d7aa1c) MSP API: 1.39

board_name CRAZYBEEF3FR
manufacturer_id 

feature -MOTOR_STOP
feature -TELEMETRY
feature AIRMODE
beacon RX_LOST
beacon RX_SET
aux 0 0 0 1700 2100 0
aux 1 1 1 900 1300 0
aux 2 2 1 1300 1700 0
aux 3 13 2 1700 2100 0
aux 4 35 3 1700 2100 0
set gyro_sync_denom = 2
set min_check = 1010
set max_check = 1990
set dshot_idle_value = 490
set motor_pwm_protocol = DSHOT300
set vbat_max_cell_voltage = 45
set vbat_min_cell_voltage = 30
set vbat_warning_cell_voltage = 32
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 2
set osd_rssi_alarm = 35
set osd_vbat_pos = 2401
set osd_rssi_pos = 2423
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2433
set osd_crosshairs_pos = 2253
set osd_ah_pos = 2126
set osd_craft_name_pos = 2441
set osd_warnings_pos = 2409
set vcd_video_system = NTSC
set frsky_spi_tx_id = 49,138
set frsky_spi_offset = -11
set frsky_spi_bind_hop_data = 2,22,42,62,82,102,122,142,162,182,202,222,7,27,49,67,87,107,127,147,167,187,207,227,12,32,52,72,92,112,132,152,172,192,212,232,17,37,57,77,97,117,137,157,179,197,217,0,0,0
set frsky_x_rx_num = 4
profile 0

rateprofile 0

# 

Video: https://youtu.be/YgeNb_ozq-0

etracer65, 2018-07-02:
Have you gone back to RC3 or even BF3.3.3 and confirmed that the problem doesn't exist?  Need to eliminate your hardware as the problem first.

GeraldoGarcia, 2018-07-03:
Well... I discovered a few things:

The difference between the RSSI on the radio and on the OSD, besides the scale difference, is due to the loss of telemetry itself. When the radio accuses "telemetry lost" the RSSI freezes until it gets telemetry again, meanwhile on the OSD the RSSI keeps dropping as the distance increases. That is why the radio may be displaying something like 70 while the OSD is showing 30. So the difference by itself is not an issue. The abnormally low RSSI and the frequent loss of telemetry only 7 meters away is still an issue.

I rolled back to RC3 and even RC1 and to my surprise the problem persisted. I guess it was a coincidence that it manifested fist right after the update to RC4. I will check the hardware again, I am sure the antenna is OK, but something is wrong for sure.

Sorry to bother you with that, looks like it is not a betaflight issue.

closed 2018-07-03 by GeraldoGarcia

GeraldoGarcia, 2018-07-03:
@mikeller 

> The reason for this is that the RSSI that is reported by the TX is actually the RSSI value that is sent back by the RX, since the TX module has no way to report its own RSSI.

Yes, that is what I realized eventually. I am still puzzled by the issue, I mean the sudden reduction on RSSI and the constant loss of telemetry, but it doesn't look like a betaflight issue.

Thanks.