betaflight/betaflight #6272: CRAZYBEEF3FR Loss of telemetry and radio

JLOFPV, 2018-07-02:
Loss of telemetry and radio with fall. You have to unplug the battery and plug it in so you can link again.
With beta 3.3.3 it was already happening, but apparently it was solved in beta 3.4 rc3, I have flown more than 20 batteries without problems, today I have updated to rc5 and the bug appears in every battery I fly.
I use the same configuration in all tests.

mikeller, 2018-07-03:
By providing a `diff` like you are asked to do when opening an issue you will increase the chances that anybody is actually willing and able to help you.

JLOFPV, 2018-07-03:
sorry

mikeller, 2018-07-03:
Try disabling the telemetry feature. Telemetry is still buggy in FrSky X SPI.

Also (but unrelated), change the motor protocol to Dshot300, the ESCs on this board are not capable of reliably doing Dshot600.

JLOFPV, 2018-07-03:
But in rc3 it works perfect.

GeraldoGarcia, 2018-07-04:
Well @JLOFPV , I hate to tell you... but I tested all the RCs with telemetry on and off and sure enough they all had the "freeze" issue with telemetry on.
I just rolled back to RC3 to test another problem and had a shot with telemetry on, at the middle of the first battery it fell to the ground with the characteristic frozen OSD (not a failsafe). Turned off telemetry and the problem was gone.

mikeller, 2018-07-04:
@GeraldoGarcia: One thing about telemetry is that how much data is sent as telemetry is dependent on what features are activated in the firmware, and this also affects the amount of time required to generate the telemetry data. I.e. telemetry might work for a configuration with less features activated, but not for another configuration with more features.

JLOFPV, 2018-07-04:
Is it possible to leave the voltage and lua activated only in the telemetry?

mikeller, 2018-07-04:
@JLOFPV: The battery voltage is already sent back as A1 even when telemetry is not enabled. lua support is currently not possible without telemetry, and I don't know if that would work, considering that some MSP commands (like writing the configuration) can take a considerable amount of time, possibly causing the RX to hang.

closed 2018-08-10 by stale[bot]

mikeller, 2018-11-03:
@Heathbn: See https://github.com/betaflight/betaflight/issues/6272#issuecomment-402090901.

Heathbn, 2018-11-03:
Thanks, Mikeller
However, as My comment states. I'm looking for more options.
Do you know if anyone has solved this any other way or pinpointed the telemetry that is the problem?
I was also wondering if using D8 would fix the issue as it has less latency? 
Thanks again