betaflight/betaflight #6274: OMINIBUS F4 USE DSMX

Skyezheng1, 2018-07-03:
Hi:
 probrem：I use OMINIBUS F4, RX1 can use DSM2, but DSMX can not work.How can I get my DSMX to work on UART RX1?Thanks

etracer65, 2018-07-03:
Are you selecting `SPEKTRUM2048` for the Serial Receiver Provider?

Skyezheng1, 2018-07-04:
Yes, I choose SPEKTRUM2048, but it can't work.Can change the CLI command to make it work?

jflyper, 2018-07-04:
@Skyezheng1 
I have no trouble, Omnibus F4 AIO (gen1) flashed with v3.3.3 and v3.4.0-RC5, setup for DSMX on UART1. I suspect your receiver is not bound to use DSMX.

This is the relevant part of my config `diff` output for above stated test.
```
# version
# Betaflight / OMNIBUSF4 (OBF4) 3.3.3 May 31 2018 / 13:23:55 (3565182d3) MSP API: 1.38

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# master
set serialrx_provider = SPEK2048
```

Skyezheng1, 2018-07-04:
@jflyper 
Thank you for your answer.
# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
This is the two different type of receiver.

# feature
Enabled: RX_SERIAL

# SERIAL
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

set serialrx_provider = SPEK2048

# map
map TAER1234

This is my message. Thank you again.

jflyper, 2018-07-04:
@Skyezheng1 

IS YOUR RECEIVER BOUND IN DSMX MODE?

Skyezheng1, 2018-07-04:
@jflyper 
Yes, I try RX6, it can work.So the receiver is no problem

Skyezheng1, 2018-07-04:
@jflyper 
About UART1, are there any other settings？

jflyper, 2018-07-04:
@Skyezheng1 
Ah, no, I'm talking about the receiver itself.

closed 2018-08-10 by stale[bot]