betaflight/betaflight #6275: RC4 OmniNXT F7

mooiweertje, 2018-07-03:
1. Using the same rates as always. Now the quad turns a lot faster than normal.
2. Use my rates.

# version
# Betaflight / OMNINXT7 (ONX7) 3.4.0 Jul  2 2018 / 12:34:38 (463d7aa1c) MSP API: 1.39

board_name OMNINXT7
manufacturer_id 
mcu_id 004b00453136511136323831
signature 

# reset configuration to default settings
defaults nosave

# name
name Real1S

# resources

# mixer

# servo

# servo mix

# feature
feature -LED_STRIP

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# led
led 0 8,9::C:11

# color

# mode_color
mode_color 6 0 1
mode_color 6 1 11
mode_color 7 0 2

# aux
aux 0 0 0 1300 2100 0
aux 1 0 1 1300 1700 0
aux 2 1 0 1300 1700 0
aux 3 13 1 1700 2100 0
aux 4 26 3 1700 2100 0
aux 5 28 0 1700 2100 0
aux 6 35 1 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_calib_duration = 250
set gyro_calib_noise_limit = 96
set gyro_use_32khz = ON
set gyro_to_use = SECOND
set acc_calibration = 52,-24,-66
set mag_hardware = NONE
set rssi_channel = 16
set max_aux_channels = 12
set serialrx_provider = SBUS
set blackbox_p_ratio = 128
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set align_board_yaw = 90
set current_meter = NONE
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 180
set gyro_cal_on_first_arm = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_vbat_pos = 2498
set osd_rssi_pos = 34
set osd_tim_2_pos = 2518
set osd_vtx_channel_pos = 2081
set osd_current_pos = 39
set osd_mah_drawn_pos = 48
set osd_altitude_pos = 2293
set osd_warnings_pos = 2505
set osd_disarmed_pos = 2282
set osd_core_temp_pos = 2104
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bb_no = OFF
set vtx_band = 2
set vtx_channel = 6
set vtx_freq = 5828
set vcd_video_system = PAL

# profile
profile 0

set dterm_notch_cutoff = 0
set p_pitch = 62
set d_pitch = 23
set p_roll = 52
set i_roll = 43
set d_roll = 18
set p_yaw = 76
set i_yaw = 60
set d_yaw = 10

# profile
profile 1

set p_pitch = 60
set p_roll = 59

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 0
set thr_expo = 30
set roll_rc_rate = 200
set pitch_rc_rate = 200
set yaw_rc_rate = 200

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

etracer65, 2018-07-03:
Maybe you made a mistake on your rate setup? You have configured 1333dps on all axes.
```
set roll_rc_rate = 200
set pitch_rc_rate = 200
set yaw_rc_rate = 200
```
Nothing has changed in regards to rates.

mooiweertje, 2018-07-04:
⁣Dear Bruce,

I put 200 there years ago and I liked it from the start. Maybe this F7 board with BF3.4 is the first one that is fast enough to actually achieve this rate?

Kind regards,
 ~ Jannes Bosma

​

McGiverGim, 2018-07-04:
Check the setpoint weight. It does not change the max rate but it changes the feeling of sticks. It acts faster than with the old default of zero.

mooiweertje, 2018-07-04:
⁣Dear Miquel,

No it's really a different degrees per second. I lowered it to 1,5 and now it's more similar to what it was. It's flying fine.

I'm also using an frsky receiver now while I'm used to Devo.

Kind regards,
Jannes Bosma

​

mooiweertje, 2018-07-13:
I always thought degrees per second would be actual degrees per second. But I now start to get the idea it is just a bogus quantity.

etracer65, 2018-07-13:
No, you are incorrect.  The configured rate is in degrees/second.  If you were to record a blackbox log of a roll or flip using full stick deflection you will see that the rotation rate matches the configured rates.

What can change between different quads and different tuning options is how quickly the quad attains the configured rate.  The power of the motors, the props, the weight of the quad, the P gains and setpoint weight all effect this.  So while the overall rate is the same, if the quad attains the configured rate quicker after being commanded by stick movement you'll perceive this as "faster rates".  So as an example, lowering setpoint rate will make the quad more slowly attain the configured rate and thus maybe feel like an overall lower rate to you (but it's not in reality).

McGiverGim, 2018-07-13:
No. The rates are the real desired degrees by second. The only way to verify it is with the blackbox.

closed 2018-08-19 by stale[bot]