betaflight/betaflight #6278: smix reverse freezes configurator / doesn't work, 3.4 RC3

ledimestari, 2018-07-03:
I'm building a bicopter on BF 3.4 and Matek F722 flight controller and what I noticed is that yaw stabilized direction appears to be wrong.  By reversing servos I can get either yaw or pitch to stabilize correctly but other one is always wrong.
I thought I could solve this by following servo reversing guide in Mixer.md to reverse yaw stab direction but it doesn't work. 

Whenever I enter "smix reverse 1 2 r" which apparently should reverse servo 1 on yaw stab axis it freezes the CLI and I need to power cycle to get it back working. 

As said, 3.4 RC3 on Matek F722 board. Sbus receiver, motors wired normally and servos on pads S5 and S6. 

diff:
```
# diff

# version
# Betaflight / MATEKF722 (MKF7) 3.4.0 Jun 24 2018 / 09:56:39 (1c081ae8c) MSP API: 1.39

board_name MATEKF722
manufacturer_id 

# name

# resources
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 B01
resource SERVO 2 A08

# mixer
mixer BI

# servo
servo 0 1000 1940 1520 100 -1
servo 1 1000 1940 1520 100 -1
servo 4 1100 1940 1485 -100 -1
servo 5 1100 1940 1455 -100 -1

# servo mix

# feature
feature RSSI_ADC
feature LED_STRIP
feature AIRMODE

# beeper

# beacon

# map

# serial

# led
led 0 0,0::CT:2
led 1 1,0::CT:2
led 2 2,0::CT:2
led 3 3,0::CT:2
led 4 4,0::CT:2

# color
color 2 359,0,255

# mode_color
mode_color 7 0 8

# aux
aux 0 0 0 1300 2100 0
aux 1 1 2 1700 2100 0
aux 2 13 2 1300 2100 0
aux 3 15 5 1300 2100 0
aux 4 19 3 1300 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = -29,13,-2
set mid_rc = 1520
set max_check = 1940
set min_throttle = 1030
set motor_pwm_rate = 200
set yaw_motors_reversed = ON
set deadband = 5

# profile
profile 0

set dterm_notch_cutoff = 0
set p_pitch = 35
set i_pitch = 10
set d_pitch = 20
set p_roll = 30
set i_roll = 10
set d_roll = 20
set p_yaw = 40
set i_yaw = 10

# rateprofile
rateprofile 0

set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50
set tpa_rate = 0
```

etracer65, 2018-07-03:
@ledimestari Could you try this firmware?  It should fix the problem.  Note that this is built from 3.4RC5 instead of RC3.

[betaflight_3.4.0_MATEKF722.hex.zip](https://github.com/betaflight/betaflight/files/2161364/betaflight_3.4.0_MATEKF722.hex.zip)

ledimestari, 2018-07-04:
@etracer65 Thank you! this fixed the probelm! 

Maybe I shouldn't use pre-releases on unusual configs like bicopter, but works now.

mikeller, 2018-07-04:
@ledimestari: No, you totally should! Thanks to you doing this, a bug was found and fixed before the stable release. If that had not happened, we probably would have had to deal with a lot of sad tricopter pilots because of 3.4 brokenness...

ledimestari, 2018-07-04:
Feels good to be useful. 
Keep up your good work, I'll close this now. :)

closed 2018-07-04 by ledimestari

etracer65, 2018-07-04:
@ledimestari Thanks for testing and confirming the fix.  And while we do appreciate you testing out the new 3.4 version, in this case the bug is also in 3.3 so you would have stumbled across it either way.  Just goes to show that not many people are using servo reversing.

ledimestari, 2018-07-04:
@etracer65 I didn't get around trying 3.3 yet, that was my next option. Good that this was solved here, saved me a lot of time. 
I do believe not so many people use servo reversing but it's one of those features that you really need when you need it. It's nice that I can use betaflight and all its features also on little different builds. Keep supporting stuff :)

PR #6280 (2018-07-06): Fix null pointer reference for "smix reverse"
Fixes #6278 

The short syntax `smix reverse` is meant to print the table. When the logic was checking for parameters it was failing to deal with the null pointer when no parameters were present. Additionally the `smix reverse` was called at the end of a successful command to display the table so even though the command was succeeding it was crashing on the null pointer reference when trying to display the result.

Also some stylistic cleanup