betaflight/betaflight #6282: FF_PIKOBLX: close port error

RipperDrone, 2018-07-04:
This bug is persistent since BF 3.x: PIKOBLX cannot close port in USB serial connection - you always have to unplug USB cable, then plug in again to re-connect.

Particularly after all 'save' operations on different / all tabs.

When pressing 'disconnect' in BFC, then 'connect', everything is fine. Not so for 'save', then trying to re-connect...

MJ666, 2018-07-04:
Issue #5835 already looks to be the same problem and was closed since there was no activity anymore. I provided already details here? BTW I just tested BF 3.3.2 and BFC 10.3.1 again and my PikoBLX clone works as expected and reconnects after saving at configurator pages and after using the CLI. Also disconnect and reconnect is working as expected. Sorry but I can not see and reproduce the problem you have?

RipperDrone, 2018-07-04:
what could I do to help diagnose further? Have a genuine PikoBLX, have ried 2 different USB cables, flashed different firmwares 3.x+. Always freezes after save&reboot, does not reboot, does not close port / disconnect successfully...

RipperDrone, 2018-07-04:
my BFC version is 10.3.1 as well now...

MJ666, 2018-07-04:
just tested an genuine PikoBLX with BF 3.3.0. this always reconnects after saving and manual disconnect and reconnect. If the CLI is used it does not reconnect automatically and reconnect is not possible. Anyhow restarting BFC will allow to connect again without touching the FC. Not sure but for me the problem looks to be in the BFC. As mentioned before in #5835 i know this behavior from some other FC's also.

MJ666, 2018-07-04:
Can you connect manually after this without restart BFC? Did you try to restart BFC and leave the FC alone (no disconnect of the USB cable)?

RipperDrone, 2018-07-04:
Just tried twice: CLI - save --> shows 'rebooting' but doesn't

Tried to press connect several times --> COM x not detected, no reconnect possible
Restarted BFC --> doesn't show COM X, still no connect possible
unplugged/replugged USB cable --> shows COM x again, connect ok

MJ666, 2018-07-04:
Mhhh, this is not something i can see/reproduce here? The COM port always comes back after the reboot. Do you have the latest STM VCP drivers installed?

RipperDrone, 2018-07-04:
think so, will try to re-install...

RipperDrone, 2018-07-04:
v1.5.0 reinstalled, after uninstalling existing driver. Still same result: Doesn't show COM x after reboot attempt. HAve to physically unplug/plug in USB cable :-/

MJ666, 2018-07-04:
With save from cli the genuine PikoBLX even reconnects. It only don't connect if move to an different page from CLI in BFC.

RipperDrone, 2018-07-04:
here, both ways it does not reconnect. Got a genuine PikoBLX as well... seems that BFC and FC timings don't work together well...?

RipperDrone, 2018-07-04:
yeah, two of them. Ok, then I'll close this issue even though unresolved. Thanks anyway... :-)

closed 2018-07-04 by RipperDrone