betaflight/betaflight #6284: Betaflight F4 & Frsky receivers

wdw222, 2018-07-05:
1. Betaflight F4 and Frsky receivers producing low RSSI values;
2. Tested both R-xsr and Xsr receivers on the two separate betaflight F4 flight controllers. In each instance the RSSI values were low, looks like 50% of the normal value. Tested the same receivers on a hobbywing xrotor flight controller and the RSSI values were fine;

3. Betaflight F4, Hobbywing FC . Receivers uses SBUS protocol, Taranis QX7;

4. Betaflight F4(BETAFLIGHTF4) = 3.4.0-RC5, Hobbywing FC(OMNIBUSF4SD) = 3.4.0-RC5

```
# Betaflight / BETAFLIGHTF4 (BFF4) 3.4.0 Jul  2 2018 / 12:07:00 (463d7aa1c) MSP API: 1.39

board_name BETAFLIGHTF4
manufacturer_id 

# name
name Yeti_2

# resources

# mixer

# servo

# servo mix

# feature
feature AIRMODE

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 0 115200 57600 0 115200
serial 2 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 13 2 1700 2100 0
aux 2 35 3 1700 2100 0

# adjrange
adjrange 0 0 3 900 1075 12 3 0 0
adjrange 1 1 3 1425 1600 12 3 0 0

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 120
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 2000
set rssi_channel = 16
set rssi_scale = 41
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1300
set vbat_max_cell_voltage = 44
set vbat_min_cell_voltage = 34
set vbat_warning_cell_voltage = 36
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 111
set ibata_scale = 267
set beeper_dshot_beacon_tone = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 35
set osd_cap_alarm = 1300
set osd_vbat_pos = 2433
set osd_rssi_pos = 2061
set osd_tim_2_pos = 2455
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 234
set osd_ah_pos = 234
set osd_mah_drawn_pos = 2444
set osd_warnings_pos = 329
set cpu_overclock = 192MHZ
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 2
set vtx_freq = 5645
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass_hz = 80
set dterm_lowpass2_hz = 160
set dterm_notch_cutoff = 0
set anti_gravity_gain = 1000
set dterm_setpoint_weight = 120
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set iterm_windup = 50
set throttle_boost = 2
set p_pitch = 32
set i_pitch = 65
set d_pitch = 23
set p_roll = 30
set i_roll = 60
set d_roll = 21
set i_yaw = 60
set d_yaw = 20

# rateprofile
rateprofile 0

set yaw_rc_rate = 102
set roll_srate = 34
set pitch_srate = 33
set yaw_srate = 34
set tpa_rate = 15
set tpa_breakpoint = 1050
 ```

grizewald, 2018-07-05:
it looks to me like you restored a dump from 3.3.
_set rssi_scale = 41_
Set rssi_scale to 100 instead.

wdw222, 2018-07-05:
O wow! Thanks for the relpy!
Will definitely give that a go.
And you are right.... I did restore  a dump from 3.3

Thanks
Wynand

grizewald, 2018-07-05:
Restoring a dump from a different release is a bad idea. Quite often, the names, ranges and function of CLI settings change between releases and restoring a full dump can lead to strange things happening.

A better idea is to save a 'diff' instead and then trim the diff down to the bare essentials like your receiver setup and any custom board or remapping changes. There are a *lot* of things that have changed between 3.3 and 3.4 and the best way is to just set things up from scratch again. Then you can be sure you have not unintentionally set things up wrong.

mikeller, 2018-07-07:
@wdw222: If this is fixed for you, can you please close?

closed 2018-07-07 by wdw222