betaflight/betaflight #6285: BF3.4 issues

raymonkroeze, 2018-07-05:
Hello, I've tried Betaflight 3.4 on multiple setups and all setups are a bit different. I have some serious issues with BF3.4. With the default pids all the quad flies like shit. very loose and it can't hold it's attitude in fast movements. So it needs tuning (not like previous versions). So last weekend I've tuned the quad to almost perfection. flew perfect, no propwash. It sounded just very clean and smooth. Yesterday I flew that exact setup and it flew like shit again. Nothing has happend to the quad since last weekend. My other setups I won't even talk about because I could not get it to fly good. I fly with new gear props etc. Just some feedback guys. don't need answers because I'm not gonna fly with BF3.4

raymonkroeze, 2018-07-05:
yes sorry. I should have provided more info when posting this message. My setup:
Prototype Aikon Freestyle frame 230 size and topmounted lipo.
Emax RSII 2206 2300kv motors
Aikon Electronics SEMF esc's Dshot600
Aikon flightcontroller (If tried running 32/8khz, but too much nasty vibrations so switched to 8/8khz)
GEPRC 5*4*3 props and I've also tried the HQ51*51*3 props
I don't have the maiden footage any more. But basically the default Pgains are too low to keep the quad stable. I wants to dip a lot and it has this loose feeling like the Pgain is too low. I've spend a whole morning tuning the quad. Don't remember the PID's any more exactly. but something like this:
ROLL: 52 50 29
PITCH: 64 63 33
YAW: 85 45
What i've noticed, the PIDs in the BF3.4 seems a lot more effective than the previous version. a value of 2 on the Pgain seems to do a lot more than in 3.3 for example. But last sunday I was tuning with a little bit of wind and I got it to fly really nice and I really love the feeling of 3.4. I got rid of pretty much all of the propwash. Yesterday I was flying again and I had this nasty motor noice and really nervous behaviour. And it wasn't like high Pgain oscilation but more of a rawisch vibration sound. Like I was flying with 32khz that day. I've read on your page 32khz might need more filtering so that's way I didn't run 32khz. With BF3.3 I'm running 32/8khz with no issues at all.
I've got a video of yesterday, but you really can't see anything weird on this. Maybe the sound of the motors but that isn't really that obvious on the gopro footage. If you want I can upload a vid for you.

watageek, 2018-07-05:
I'm not convinced; BF3.4 IS awesome. Maybe the culprit on your setup is the noise polluting the gyro. leave all the PID's and setup at default at first (and/or follow the tuning guide wiki for 3.4 )[https://github.com/betaflight/betaflight/wiki/Tuning-Tips-for-Betaflight-3.4](url)

On my sloppy/beat-up daily driver quad, I have worn out bearing and bent/nicked prop ( HQ 4x4.3x3 V1S). Just a thought, I share my lowpass Hz with you

set dterm_lowpass_hz = 54 ( or 70)
set dterm_lowpass2_hz = 140
set gyro_lowpass_hz = 90
set gyro_lowpass2_hz = 180

raymonkroeze, 2018-07-05:
Maybe you and I have different standards if it comes to how a quad flies. It's not one setup. I've tried 2 setups and some buddies of me also did try 3.4 with same result.
I have one of the most cleanest running quads mate. I always can turn off all of the notch filters be for with no issues. And if I need to mess around with different settings to get the quad flying a bit good I might as well fly with kiss.

ProcessDestroyer, 2018-07-05:
What version of Betaflight Configurator are you running?  There's a bug in 10.3.0 that disables gyro LPF.  10.3.1 fixes.

fixmania, 2018-07-20:
I agree that. Many people with me have experienced vibrations like jello. Even though defaults setup, motors are hotter than before 3.3.3 version. It just shows on 1080p above screen not goggles.

gmdfpvpilot, 2018-08-05:
I was trying out the BF 3.5 RC1 and it happened to me the same after tuning and recording the changes and when I switched to another equivalent battery the behavior was all erratic. The motors always have high temperatures.

Dropsleutel, 2018-08-09:
I also have feeling that something not yet optimal motors run hotter for me than before. 3.5 rc2 configurator 10.3.1, tried on few setups.   Although quad behaves flies pretty well at defaults.  loweredthe filter by 30% not much of difference in temps….

closed 2018-09-21 by stale[bot]