betaflight/betaflight #6290: D term going crazy when rc_smoothing = FILTER

antoni9, 2018-07-05:
1. Describe your problem;
Smoked a motor

2. Include ways to reproduce the problem;
Issue present in both RC2 and RC5
When rc_smoothing = FILTER D is going wild, steps in rc_command are only "smoothed", peaks an d troughs still clearly audible when flying, see logs - at each trough P is rising abruptly, preceded by a D kick.

I was doing hover tests at home because I could hear something that sounded like propwash and clicky sounds in the field.

Log where motor dies:
[BTFL_BLACKBOX_LOG_ANTONIG_20180705_225451.txt](https://github.com/betaflight/betaflight/files/2168466/BTFL_BLACKBOX_LOG_ANTONIG_20180705_225451.txt)

Log with = INTERPOLATION
[BTFL_BLACKBOX_LOG_FLEA_20180705_233119.txt](https://github.com/betaflight/betaflight/files/2168467/BTFL_BLACKBOX_LOG_FLEA_20180705_233119.txt)

4. Add the used configuration and firmware version;
3.4 RC2 DYS F4 Pro V2, Tekko 4in1, AokFly 2306

fujin, 2018-07-05:
Please post your full diff as well as the output of `rc_smoothing_info` CLI command with radio powered on / RC link established.

antoni9, 2018-07-05:
Wrong log, this is the one with = INTERPOLATION
[BTFL_BLACKBOX_LOG_ANTONIG_20180706_002820.txt](https://github.com/betaflight/betaflight/files/2168516/BTFL_BLACKBOX_LOG_ANTONIG_20180706_002820.txt)

[diff.txt](https://github.com/betaflight/betaflight/files/2168518/diff.txt)

# rc_smoothing_info
# RC Smoothing Type: INTERPOLATION

fujin, 2018-07-05:
Your issue report is about `FILTER` type, not `INTERPOLATION`. Please provide log for `FILTER` type flight as well as `rc_smoothing_info` while in FILTER mode if that is where the problem lies.

It would also be beneficial for you to use `debug_mode = RC_SMOOTHING` while logging.
```
debug(0) = raw un-smoothed rc channel data
debug(1) = raw un-smoothed setpoint derivative
debug(2) = filtered setpoint derivative before applied to setpoint weight
debug(3) = auto-calculated filter cutoff frequency base after sampling the rx frame rate
```

etracer65, 2018-07-06:
I'm not sure exactly what you're trying to report.  Your diff shows you using interpolation.  I also noticed that you're running RC2 and we're currently at RC5 with many fixes in between.  Both logs have normal looking D-terms and the motor traces look smooth.

As noted above, please provide the output of `rc_smoothing_info` with your receiver powered and your TX on.

antoni9, 2018-07-06:
I already smoked a motor in a 17sec flight, not doing that again.. afraid
it's the only logs I can provide. The only change made was to rc_smoothing
everything else=same

Been flying much more agressive today, no issues.

watageek, 2018-07-06:
It has been state in other post. There is a bug with the 10.3.0 configurator: it set gyro_lpf =0.

You can smoke a motor for no apparent reason. It happens... It might be unrelated. It would be a good time to start winding coils. Look for my thread and JackErbes, ManuelV, Leonard3K, SkyLar, Etc on RC group (and facebook) for help.

fujin, 2018-07-06:
@antoni9 You do not need to fly to give us the information required to identify a problem with FILTER-type mode on your setup. Just bring up radio RC link, set type to `FILTER`, then `rc_smoothing_info` in CLI. Bang the sticks etc a couple of times, make sure it looks OK on the Radio tab.

Debug logging would be beneficial to investigate further. Could you also please describe what radio protocol/provider you are using? e.g. TBS CSRF, FrSky X/D16 8ms, etc?

If you are unwilling to provide diagnostic information, without a way to reproduce, we're stuck.

antoni9, 2018-07-07:
I can do that next week

antoni9, 2018-07-07:
# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 7.748ms
# Input filter type: BIQUAD
# Active input cutoff: 58hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 58hz (auto)

FlySky i6X with M9 gimbals

watageek, 2018-07-07:
>
The calculation provided in this video gives 71.4Hz, why are the cutoffs not set to that value?
https://www.youtube.com/watch?v=M50fKpvFjT8&t=4s

the calculation is correct. You are missing a step in the calculation

1/(7.748/1000)=125Hz
125hz/2*.9=56.25Hz

etracer65, 2018-07-07:
Everything looks good with RC smoothing and the detected RX rate.  The cutuffs were calculated properly based on the RX frame rate.  I don't see any problems.

antoni9, 2018-07-12:
Debug doesn't work, I get the exact same traces from debug(0123) as I get
from rc_command

I flew with fliter and the ladder is clearly audible and visible in the
goggles, is this normal? Changed the cutoff to 45 and it seems a bit less
audible, but still present. Am I correct in assuming that the higher the
cutoff hz correspond to the faster protocols? Tried it again on auto and 1
motor came down too hot to touch after 30sec. Others were warm.

etracer65, 2018-07-12:
Where is the log?  Also please provide a `diff` of the settings used.  The information you posted previously is ambiguous as you seemed to provide a log and diff with interpolation enabled.

Also you made some mistake on the debug mode.  It's not possible for the debug mode to return the rcCommand values in the debug values.

Be sure to use `set debug_mode = RC_SMOOTHING`.

There is clearly something wrong with your build.  RC smoothing really can't make your motors hot particularly while you're hovering around.  Also there's no possible way that only one motor would be affected.  Is the hot motor the same one you had a problem with previously?  It's much more likely to be an ESC problem (hardware/firmware/settings) then anything else.

watageek, 2018-07-12:
Under all reserve, There is inconsistency in the info you are giving

I think you are misleading yourself... 

Revert to interpolation instead of Filter ( everything default too), if you think this is the culprit. Hot motor can be caused by having a physical problem (bearing/props/shaft/short/weaken magnet/fake motor) and/or gyro and D lpf filter set too high. Have you tried to lower D to 70hz (50hz) 
A beat up quad on a sloppy frame that often is flown with bent props:
(3.4 tuning tips)
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 90
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 180
set dterm_lowpass_hz = 70
set dterm_lowpass2_hz = 140

Again you are making us blind folded throwing dart in the dark without any data in relation to the situation you are having. 

I have the feeling your problem relies more on the motor/esc/noise or something you have overlooked by overconfidence/assumption more than the firmware.

watageek, 2018-07-12:
sorry etracer ( kind of a double post). You are too swift... ;)

watageek, 2018-07-12:
He seems being plague with motor issues since october
[image]

antoni9, 2018-07-12:
@watageek, that log was added here by mistake, it's a different craft. 

Sorry, forgot to attach the log as I was in the park. I went back to interpolation due to the hot motors and stuttering while flying, I was pretty aggressive with the same props on all of the below logs, everything else on the craft and config is the same, except for rc_smoothing.

Filter does feel faster on the sticks but it's too twitchy, perhaps gyro setpoint needs to come down?

[BTFL_BLACKBOX_LOG_ANTONIG_20180712_184414_with_debug.txt](https://github.com/betaflight/betaflight/files/2189924/BTFL_BLACKBOX_LOG_ANTONIG_20180712_184414_with_debug.txt)

[BTFL_BLACKBOX_LOG_ANTONIG_20180712_191858toohottotouch.txt](https://github.com/betaflight/betaflight/files/2189610/BTFL_BLACKBOX_LOG_ANTONIG_20180712_191858toohottotouch.txt)

[BTFL_BLACKBOX_LOG_ANTONIG_20180712_214431_INTERPOLATION.txt](https://github.com/betaflight/betaflight/files/2189920/BTFL_BLACKBOX_LOG_ANTONIG_20180712_214431_INTERPOLATION.txt)

[diff.txt](https://github.com/betaflight/betaflight/files/2189936/diff.txt)

etracer65, 2018-07-12:
Well none of these logs have the requested debug_mode set so I can't see too much.  However by examining the nature of the rcCommand traces it looks like your radio is not updating the samples as often as it's sending frames.  In fact the actual interval between the changed data seems to be jumping around.  But it's extremely difficult to tell without the correct debug mode.  These strange rc channel update intervals are also appearing under the interpolation logs.

Just try a quick test with:
```
set debug_mode = RC_SMOOTHING
set rc_smoothing_type = FILTER
```
**Take the props off** and just do a quick test recording a log where you arm and yank the roll stick back and forth about 5 times.

Then change the debug mode to:
```
set debug_mode = RC_SMOOTHING_RATE
```
And repeat the test.  Again, be sure to set the debug modes as specified and put the activity on the roll axis (so the logging will pick it up).

watageek, 2018-07-12:
It's a log from nearly two weeks old
[image]

you have a bad motor, most evident needs to be adressed first.

watageek, 2018-07-12:
[image]

antoni9, 2018-07-12:
@watageek, it's from today, I assure you

antoni9, 2018-07-12:
[debug_logs.txt](https://github.com/betaflight/betaflight/files/2190125/debug_logs.txt)

etracer65, 2018-07-12:
I'm still not seeing the required debugging information because even if you set the debug_mode correctly (which I can't tell) you left `rc_smoothing_type = INTERPOLATION`.  As stated above it needs to be set to `FILTER`.

antoni9, 2018-07-12:
[rc_smoothing_rate.txt](https://github.com/betaflight/betaflight/files/2190185/rc_smoothing_rate.txt)
[rc_smoothing.txt](https://github.com/betaflight/betaflight/files/2190186/rc_smoothing.txt)

etracer65, 2018-07-12:
You have some kind of problem with your tx/rx.  I'm not familiar with Flysky so I can't tell you if it's a configuration item you can change or faulty hardware.

Your rx is sending IBUS frames consistently every 7.75ms which is correct for IBUS I believe.  This is the detected frame rate.  However it looks like your radio is only sending updated channel data somewhat randomly.  Most of the time it looks like every third frame, but it jumps around and sometimes it's ~7.75ms (1 frame), sometimes ~15ms (2 frames) and sometime ~23ms (3 frames).  See the screenshots which show the raw data received by the flight controller.  There should be consistent steps every rx frame or 7.75ms.

[image]

[image]

antoni9, 2018-07-12:
Do you think it's better to use FILTER with the frequency in Hz matching 23ms, or use INTERPOLATION?
Thanks for the debugging.

etracer65, 2018-07-12:
I'm not sure either are very good.  You should really try and figure out what's going on with your radio.  23ms is quite slow.

Every time the rate changes interpolation will generate a big dterm spike becasue you're running a high setpoint weight.

RC smoothing will handle the changes better but you'll have to manually set the cutoffs.  In your case you would want to use 19hz for both filters.

```
set rc_smoothing_type = FILTER
set rc_smoothing_input_hz = 19
set rc_smoothing_input_type = BIQUAD
set rc_smoothing_derivative_hz = 19
set rc_smoothing_derivative_type = BIQUAD
```

But frankly neither interpolation or filter methods are going to be very good at dealing with the constantly varying channel input rates.

antoni9, 2018-07-13:
It looks like this is an issue with the protocol not the hardware. Tried with a different Rx and the result is the same.

Rx1: https://www.banggood.com/Flit10-2_4G-10CH-Micro-Telemetry-Flysky-Compatible-Ibus-Receiver-for-FS-I6X-FS-i6S-Turnigy-Evolution-p-1260238.html?rmmds=myorder&cur_warehouse=CN

Rx2: https://www.banggood.com/Flysky-X6B-2_4G-6CH-i-BUS-PPM-PWM-Receiver-for-AFHDS-i6s-i6-i6x-Transmitter-p-1101513.html?rmmds=myorder

[image]

[flea.txt](https://github.com/betaflight/betaflight/files/2192573/flea.txt)

atomiclama, 2018-07-13:
@antoni9 in a previous post you say your radio is: -

> FlySky i6X with M9 gimbals

I take it you have modified your TX. Could there be a problem there?

I have the same TX but different RX, can't say I have noticed any problems with the RX data update rate.
Will get some logs when I get a chance.

etracer65, 2018-07-13:
More likely that it's some problem with the over-the-air protocol between your TX and RX or some configuration/binding problem in the TX or RX.

atomiclama, 2018-07-13:
Yeah damaged antennas or components, doubt it's a config/bind problem as that side either works or it don't, it's a basic TX/RX.

atomiclama, 2018-07-13:
This is with a Flysky i6 TX and ia6B RX running on bench.
Hope it can help someone. Looks ok to me.

[LOG00078.zip](https://github.com/betaflight/betaflight/files/2193106/LOG00078.zip)

```
# rc_smoothing_info
# RC Smoothing Type: FILTER
# Detected RX frame rate: 7.696ms
# Input filter type: BIQUAD
# Active input cutoff: 58hz (auto)
# Derivative filter type: BIQUAD
# Active derivative cutoff: 58hz (auto)
```
Running RC5 
```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.4.0 Jul  2 2018 / 12:33:04 (463d7aa1c) MSP API: 1.39

board_name OMNIBUSF4SD
manufacturer_id

# name
name tikka

# resources

# mixer
mixer CUSTOM
mmix 0  1.000  0.667 -0.859  1.000
mmix 1  1.000 -1.000  0.065  1.000
mmix 2  1.000 -0.474  0.793 -1.000
mmix 3  1.000  0.474  0.793  1.000
mmix 4  1.000  1.000  0.065 -1.000
mmix 5  1.000 -0.667 -0.859 -1.000

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature AIRMODE

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1625 2100 0
aux 1 1 4 1300 2100 0
aux 2 13 0 1750 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 140
set acc_calibration = 8,-117,311
set mag_hardware = NONE
set min_check = 1020
set rssi_channel = 14
set rc_smoothing_type = FILTER
set serialrx_provider = IBUS
set blackbox_p_ratio = 64
set blackbox_record_acc = OFF
set min_throttle = 1030
set max_throttle = 1960
set motor_pwm_protocol = ONESHOT42
set failsafe_delay = 5
set align_board_pitch = 180
set align_board_yaw = 180
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set current_meter = NONE
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_rescue_ground_speed = 2500
set gps_rescue_throttle_p = 120
set deadband = 4
set yaw_deadband = 6
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_alt_alarm = 150
set osd_tim1 = 2561
set osd_vbat_pos = 2124
set osd_rssi_pos = 2132
set osd_tim_1_pos = 2402
set osd_gps_speed_pos = 2410
set osd_gps_lon_pos = 2448
set osd_gps_lat_pos = 2434
set osd_gps_sats_pos = 2118
set osd_home_dir_pos = 2382
set osd_home_dist_pos = 2415
set osd_altitude_pos = 2422
set osd_pidrate_profile_pos = 89
set osd_warnings_pos = 2089
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set debug_mode = RC_SMOOTHING
set vtx_band = 0
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5790
set vcd_video_system = NTSC

# profile
profile 2

set dterm_lowpass_hz = 120
set dterm_lowpass2_hz = 0
set dterm_notch_cutoff = 0
set dterm_setpoint_weight = 75
set iterm_relax = RPY
set p_pitch = 61
set p_roll = 54
set i_roll = 47
set p_yaw = 76
set i_yaw = 55

# rateprofile
rateprofile 0

set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set tpa_rate = 20
set tpa_breakpoint = 1050
```
Looks like I'm getting consistent updates 

[image]

etracer65, 2018-07-13:
@atomiclama Yes, that looks completely normal.  The channel update rates are consistent and match the rx frame rate to the flight controller.

antoni9, 2018-07-13:
Tried with a different Tx [i6] and traces looks consistent.

Tried with different antennas but it didn't get fixed, I suspect the gimbals are not playing well with the FlySky internals :( any suggestions?

closed 2018-07-13 by antoni9