betaflight/betaflight #6295: Spektrum Serial RX response intermittent on uart r2 and uart r3

jpopillion, 2018-07-06:
I'm running SP Racing F7 dual on BF 3.4 rc5 with a Spektrum Serial RX.  I bind the radio with the rx and test everything runs good then plug it all back in and sometimes it stays bound and not responsive in BF configurator and other times I have to rebind it which only works for a little while then unresponsive again.

1. I connected it to uart r2 and experienced intermittent connections sometimes resulting in my having to re-bind.  
2. I switched to uart r3 and had the same problem. 
3. I then turned off "spektrum_sat_bind_autoreset=OFF", spektrum_sat_bin has always been set to 0 and has never had another value.

setup as:

Entering CLI Mode, type 'exit' to return, or 'help'

# get serial
serialrx_provider = SPEK2048
Allowed values: SPEK1024, SPEK2048, SBUS, SUMD, SUMH, XB-B, XB-B-RJ01, IBUS, JETIEXBUS, CRSF, SRXL, CUSTOM, FPORT
Default value: SBUS

serialrx_inverted = OFF
Allowed values: OFF, ON

serialrx_halfduplex = OFF
Allowed values: OFF, ON

serial_update_rate_hz = 100
Allowed range: 100 - 2000

# get bind
spektrum_sat_bind = 0
Allowed range: 0 - 10

spektrum_sat_bind_autoreset = OFF
Allowed values: OFF, ON

Turned on the Radio
Plugged into Betaflight Configurator 10.3.1
Plugged in the battery and checked the receiver tab everything is working.  Armed on bench (no props) several times.  
Unplug repeat steps above bind light on Serial RX is bound but Receiver tab is showing no activity and I have "Arming disable flags: RXLOSS CLI MSP" in the status of CLI.

# status
System Uptime: 48 seconds
Voltage: 161 * 0.1V (4S battery - OK)
CPU Clock=216MHz, Vref=3.34V, Core temp=52degC, GYRO=ICM20602
SD card: Manufacturer 0x3, 31166976kB, 02/2018, v8.0, 'SL32G'
Filesystem: Ready
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 0, config size: 2528, max available config: 16384
CPU:38%, cycle time: 33, GYRO rate: 30303, RX rate: 33, System rate: 9
Arming disable flags: RXLOSS CLI MSP

flint723, 2018-07-09:
Still an issue with 3.4 official?  Also what board, and what serial RX exactly? (send product link)

jpopillion, 2018-07-15:
This is not an issue anymore with the official release.

Thank you,

Jason Popillion
Chief Technology Officer

Desk:  515-288-5853
Cell:    515 291-0782
Fax:    515-243-4537
jpopillion@gcommerceinc.com<mailto:jpopillion@gcommerceinc.com>

GCommerce, Inc.
215 10th  St.
Suite 1220
Des Moines, IA 50309

Visit GCommerceinc.com<http://www.gcommerceinc.com/> for more information regarding Match3<http://www.gcommerceinc.com/ProductsServices/Match3.aspx> and our Virtual Inventory Cloud<http://www.gcommerceinc.com/ProductsServices/VirtualInventoryCloud.aspx>!
[Description: Description: Description: Description: cid:image003.png@01CBC90A.E1557080][PBEPRO]

closed 2018-07-15 by mikeller