betaflight/betaflight #6296: 3.4 RC5 Turtle Mode Motors spinning backwards

wrevenaugh, 2018-07-07:
I have two quads on 3.4 RC5 that have the same problem.  One is running with clean air mod and the other isn't.  The problem is in turtle mode, the blades on the free motors spin backwards, "sucking into the ground" rather than blowing towards it to flip the quad. One is an F3 FC and one is an F4.

On the f4:  (which is also running with the clean air mod):
 
``` 
# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.4.0 Jul  2 2018 / 12:33:04 (463d7aa1c) MSP API: 1.39
board_name OMNIBUSF4SD
manufacturer_id 
# name
name DarkMax
# resources
# mixer
# servo
# servo mix
# feature
feature -RX_PARALLEL_PWM
feature -ANTI_GRAVITY
feature RX_SERIAL
feature MOTOR_STOP
feature LED_STRIP
# beeper
# beacon
# map
# serial
serial 5 64 115200 57600 0 115200
# led
led 0 0,0::CT:2
led 1 1,0::CT:2
led 2 2,0::CT:2
led 3 3,0::CT:2
# color
# mode_color
# aux
aux 0 0 0 1750 2100 0
aux 1 1 1 900 1300 0
aux 2 2 1 1300 1700 0
aux 3 4 1 1300 2100 0
aux 4 13 2 900 1250 0
aux 5 28 1 1300 2100 0
aux 6 30 1 1650 2100 0
aux 7 35 2 1300 1700 0
# adjrange
# rxrange
# vtx
# rxfail
# master
set mag_hardware = NONE
set baro_hardware = NONE
set baro_noise_lpf = 0
set baro_cf_vel = 0
set baro_cf_alt = 0
set min_check = 1000
set max_check = 2000
set rssi_scale = 30
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 30
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set yaw_motors_reversed = ON
set servo_lowpass_hz = 400
set small_angle = 180
set deadband = 16
set yaw_deadband = 12
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2418
set osd_rssi_pos = 22
set osd_tim_2_pos = 2305
set osd_flymode_pos = 2329
set osd_throttle_pos = 129
set osd_vtx_channel_pos = 200
set osd_crosshairs_pos = 2253
set osd_current_pos = 97
set osd_mah_drawn_pos = 111
set osd_craft_name_pos = 2091
set osd_gps_speed_pos = 66
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 386
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 161
set osd_pid_roll_pos = 322
set osd_pid_pitch_pos = 354
set osd_pid_yaw_pos = 386
set osd_debug_pos = 1
set osd_power_pos = 47
set osd_pidrate_profile_pos = 418
set osd_warnings_pos = 2281
set osd_avg_cell_voltage_pos = 2411
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set vcd_video_system = NTSC

# profile
profile 0

set dterm_notch_cutoff = 0
set iterm_relax = RP

# rateprofile
rateprofile 0

set thr_expo = 62
set yaw_rc_rate = 135
set roll_expo = 50
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 43
```
BTW, what are the three ticks for???  
THANK YOU SO MUCH!!!!!
Bill

mikeller, 2018-07-07:
@wrevenaugh: What ESCs are you using, and what firmware version is running on them?

wrevenaugh, 2018-07-07:
30A 4-in-1 BLHeli_S, and BLHeli software version 16.6, dshot 600

mikeller, 2018-07-08:
@wrevenaugh: You need 16.7 for motor reversing by Dshot command to work.

wrevenaugh, 2018-07-08:
THANK YOU SO MUCH!!!!  Works perfect now - confirmed.  BLESS YOU MAN!

mikeller, 2018-07-09:
@wrevenaugh: If this has fixed it for you, can you please close this?

closed 2018-08-15 by stale[bot]