betaflight/betaflight #6298: Amperage Meter Error in Matek F405 CTR

brunoleaomaia, 2018-07-07:
I've Flashed my Matek F405 CTR with RC5. The Amperage is always incorrect. disarmed is measuring 65A with default scale for this FC (179). I tried adjust with a wattmeter but the max value was reached without correct result.

mikeller, 2018-07-07:
@mateksys: You might be able to help here.

hokasan, 2018-07-08:
Im using Matek F405 CTR with RC5 and my amperage is 0.41A when disarmed. Seems reasonably correct when hovering. Just fyi.

MATEKSYS, 2018-07-08:
Hard to help without having wiring and board photos.

brunoleaomaia, 2018-07-08:
@MATEKSYS I wired this by the manual, It's a brand new board. I'm using Matek VTX HV too.

The picture of wiring https://drive.google.com/file/d/1FA-IHzZ8_esUPd__xRwA7h2D-AOmGPW7/view?usp=sharing

brunoleaomaia, 2018-07-08:
@MATEKSYS I've rollback to 3.3.3 and same issue, is it a problem in current sensor?

MATEKSYS, 2018-07-08:
Looks like there is dry joint on current sensor chip. check the pins
[image]

brunoleaomaia, 2018-07-12:
[image]
Sorry for the delay, I just got to disassemble the quad yesterday. Can you see anything wrong @MATEKSYS ?

brunoleaomaia, 2018-07-16:
Solved! I made two things, I don't know what fixed it. 

1) I cleaned the FC removing all soldering flux. 

2) The GND of some one of boards (split or matek vtx) was connect to the frame by the iron screw (I had GND signal in the standoffs). I changed it to Nylon screws. 

Thanks All

closed 2018-07-16 by brunoleaomaia