betaflight/betaflight #6301: after updating betaflight....no connection anymore!

Jorrit-TheOneAndOnly, 2018-07-07:
After updating my betaflight my F3 EVO brushed doesnt connect anymore. The STM32 bootloader also doesn't pop up in zadig anymore. I experience this problem on my windows 7 laptop (the updated one) and after struggling for a while I also tried to install everything on my windows 10 laptop. Unfortunately no succes neither.

My guess is this has something to do with the new release from 14 days ago. This afternoon (when I ignored the new release note) everything still worked fine, I plugged it in and everything worked as usual. Made some modification on my (self-made) drone and needed to recalibrate therefor. I thought the update wouldnt harm, followed the instructions, and now wish I didnt.  After updating betaflight doesnt see my COM port anymore and therefor doesnt connect. To make it really strange, STM32 bootloader also disappeared from Zadig. 

How can I rewind this horrible update? Or what can I do to simply connect again?

mikeller, 2018-07-08:
@Jorrit-TheOneAndOnly: GitHub is for questions and discussion around the **development** of the firmware and tools. For questions around the configuration and use of Betaflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?query=betaflight%20cleanflight%20configuration%20use).

closed 2018-07-08 by mikeller

Jorrit-TheOneAndOnly, 2018-07-08:
Well, the old methods worked fine and I could indeed find everything I need to configure and use betaflight at several places. But now something has changed and it doesnt work anymore the way it used to work. There is some new issue that needs to be solved now and that issue isnt addressed yet in those groups...

mikeller, 2018-07-08:
Then go and ask in these groups.