betaflight/betaflight #6306: Kakute F7 problem

MartinFPV, 2018-07-08:
With my new F7 Fakute with Bf 3.4 final @ Dshot 1200 run only engine 1 and 2 on. 
Motors 3 and 4 do not start. When controlled by the Betaflight configurator 10.3.1. 
Only after switching to the Mulitschot protocol will all 4 motors start up.
The copter can also not be armed.

At BF 3.3.2, all 4 engines run immediately on Dshot1200

Setup:
Kakute F7@BF 3.4final
Typhoon 32 V2 @ Dshot 1200 BlHeli 32.4
Brotherhobby R3 1660kv

https://youtu.be/sTwiZYLETAQ

# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 3.4.0 Jul  8 2018 / 04:16:15 (cce8a59be) MSP API: 1.39

board_name KAKUTEF7
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set motor_pwm_protocol = DSHOT1200

# profile
profile 0

# rateprofile
rateprofile 0

jflyper, 2018-07-08:
@MartinFPV 
Can you do CLI `resource list` and `dma`?

Also, it would be great if you can select and mark them using the "<> (insert code)" tool icon?

mikeller, 2018-07-08:
@MartinFPV: In addition to @jflyper's requests, can you try `set dshot_burst = true`, `save`, and then check if it's working?

MartinFPV, 2018-07-08:
@jflyper sure

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: USB_DETECT
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: MPU_EXTI
E02: SPI_SCK 4
E03: FREE
E04: MPU_CS
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: CAMERA_CONTROL
E14: FREE
E15: FREE

Use: 'resource' to see how to change resources.

# dma list

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 4
DMA2 Stream 3: MOTOR 3
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

MartinFPV, 2018-07-08:
@mikeller 

After set dshot_burst = on, save   in CLI

No change Motor 3 and 4 still not run.

jflyper, 2018-07-08:
@MartinFPV Resources and DMA look fine... Did you swap working and non-working motor positions at ESC to identify which side of the connection has the problem?

agbagb, 2018-07-12:
I have matek f722-std, and have similar trouble.
BF 3.4.0, default config, just turn on DSHOT600 and 32k\32k and all motors stop working.
Multishot and 32k\32k all motors work.
DSHOT600 and 8k\8k all motors work.

davidlittlefarmer, 2018-07-15:
I have same issue, 
BF 3.4.0, default config. Kakute F7 with Tekko32
Multishot is working, but even DSHOT600 and 8k/8k doesn't work for me.

TheRealOrange, 2018-07-15:
I also have the same issue, Kakute F7. I tried re flashing the FC and reflashing the ESCs but it didn't help. All the DShot options don't work, but Multishot does work. Any help would be greatly appreciated

DieHertz, 2018-07-16:
@MartinFPV your `resource list` output is from the case where `dshot_burst = off`, could you try with `dshot_burst = on` again and providing a `resource list` as well?

MartinFPV, 2018-07-16:
Hi @DieHertz 

yes sure, after work when I'm back home.

TheRealOrange, 2018-07-16:
Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

This is the dma list output after dshot_burst = on

DieHertz, 2018-07-16:
Looks ok, and still not working? You best bet is wait until Michael or I have a look at it. I have some holybro hardware samples on their way to me

mikeller, 2018-07-16:
Digging it out right now...

mikeller, 2018-07-16:
Doesn't look good, M3 and M4 both look mangled with Dshot600:

Without DMAR:

[image]

With DMAR:
[image]

Apart from that I could not spot any conflicts.

I did a quick test with optimisations disabled in `common_fc_pre.h` (since it's an F745), did not make a difference.

DieHertz, 2018-07-16:
@mikeller Try setting FLASH and FLASH1 to AXIM in f74x linker file

mikeller, 2018-07-16:
@DieHertz: Nope, still the same:

[image]

davidlittlefarmer, 2018-07-16:
I dont't know if this helps but i ran motors at 1055 in motor testing and put oscillator on motor 1 pwm output and got nice square wave with 0.3v amplitude with 1 us duty cycle but on motor 3 i got only 0.1 and 0.2 voltage spikes, and 0.6 us duty cycle. I have old analog tektronix so i dont know the numbers exactly, but i used matrix on screen to calculate that numbers.
BF 3.4.0 KTF7 32/16khz DSHOT 1200 4.5% motor idle throttle value

TheRealOrange, 2018-07-18:
Hello @DieHertz 
Are there any further updates? I'm sorry to disturb you, but I am curious why only motor 3 & 4 seem to be experiencing this problem. I did some very very superficial testing an found that the problem persisted when I swapped motor 1 & 3 in resource list, the issue did not move with the motor, it stayed on the same hardware locations (so now motor 4 and 1 don't work but motor 2 and 3 do). Thank you for trying to help.

DieHertz, 2018-07-18:
I don't have the hardware yet.

sputnik378, 2018-07-18:
I have the exact same issue, I will patiently be flying multishot waiting for a solution...

zadeeb, 2018-07-19:
Well, at least I'm not alone in experiencing this very same issue.  Currently running the latest BF build(as of a few nights ago) and motors 3 & 4 are a no go with DSHOT1200/600/300, but work just fine running Multishot.  I'll try rolling back BF to 3.3.2 to see if that improves the issue, but I do hope that this BF issue is resolved soon.

sputnik378, 2018-07-19:
I am on Butterflight, and it was the exact same. I am now using Proshot 1000 and a modified HEX so that I can still have turtle mode. Decided not to maiden it today, but will give it a flight in the morning. I also noticed that anytime I went over 8k/8k it would jack my CPU utilization to 100% which seems odd to me on an F7 board.

tonino1991, 2018-07-19:
today I will try to connect both esc to the m5 and m6 pads and try to "resource" them in betaflight 3.4 to check if they work with dshot, not the ideal solution but at least a temporary one better than use multishot

I hope a dev takes a look at this issue soon.

nerilorenzo, 2018-07-20:
I can confirm that I have the same problem with betaflight 3.4 and 3.3.3
but with 3.3.2 all motors works correctly

jflyper, 2018-07-20:
@sputnik378 
None of the ButterFlight version is not ready for F7 yet.

nerilorenzo, 2018-07-21:
Sorry for the question, but there is some one that can estimate how much time is need to solve the problem?

TheRealOrange, 2018-07-21:
@DieHertz sorry to trouble you, have you received the hardware samples yet?
Small input, I wonder if the problem is with the timer setup in the target file? I don't really understand the arguments given to the macro to define the timer:
```
    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_MOTOR, 0, 0), // M1 , DMA1_ST7
    DEF_TIM(TIM3, CH4, PB1,  TIM_USE_MOTOR, 0, 0), // M2 , DMA1_ST2
    DEF_TIM(TIM1, CH1, PE9,  TIM_USE_MOTOR, 0, 2), // M3 , DMA2_ST2
    DEF_TIM(TIM1, CH2, PE11, TIM_USE_MOTOR, 0, 1), // M4 , DMA2_ST4
````
I understand that the `TIMx` argument sets the timer to use and `CHx` the output channel of the timer, to the port `Pxx` but what do the two numbers at the back mean, and why is motor 3&4 at 2 and 1. (please correct me if I am wrong) How do you debug the code during runtime?

DieHertz, 2018-07-21:
These numbers are DMA variant numbers, on TIM1CH1 and TIM1CH2 there are 3 and 2 DMA Stream variants to select from, these values see correct.

Are you guys using PPM receivers by any chance? Or maybe you're using Camera Control?

TheRealOrange, 2018-07-21:
I am using a serial based receiver and RunCam camera control (which is serial based too ad as a sidenote, it doesn't work, but one issue at a time XD) So I don't think these will affect the DMA outputs.

DieHertz, 2018-07-21:
Try this HEX.
[betaflight_3.4.1_KAKUTEF7.hex.zip](https://github.com/betaflight/betaflight/files/2216271/betaflight_3.4.1_KAKUTEF7.hex.zip)

DieHertz, 2018-07-21:
So it's likely not LL DSHOT rewrite which I reverted in this build.
No word on my hardware samples yet, they sent it using DHL and tracking got updated 0 times

bhuism, 2018-07-21:
I've just reproduced this issue, @DieHertz f I can test anything, happy to help

MartinFPV, 2018-07-21:
@DieHertz In which country do you live?

DieHertz, 2018-07-21:
Russia, just what my GitHub profile says :)

MartinFPV, 2018-07-21:
Sorry i did not look. Uff the transit time for shipments from Germany to Russia is 2 to 4 weeks. So it does not help either.

sputnik378, 2018-07-23:
I've got the quad fully built ready for testing too, if you come up with something. I'd be happy to give it a shot.

In regards to Butterflight it's not an option. To be functional currently what I have gleaned is that I have to use proshot 1000 at 8k/8k and Betaflight 3.3.2. Right?

tonino1991, 2018-07-23:
I do recommend to use betaflight 3.4 with 8k/8k/multishot, I was using it that way and it worked like a charm…

Will tell how does the st-link work with the board when I get one in the afternoon.

-Antonio Ruiz

bf 3.4.0 with multishot works too.

davidlittlefarmer, 2018-07-23:
Hi all, i probably solved the issue. Please try this image, it worked for me at DSHOT 1200 8k/8k.

I commented out from target.h this
`#define CAMERA_CONTROL_PIN PE13`
Because the same pin was used for this macro.
`#define ESCSERIAL_TIMER_TX_PIN PE13 // ( Hardware=0)`

Now i am going to meet my friend, he got also Kakute F7 and i will try to find correct pin for camera control, because i built my drone too tight and i am unable to reach pins of STM32.

[betaflight_3.5.0_KAKUTEF7.zip](https://github.com/betaflight/betaflight/files/2220015/betaflight_3.5.0_KAKUTEF7.zip)

TheRealOrange, 2018-07-23:
I'll try it tomorrow when I'm free but is it ok to just resource camera control to none? This seems promising, thank you for the effort!

DieHertz, 2018-07-23:
Yes, that's it, i asked whether some of you use camera control:
>Are you guys using PPM receivers by any chance? Or maybe you're using Camera Control?
Turns out it's enabled by default. I'll submit a fix

DieHertz, 2018-07-23:
Someone give me a two-sided detailed photo of the board, please.

Kieptaj, 2018-07-23:
Flashing THX

TheRealOrange, 2018-07-23:
@DieHertz
Here is the detailed manual http://www.holybro.com/manual/Holybro_Kakute_F7_AIO_Manual.pdf

DieHertz, 2018-07-23:
So the designers of this board didn't follow guidelines and placed camera control on a pin without exclusive timer available. We'll have to disable it and add a note to the wiki.

DieHertz, 2018-07-23:
Here's what broke it: #5796

davidlittlefarmer, 2018-07-23:
#5796 is reason, why i started looking around camera control when i found it.

utkinpol, 2018-07-23:
hi, i am trying to configure this board to be used with arducopter - and would like to confirm with the community 2 issues, if those ar showing on betaflight or not: 

are there any known issues for 2 things on this kakute f7 and f7 aio boards:
1) SD card sporadically receiving partially written files with garbage in them - what is the MHZ setting in betaflight for SD card? in ardupilot it is attempted to be set as this:
SPIDEV sdcard   SPI1 DEVID1 SDCARD_CS   MODE0 400*KHZ 25*MHZ

2) there is an issue with frsky passthrough telemetry output - signal is not getting recognized, as it seems, by the non-inverted port of the r-xsr. i tried to put a diode between tx/rx, resistors, etc - nada. is this board known to work fine with betaflight to pass s.port stream? 

pls confirm.

DieHertz, 2018-07-23:
I don't know of any similar issues reported

MartinFPV, 2018-07-23:
@DieHertz 
[image]
[image]

DieHertz, 2018-07-23:
Thanks. I can only see one CAM pad, is it for camera control?

davidlittlefarmer, 2018-07-23:
I got that board from friend F7 AIO, and DAC pin on FC is really connected to PE13 (pin number 43) through 470 ohm resistor.
In datasheet page 45.
https://www.st.com/resource/en/datasheet/stm32f745ie.pdf

@utkinpol 
I am using successfully R-XSR with F.port on T6 with no hacks on receiver and this settings
`serial 6 1024 115200 57600 0 115200`
`set serialrx_provider = FPORT`
`set serialrx_inverted = ON`
`set serialrx_halfduplex = ON`

davidlittlefarmer, 2018-07-23:
@DieHertz 
[image]
Yes, by datasheet that pad is for camera control.
http://www.holybro.com/manual/Holybro_Kakute_F7_AIO_Manual.pdf

MartinFPV, 2018-07-23:
@DieHertz 
[image]

DieHertz, 2018-07-23:
Could you check continuity between PE13 and this pad on your FC?

DieHertz, 2018-07-23:
If you're going to use camera control, you can resort to M5, M6 or LED_STRIP pins for that matter.
DSHOT should work once PR is merged to master.

davidlittlefarmer, 2018-07-23:
I have F7 AIO and between PE13 and CAM pad for wire is 470ohm smd resistor.

I can also confirm, that when i free resource from that pin, the motors spin on 3.4.0 with DSHOT 1200
`resource CAMERA_CONTROL 1 none`
I just tried this with clean freshly flashed firmware.

DieHertz, 2018-07-23:
Wow, 470 Ω. There's another place where they didn't follow the guidelines.
[image]

Thanks guys, seems like we have figured it out with a joint effort :+1:

bhuism, 2018-07-23:
I concur, 3.4.0 flies great with:

resource CAMERA_CONTROL 1 NONE

8k/8k dshot1200

utkinpol, 2018-07-23:
what is used in the betaflight code to communicate with SD card? is it set with same 400KHZ 25MHZ combination, or different?

nerilorenzo, 2018-07-24:
Thanks to have solved the issue.
For what I understand the motors 3 and 4 are incompatible with camera control. It is possible to use motor 5 and 6, instead of 3 and 4, and camera control at the same time?

[18 more replies]

PR #6436 (2018-07-24): Fix KakuteF7 motors 3 and 4
Credit to @davidlittlefarmer for finding and confirming

Fixes #6306.
Pin defined for camera control in #5796 has no dedicated timer and collides with motors 3&4. Only protocol that "works" due to it is MULTISHOT, any other protocol likely doesn't.

Camera control alternative is either of M5, M6 or LED_STRIP, but since this board has a dedicated `CAM` control pad (which doesn't work), I won't set any defaults. We should add a Wiki board description page and mention this issue there, but that will follow in another PR once I get my samples.

http://www.holybro.com/ma [... cut]
closed 2018-07-24 by mikeller