betaflight/betaflight #6309: BF 3.4 Unable to arm when gyro 32kHz enabled, CALIB flag

GenuineFlollie, 2018-07-08:
**Describe the bug**
The CALIB flag (Wait for sensor calibration to complete) is raised when arming the quad and 32kHz sampling mode is enabled.
OMNIBUSF4FW

**To Reproduce**
Steps to reproduce the behavior:
Clean install of BF 3.4 or BF 3.4 RC 5
Minimal configuration to test arming (see diff)
* Quad arms
Enable gyro 32kHz sampling mode
* Quad doesn't arm and beeps 2 long and 2 short (CALIB flag)

This is repeatable, 32kHz on, no arming, 32kHz off quad arms

**Expected behavior**
Quad should arm when gyro 32kHz sampling mode is enabled

**Flight controller configuration**
# version
# Betaflight / OMNIBUSF4FW (OBFW) 3.4.0 Jul  8 2018 / 04:24:16 (cce8a59be) MSP API: 1.39
board_name OMNIBUSF4FW
manufacturer_id 
# name
name Stingy fireworks
# resources
# mixer
# servo
# servo mix
# feature
feature AIRMODE
feature ESC_SENSOR
# beeper
beeper -ON_USB
# beacon
beacon RX_LOST
beacon RX_SET
# map
map TAER1234
# serial
# led
# color
# mode_color
# aux
aux 0 0 0 1300 2100 0
# adjrange
# rxrange
# vtx
# rxfail
# master
set acc_calibration = 104,-205,-43
set serialrx_provider = SBUS
set min_throttle = 1000
set dshot_idle_value = 300
set use_unsynced_pwm = ON
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 16000
set align_board_yaw = 180
set beeper_dshot_beacon_tone = 4
set pid_process_denom = 2

# profile
profile 0
# rateprofile
rateprofile 0

**Setup / Versions**
 - OMNIBUS FIREWORKS V2 @local store
 - Xm+, Unify pro race, Runcam Micro, BF 32 bits escs
 - Not relevant

**Additional context**
Add any other context about the problem here.

GenuineFlollie, 2018-07-16:
Setting the gyro_calib_noise_limit to a value of 150 doesn't seem to work for me.

mikeller, 2018-07-17:
@GenuineFlollie: Looks like the gyro on your board does not work well with 32k. The reality is that 32k mode is not officially supported by the gyro's manufacturer, probably because a high percentage of gyros made are really noisy when used this way. Not much you can do there, except change the board and hope that the gyro on the new board works in 32k.

GenuineFlollie, 2018-07-22:
Ok, thanks mate. I'll just stick with 8k/4k for now.

closed 2018-07-22 by GenuineFlollie