betaflight/betaflight #6310: Support for ICM-20948

AlexRochelle, 2018-07-08:
Adding support for the 20948 or any 9 axis IMU would allow a greater variety of flight controllers to work. It would be beneficial to have a driver for the 20948 in betaflight.

etracer65, 2018-07-08:
Do you have any examples of flight controllers using this chip?

AlexRochelle, 2018-07-08:
Not as of yet but I suspect that some flight controllers will start using it. Also, I am working on developing my own flight controller and would like to use the 20948

AlexRochelle, 2018-07-08:
The Cube flight controller from pixhawk uses the 20948 and there are a few other people that want support to use the 20948 as an external compass.

etracer65, 2018-07-08:
Well Betaflight is open source you you're free to contribute the appropriate driver code when you develop your own flight controller.  It's unrealistic to expect that anybody is going to develop it on spec without any boards that need it (or boards to test with).

AlexRochelle, 2018-07-08:
Ok. I will try developing a driver for it.

jflyper, 2018-07-08:
- Isn't ICM-20948 a relabelled MPU9250?
- I don't recommend using I2C connected internal compass for Betaflight quads, as compass reading operations interfere with acc/gyro operations.

AlexRochelle, 2018-07-08:
It is effectively an updated 9250 with faster SPI communication. It also uses a lot less power than the 9250. They are pin for pin compatible with each other but I do not know if they will communicate the same.

jflyper, 2018-07-09:
> It is effectively an updated 9250 with faster SPI communication.

Wasn't 9250 capable of handling fSCLK = 20MHz @ VDD=3.3V?

MJ666, 2018-07-09:
VDDIO is only 1.71V to 1.95V, This will not work together with the usual 3.3V we use for most other chips and would require a additional power supply for this. The ICM-20948 is basically a ICM-20648 with an AK09916 connected to the slave I2C port. This is similar to the MPU9250 and will have the same limitations. Even SPI communication is finally slower. I think it is not the best choice to be used on actual FC's.

AlexRochelle, 2018-07-09:
Ok. Thank you.

closed 2018-07-10 by AlexRochelle