betaflight/betaflight #6311: BF 3.4 BARO and GPS HOLD modes not working on Matek FC

Hydrologisst, 2018-07-08:
Hello, 

I have Matek F411 mini and F405 CTR FC.
Baro or gps hold modes are not working at all.
On OSD screen fly mode not changing when activating baro or gps hold mode.

On all 3.4 candidate versions didin't worked.

Where is problem?

McGiverGim, 2018-07-08:
The alt hold and gps modes had been removed. Only the new GPS Rescue is available. Did the work for you? I tried them in earlier versions and the didn't work as expected.

s0up, 2018-07-09:
Neither of these modes work (and haven't for quite a long time) but they should probably be removed on the configurator end as well instead of just the firmware itself.  Same with nav PIDs :P

mikeller, 2018-07-09:
@s0up: The modes shown in the configurator are entirely driven by the flight controller.

tavdog, 2018-07-11:
I'm configuring a matek f405-osd for gps rescue and I still see gps hold and gps home listed in the modes tab.

mikeller, 2018-07-11:
@tavdog: That's a bug. #6318 fixes it.

ghostkcleung, 2018-07-15:
Yes, GPS Hold is removed.

https://youtu.be/s9WLOH96znI

closed 2018-08-21 by stale[bot]