betaflight/betaflight #6313: 3.4 Release - slow LUA script update and intermittent "sensor lost" warnings

Klemvore, 2018-07-08:
**Describe the bug**
LUA script on FrSky x9d+ taking 10+ seconds to update(sometimes it doesn't at all)
Systematic deletion of unused sensors at radio does not cure "sensor lost" warning 

**To Reproduce**
Upgrade from 3.4 RC5 to 3.4 release using 3.4 RC5 "diff" 

**Expected behavior**
As in 3.4RC5 the lua scripts updated almost immediately, and there was no "sensor lost" warnings.

**Flight controller configuration**
```
# diff
# Betaflight / RG_SSD_F3 (RGF3) 3.4.0 Jul  8 2018 / 04:29:25 (cce8a59be) MSP API: 1.39

board_name RG_SSD_F3
manufacturer_id 
name Loki

feature -RX_PPM
feature RX_SERIAL
feature TELEMETRY
feature RSSI_ADC
feature AIRMODE
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
map TAER1234
serial 1 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
aux 0 0 0 1700 2100 0
aux 1 13 1 1700 2100 0
aux 2 26 4 900 1300 0
aux 3 35 3 900 1300 0
aux 4 36 1 1700 2100 0
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 180
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 2
set vtx_freq = 5843
profile 0

set dterm_notch_cutoff = 0
set iterm_relax = RP
set i_pitch = 52
set d_pitch = 31
set i_roll = 47
set d_roll = 30
set p_yaw = 70
rateprofile 0

set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_breakpoint = 1050
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
    RG_SSD_F3 (RGF3) - RotorGeeks.com
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
    FrSky R-XSR (updated Firmware), TBS Unify.. 
 - how are the different components wired up.
   Correctly for the last year...? 

**Additional context**
None - 3.4 flies amazingly! Just these 2 little nuances. Keep up the great work team!

DieHertz, 2018-07-09:
I strongly recommend a diy hardware inverter then

darksonny, 2018-07-09:
I have that sensor lost warnings with the Crossfire as well.
Seems like something to do with GPS, but don't ask me why.

Klemvore, 2018-07-09:
Sorry, It is an XSR RX. But I don't think that the hardware inversion hack is a solution that should be implemented as it worked flawlessly previously.  i see there is the "F.Port" firmware on the FrSky website though. My question/concern about that firmware is will the Firmware just work or do I need to disassemble the quad, and solder to a different port, or just upgrade the firmware and send it.

roeemz, 2018-07-09:
I think you shouldn't use f.port on soft serial, but I am not sure...

You will need some disassembly to update the receiver (I didn't succeed with serial pass through through a soft serial)

Klemvore, 2018-07-09:
Just to make certain all my configuration settings were as they should be. I cross checked them against another quad with the same overall setup (I inadvertently missed shutting off the ACC and checked the RSSI signal) . I re-discovered the sensors on the TX, and I now get no warnings! However; the lua scripts time to populate the fields has gotten better, but still not on par with what they use to be. 

It appears that shutting off the: 
Acc (typically have it off anyhow. I must have missed this during setup)
RSSI_ADC | Analog RSSI input <--culprit for the "sensor lost" warning I am guessing.

roeemz, 2018-07-09:
I just unmapped all my uarts(except for the tramp protocol) and turned off acc and it seems to be fixed.
I guess Ill start turning on things slowly and see what breaks it when I have some more time.

I am assuming it might be the ESC telemetry or maybe some malfunctioning / shorted uart

hsuq510, 2018-07-10:
Having this issue also, got better response on every F4 using 4/4 intestead 8/8. I got the issue major with softserial, but physical uart lags as well, also everytime LUA hangs or waits, I got sensor lost. Something in telemetry code changed ?

Alpha200, 2018-07-10:
I have the same issue with my Omnibus F3. At only 17% CPU Load and SmartPort on a physical uart port, I get permanently sensor lost warnings. Mainly the telemetry of my KISS ESCs is lost and slowly updated.

Also same issue with the lua script on the FrSky QX7.

Alpha200, 2018-07-10:
"Downgraded" to RC5 and the error is gone.

kako0sb, 2018-07-11:
have the same issue, maybe in the next update will it fix...

mikeller, 2018-07-11:
#6300 was a fix to an issue causing a delay in the SmartPort protocol to be not properly observed. This delay seems to be required to make it work for older RX, but it looks like it's causing issues of its own with (presumably) newer firmware.

We'll have to look into it, and figure out a compromise that works with a reduced delay.

CharlosFPV, 2018-07-15:
Got the same issue telemetry drops frequently, sensor lost. Fport on a f4 fc connected to a UART. What is RC 5? I’m not familiar with these terms first time frsky user.

mikeller, 2018-07-15:
@Klemvore: Can you please try to see if this fixes it:

[betaflight_3.5.0_RG_SSD_F3.zip](https://github.com/betaflight/betaflight/files/2195408/betaflight_3.5.0_RG_SSD_F3.zip)

Others: Let me know what targets you need, or build from https://github.com/mikeller/betaflight/tree/reduced_smartport_delay.

roeemz, 2018-07-15:
Ill test it hopefully tonight on omnibus fireworks (omnibusf4fw).

mikeller, 2018-07-15:
@roeemz: 
[betaflight_3.5.0_OMNIBUSF4FW.hex.zip](https://github.com/betaflight/betaflight/files/2195545/betaflight_3.5.0_OMNIBUSF4FW.hex.zip)

Klemvore, 2018-07-15:
@mikeller: I tried the new hex. I am still getting the "sensor lost" warnings, and the pages are still slow to populate. When trying to make changes via LUA I always see a "retrying" message.

CharlosFPV, 2018-07-15:
@Mikeller do you have a hex for this f4? 
(Luxf4osd).

kapstaad, 2018-07-15:
@mikeller FWIW, I'm using a Hobbywing Xrotor Mini stack (OMNIBUSF4SD target) running BF 3.4.0 and a Taranis X9D+ runing OpenTX 2.2.1 with current Lua scripts.  Experiencing the same issues; slow telemetry updates, "sensor lost" warnings, and "retrying..." errors whenever attempting to use the TX to send e.g. VTX power setting changes.  Would be happy to help test potential fixes.

mikeller, 2018-07-15:
Looking into putting #6098 into 3.4.1 to address these issues with a fix for FPort.

klutvott123, 2018-07-15:
I have been having the same issues. "sensor lost" all the time. Seems like using the lua script only makes it worse. 

I tried compiling it with mikellers reduced smartport delay and then the lua script didn't work at all. 
I then reduced the smartport delay to 100 and that seems to be working. Flew 3 packs and got one sensor lost, but it came back again immediately. It also lets me change pids etc as much as i want. Lua script pages are loaded as fast as before.

Using SPRACINGF3 clone with R-XSR receiver running latest 180615 beta f.port firmware. 
Taranis q x7 with opentx 2.2.1. Lua script version 1.0.2.

mikeller, 2018-07-15:
@klutvott123: Thanks for testing. Happy to go for 100 us delay if that works, but it will need testing with XSR as well, and preferably R9slim / R9mini.

RubberDuck00, 2018-07-15:
2.2.1 OpenTX Taranis plus / X4R-SB / wraith32 v2 (blheli32 - DSHOT1200) / JBF4v1.3 / 3.4.0 Betaflight
I'm getting "sensor lost" warnings on the Taranis when enabling 4x ESC telemetry on corner RX4 pads.  No issues if I disable ESC in betaflight.  I get 12 sensors without ESC telemetry and around 30 with.  Additional sensors are hard to tell apart as each ESC has the same telemetry list.  For example, VFAS is repeated for each telemetry input.  LUA script makes the "sensor lost" event occur more often when ESC telemetry is enabled on all 4 ESC's.  Earlier today I was running the original JBF4 with one ESC telemetry with no issues.  I just changed out the FC and connected all 4 ESC telemetries.  No other changes to the system.

Klemvore, 2018-07-15:
Update: Flashed the FPort firmware to two XSR receivers. Updated to BF 3.5 hex. LUA scripts now load the fields faster, but still have the "sensor lost" and "retrying" when saving changes ("sensor lost" now only happens during the attempt to save").

mikeller, 2018-07-16:
@Klemvore: 'sensor lost' is an issue of its own: The way SmartPort is implemented the flight controller cycles through sending all supported telemetry values. So if there are more sensors / features activated, the interval between individual values gets longer, and sometimes too long, causing OpenTx to report 'sensor lost'. When you enter the lua script, the MSP data is sent, using some of the telemetry bandwidth, making intervals even longer.

I am not sure that we can do something about this on our side - there is only limited bandwidth available, and as a consequence the more data we try to send per iteration the longer the intervals will get. In reality, the way that OpenTx detects sensor presence / absence is completely geared towards each of the sensor data frames being provided by a separate hardware device, and not our case where all of the sensor frames are sent by the same hardware, and as a consequence it's either 'all sensors good' or 'all sensors lost', and no way for an individual sensor being lost.

Maybe we should consider changing OpenTx to be configured to only report 'sensor lost' if there are no sensor update frames at all received, but not if the interval for an individual sensor gets too long...

Klemvore, 2018-07-16:
@mikeller: I'm good with the way it is working right now. Updating to Fport seems to have fixed the major lag issue I was having during the LUA field update process. There are a lot of sensors coming through on telemetry that I don't use. I'll look into removing them and see how it goes. Perhaps If there was an option in BF to be able to turn certain "sensors" off the end user could control the amount of bandwidth being transmitted. to a certain degree of course. hypothetical. lol

 Thanks for the help, and keep up the good work!

roeemz, 2018-07-16:
Sorry. Didnt have time to test this yet. Ill do some XSR, RXSR, r9 slim and mini soon...

Klemvore there are two feature requests asking to turn off telemetry sensors, maybe it will be added soon :)

And f. Port is not a good solution because unfortunately not all FCs are supported at the moment.  My personal attempt with f. Port was good too and it seems to be more reliable than s.port!

paulo-serrao, 2018-07-16:
I'm also experiencing frequent sensor lost warnings, using BF3.4 and R-XSR on s.port (171009 fw). 

On Taranis I have deleted all the sensors I don't use, but I still get those sensor lost warning on the VFAS telemetry, yet I see other sensors updating much more frequently (like A4 which is VFAS/#cells), but the VFAS seems to taking much longer to update causing the warning.

I do have ESC telemetry enabled on FC (but deleted on Taranis side), not sure if that is what is causing this issue (too much telemetry data to send?), but is that is the case, then there should be a CLI option on BF to choose what telemetry the FC sends.

Maybe this is somehow related with https://github.com/betaflight/betaflight/issues/5785

mikeller, 2018-07-16:
Try these, @Klemvore, @roeemz, @kapstaad (from #6382):

[betaflight_3.5.0_OMNIBUSF4SD.hex.zip](https://github.com/betaflight/betaflight/files/2197632/betaflight_3.5.0_OMNIBUSF4SD.hex.zip)
[betaflight_3.5.0_OMNIBUSF4FW.hex.zip](https://github.com/betaflight/betaflight/files/2197633/betaflight_3.5.0_OMNIBUSF4FW.hex.zip)
[betaflight_3.5.0_RG_SSD_F3.hex.zip](https://github.com/betaflight/betaflight/files/2197634/betaflight_3.5.0_RG_SSD_F3.hex.zip)

(Please report back.)

roeemz, 2018-07-16:
It seems better with ESC telemetry off (lots of duplicate sensors with ESC telemetry on, but that is not the issue here...).

When ESC telemetry is turned off I am not getting any sensors lost.

regarding lua scripts:
changing to pit mode and R8: "saving" "retrying" and then it is updated - working almost as intended.
turning off pit mode right after: "saving" "retrying" - but it fails, no change is done, the LUA script does show that it failed, it still says pit mode.
3rd attempt to turn off pit mode - success

changing to R1 - syntax error, not enough memory (am using the LUAC script)
changing to R1 again - saving, retrying, sensor lost, doesnt change
R1 again - saving, retrying, stay on R8.
changes to R2 - smooth, not even retrying.
changing to r3 - smooth
changing back to r2 - smooth
changing to r1 - saving, retrying, fails
changing to r3 again - smooth
changing to r4 - saving, retrying, stays on r3
changing to r4 again - saving, retrying, stays on r3
changing to r5 - smooth
changing to r6 - smooth
changing to r7 - smooth
changing to r8 - saving, retrying, changed to r8.

using R9 Slim
Fireworks v2
sbus connected to serial_rx 1
s. port connected to serial_tx 11 which is where serial_tx 1 was

latest lua scripts version (1.0.3 I believe?) and x9dp-openTx v2.2.2-20180525 firmware from frsky-rc website

roeemz, 2018-07-16:
I removed all sensors.
R1 and R4 fail after "Retrying"

but everything else is updated quickly.

(btw, there is a bug, if all sensors are off, the LUA script is flashing with "no telemetry" but it is still working)

Changing to pit mode still doesnt work on some channels...

Klemvore, 2018-07-16:
@mikeller: Tested the last 3.5 hex. I no longer get the "lost sensor" warning. I only get the "saving">"retrying" message. 
This is the Diff

# diff
# Betaflight / RG_SSD_F3 (RGF3) 3.5.0 Jul 15 2018 / 14:37:22 (b504a4869) MSP API: 1.40

board_name RG_SSD_F3
manufacturer_id 

feature -RX_PPM
feature RX_SERIAL
feature TELEMETRY
feature AIRMODE
map TAER1234
serial 2 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 26 3 1700 2100 0 0
aux 3 36 1 1700 2100 0 0
set acc_hardware = NONE
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 180
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 2
set vtx_freq = 5843
profile 0

set dterm_notch_cutoff = 0
set dterm_setpoint_weight = 100
set d_roll = 27
rateprofile 3

set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set tpa_rate = 0

#

paulo-serrao, 2018-07-16:
The saving retrying is an old issue, it was already present on 3.3. Some combinations of channels/band/power never work. I opened an issue about it on the lua github and it seemed more suited, it was confirmed by multiple persons, but noone could solve the issue. https://github.com/betaflight/betaflight-tx-lua-scripts/issues/112

roeemz, 2018-07-16:
@paulo-serrao, oh ok, it is not the only issue tho :/
You really cant use ESC Telemetry and a telemetry receiver.

paulo-serrao, 2018-07-16:
@roeemz can you try R1 and R4 work at 100mw? At least I didn't had problem with that combination.

dmagicone, 2018-07-16:
Can we get one of these fixed versions for OMNIBUSF4 also.

mikeller, 2018-07-17:
@dmagicone: 
[betaflight_3.3.0_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/2199783/betaflight_3.3.0_OMNIBUSF4.hex.zip)

closed 2018-07-17 by mikeller

dmagicone, 2018-07-18:
Any reason that one is named 3.3.0 instead of 3.5.0 like the others?

mikeller, 2018-07-18:
@dmagicone: Erm, because that's the wrong one, sorry for that. But #6382 has been merged, so if you get the latest development build for your board you'll have it.

callagga, 2018-07-22:
Just gone to 3.4 plus also included ESC Telemetry for the first time.  I'm getting frequency sensor lost Taranis warnings too, and they seem to be the ones associated with ESC Telemetry, i.e. VFAS and Current.  

Question:  Is this the same issue?   Should I just wait for 3.4.1 release to come out?   

_Happy to try to do a fix/patch if there's something that fixes this if I could get a pointer to how to (e.g. would I need to download from latest main branch if the fixed is learned and learn to build/compile betaflight myself?)_

mikeller, 2018-07-22:
@dmagicone: 3.5 (soon to be out) will have a fix for that.

PR #6382 (2018-07-17): Removed delay from SmartPort frame processing.
Fixes #6313.