betaflight/betaflight #6315: Support F. Port as a Telemetry Provider Only

roeemz, 2018-07-08:
F4 FCs must enable soft serial to use F. Port unless they have a suitable inverter.
In order to use F. Port firmware when soft serial doesn't give adequate performance for RC control, one can use sbus for rc control while getting telemetry only from the f. port.

It will be very nice if it will be possible to choose f. port as telemetry output and not just as a serial rx provider.

DieHertz, 2018-07-08:
I don't know of Receivers which continue to output SBUS when they're flashed to FPORT firmware.
XSR and R-XSR output FPORT or SBUS , not both

roeemz, 2018-07-08:
The R9 receivers do it.

mikeller, 2018-07-09:
At least earlier versions of the FPort firmware for XSR did in fact output SBus on the SBus pin as well as FPort (potentially a relic of a hasty implementation of FPort as an add - on). So at least in theory, and for some RX, this should work.

But at the same time, this does beg the question why you are not just flashing the SBus / SmartPort firmware if you want to use SBus and telemetry separately - if you are using softserial for telemetry, using FPort uses a lot more data than SmartPort, and decoding data for softserial increases the CPU load, so SBus / SmartPort is probably the better combination.

roeemz, 2018-07-09:
The F. Port firmwares for R9 seem to work much better. I am having issues with telemetry and LUA scripts on smartport while FPort actually works. (Slow / Failing to write LUA scripts and duplicate sensors that update in very low frequencies or not at all)

The FPort versions also supply RSSI natively which is very desirable by the R9 community.

At least until F4 is dead, I think a lot of people will appreciate this!

roeemz, 2018-07-09:
Thats exactly what I had in mind, having two serial rxs and using just one of them since it will be the easiest to implement. But I trust you if you are saying that is too much :/

callagga, 2018-09-01:
Am I correct in saying:

1.  On an F4 board with no SmartPort pin, there is no way to get FPort working such that both control/telemetry/digital RSSI is working?    seems the issue is you need soft serial and then getting telemetry as well as control through soft serial doesn't work?   

2. If you drop back to Smart Port (not FPort) you do lose the digital telemetry RSSI broadcast feature?  (so you have to drop back to passing back from the Taranis?)

mikeller, 2018-09-02:
@callagga: Pretty much - using the uninverted pad on r-XSR or the uninverted hack on XSR to connect to an uninverted TX pin will make FPort work in these cases as well.

I have pondered the 'FPort telemetry only' situation, and while I can see this as some sort of workaround, I don't see it as a good one:
- the incoming RC data stream on the FPort port still has to be decoded in order to be able to detect telemetry frames and get the timing right, so when using SoftSerial for FPort using it for telemetry only will not reduce the load or data rate on it;
- the fact that SBus frames are sent on the SBus port when FPort firmware is used is just an artefact - the FPort specification does not have any provision for this, and it could be dropped with the very next firmware release.

All in all I would say, if you are at this point, and there is no other way, then using SoftSerial for FPort (including RC data) is no worse than using SoftSerial for FPort **and** using SBus in parallel.

mikeller, 2018-09-02:
@callagga: I am not saying it _will_ work. What I am saying is that it is not more likely to not work than if you used SBus over a hardware UART **and** FPort over SoftSerial - you'll just have to try and find out, and probably execute a bit more caution when testing after a firmware upgrade.

callagga, 2018-09-02:
update - couldn’t get F-port working over soft serial so have dropped back to SPort + SBuss + AnalogRSSI

jflyper, 2018-09-02:
@callagga Software serial is compatible with F.Port. Are you saying that you could not make F.Port work as a telemetry protocol running independently from serial RX on a dedicated serial port all by itself?

callagga, 2018-09-02:
I had been rebuilding my quad, going to R9M/R9Slim+ and running out of time on the weekend, and wanting it get it working.  So I didn't get it working but it could have been my fault....  What I did however:

* Upgrading Taranis/R9M/R9Slim+ to latest firmware, R9Slim+ flashed with F-Port version, using SpeedyBee FC.
* 1st try:  F-Port to Sbus => didn't work (my misunderstanding)
* 2nd try:  F-Port to S5 with Softserial => couldn't get it working, tried the different CLI setting
* 3rd:  SBus & Smart-Port to S5 with Softserial (re-flashed receiver) => got it working

I mainly was going to F-Port to get the digital RSSI, but I'm now running with the analogue RSSI output from R9Slim+ to an RSSI pin that the speedbee FC has.

callagga, 2018-09-02:
(sorry - just updated my previous post to make it clearer)

closed 2026-09-26 by blckmn