betaflight/betaflight #6319: IBUS telemetry GPS sensors not working

vkolotov, 2018-07-09:
GPS coordinates and altitude are not displayed on FlySky radio transmitter screen (displayed as all zeroes).

I've set up betaflight to transmit FlySky GPS coordinates as per is:

```
set ibus_sensor = 2,3,4,5,6,7,20,21,22,126,128,129,130,131,132
```

All sensors up to `126` are displayed correctly (voltage, rpm, amperage etc). 

The 128 - 132 sensors (0x80 - 0x84, GPS latitude, longitude, altitude etc as per [ibus_shared.h](https://github.com/betaflight/betaflight/blob/a71a2f2b92b98759dc74b523cc16cb2aadc098be/src/main/telemetry/ibus_shared.h#L66)) are not getting displayed on FlySky radio transmitter.

I can confirm that I do see longitude and altitude on my OSD.

Betaflight v3.4.0
JBardwell F4 board
FlySky FS-i6 radio (FlyPlus 1.7.5 firmware)
FS-iA6B transmitter
Adafruit Ultimate GPS module

```
# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.4.0 Jul  8 2018 / 04:04:28 (cce8a59be) MSP API: 1.39

board_name CLRACINGF4
manufacturer_id 

# name
name meccup

# resources

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature AIRMODE

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 3 4096 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1700 2100 0
aux 1 1 4 1300 1700 0
aux 2 46 4 1700 2100 0
aux 3 13 0 1700 2100 0
aux 4 19 3 1700 2100 0

# adjrange
adjrange 0 0 6 1675 2075 12 6 0 0

# rxrange

# vtx

# rxfail

# master
set rssi_channel = 10
set fpv_mix_degrees = 25
set serialrx_provider = IBUS
set motor_pwm_protocol = MULTISHOT
set bat_capacity = 1400
set vbat_min_cell_voltage = 35
set vbat_warning_cell_voltage = 36
set ibata_scale = 337
set gps_auto_baud = ON
set pid_process_denom = 1
set ibus_sensor = 2,3,4,5,6,7,20,21,22,0,0,0,0,0,0
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1200
set osd_vbat_pos = 2081
set osd_rssi_pos = 2106
set osd_tim_1_pos = 2135
set osd_tim_2_pos = 2167
set osd_flymode_pos = 2458
set osd_crosshairs_pos = 2285
set osd_ah_pos = 2158
set osd_current_pos = 2112
set osd_mah_drawn_pos = 2145
set osd_gps_speed_pos = 2088
set osd_gps_lon_pos = 2513
set osd_gps_lat_pos = 2481
set osd_gps_sats_pos = 128
set osd_home_dir_pos = 2094
set osd_home_dist_pos = 2100
set osd_altitude_pos = 2176
set osd_warnings_pos = 2441
set osd_battery_usage_pos = 2544
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vcd_video_system = PAL

# profile
profile 0

set dterm_notch_cutoff = 0

# rateprofile
rateprofile 0

set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30

```

vkolotov, 2018-07-09:
Just so you know, I've also tried to play with `IBUS_SENSOR_TYPE_GPS_FULL         = 0xfd` sensor type with no success.

mikeller, 2018-07-09:
@vkolotov: Not an expert in IBus, but have you tried with less sensors before GPS in `ibus_sensor`, e.g. `ibus_sensor = 128,129,130,131,132`?

vkolotov, 2018-07-09:
@mikeller, I've tried lots of combinations including the one that you've provided... Nothing helps

McGiverGim, 2018-07-09:
@vkolotov you don't say nothing about it, but I suppose that in 4.3.X worked, right?

vkolotov, 2018-07-09:
Ah nah sorry, it was not working for me in 4.3.3v.

McGiverGim, 2018-07-09:
Some of the telemetry need to modify the ia6b receiver too, not only the transmitter. Do you have it modified?

vkolotov, 2018-07-10:
Ok. I tried several other options. Tried even to flash the original firmware for the transmitter. Obviously there is not any GPS support in the original one.

I tried also to check/match logic in the code that is responsible for GPS coordinates in both [Betaflight](https://github.com/betaflight/betaflight/blob/ea53a18ae00bc05bebdbe3f8d7bd765c69aebdd2/src/main/telemetry/ibus_shared.c#L292) and [transmitter firmware](https://github.com/qba667/FlySkyI6/blob/b0c5643814c26441338ff50384c2c837fde1f6ab/source/source/alt.c#L33). Unfortunately without a debugger it is quite hard to understand what's happening there. It seems that it should work...

@qba667 did awesome job to add support for iBus telemetry. It is working mostly, it is just GPS coordinates don't get displayed properly. Maybe my `ibus_sensor` settings are wrong? Or maybe I have to upgrade firmware in the receiver itself (I can't see how that would fix the issue as I see only that custom firmware is addressing RSSI representation, e.g. % vs dba)?

Unfortunately there is not any proper documentation anywhere that clearly states how to do telemetry with FlySky. I'm keen to create a page that describes everything what needs to be done to add FlySky telemetry from scratch.

It is pity that FlySky is actually not bad and there is not a proper guide for it.

@qba667 if you have some time, could you please assist?

vkolotov, 2018-07-10:
Hey @McGiverGim,

> Some of the telemetry need to modify the ia6b receiver too, not only the transmitter. Do you have it modified?

Could you please point out where is that firmware? 

Are you referring to that [one](https://github.com/qba667/MAVLinkToIbus/tree/master/IA6B)? The name is a bit misleading/confusing. Is it for iBus or Mavlink?

qba667, 2018-07-10:
@vkolotov 
If most sensors are displayed correctly it seems you are using FlyPlus 1.7.5 - is that correct?
Some old IA6B are not supporting 4 bytes sensors (0x80 - 0x84) - in this case firmware update of receiver is needed. All IA6B are not supporting extreme long frames like IBUS_SENSOR_TYPE_GPS_FULL without my modded firmware. 
The firmware you refereed enables support for long telemetry format and enable telemetry on same serial together with channel data.
You can try first official firmware from: https://github.com/povlhp/FlySkyRxFirmware

vkolotov, 2018-07-10:
awesome @qba667, I'll give it a go. I'll let you know how it goes.

vkolotov, 2018-07-10:
Just one thing to clarify @qba667,
> enable telemetry on same serial together with channel data.

will I still be able to still use both RX and TX ? Or I'm forced to use single pin only?

Cheers

closed 2018-08-16 by stale[bot]