betaflight/betaflight #6320: 3.4.0 Runaway OSD

darksonny, 2018-07-09:
Hi Guys.

My setup:
7inch frame
7inch props
Airbot Fireworks  8K/8K
Blheli32, Dshot1200

Today I had a Runaway message while tuning my quad.
It suddenly kicked in doing flips and rolls, disarmed my quad and just dropped.
The Blackbox stopped at the same moment.
I've never had that before, but while investigating and sharing turns out other people experienced the same issues testing the RCs, where I just had it flying the official release.

DIFF:
# diff

# version
# Betaflight / OMNIBUSF4FW (OBFW) 3.4.0 Jul  8 2018 / 04:24:16 (cce8a59be) MSP API: 1.39

board_name OMNIBUSF4FW
manufacturer_id 

# name
name HL7

# resources
resource MOTOR 1 B05
resource MOTOR 2 A03
resource MOTOR 3 B01
resource MOTOR 4 B00
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL

# beacon
beacon RX_LOST

# map
map TAER1234

# serial
serial 1 8192 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 1 1 1300 1675 0
aux 2 27 2 1700 2100 0
aux 3 13 2 1325 1725 0
aux 4 26 1 1700 2100 0
aux 5 35 3 1700 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 70
set gyro_lowpass2_hz = 140
set acc_calibration = 73,57,-23
set mag_calibration = 109,-11,146
set min_check = 1030
set max_check = 1950
set rssi_channel = 12
set rc_smoothing_type = FILTER
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 32000
set align_board_yaw = 180
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set yaw_deadband = 5
set pid_process_denom = 1
set osd_rssi_alarm = 50
set osd_cap_alarm = 1200
set osd_tim1 = 1280
set osd_tim2 = 3841
set osd_vbat_pos = 2466
set osd_rssi_pos = 2391
set osd_tim_1_pos = 2454
set osd_tim_2_pos = 2486
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2422
set osd_ah_pos = 200
set osd_current_pos = 2434
set osd_mah_drawn_pos = 2403
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 2105
set osd_gps_lon_pos = 2113
set osd_gps_lat_pos = 2081
set osd_gps_sats_pos = 2169
set osd_home_dir_pos = 2390
set osd_home_dist_pos = 2342
set osd_compass_bar_pos = 2132
set osd_altitude_pos = 2370
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2187
set osd_nheading_pos = 2100
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2202
set osd_esc_rpm_pos = 83
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_freq = 5732

# profile
profile 0

set dterm_lowpass_hz = 90
set dterm_lowpass2_hz = 180
set dterm_notch_hz = 180
set dterm_notch_cutoff = 90
set p_pitch = 65
set d_pitch = 28
set p_roll = 50

# rateprofile
rateprofile 1

set roll_rc_rate = 95
set pitch_rc_rate = 95
set yaw_rc_rate = 113
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 30
set tpa_breakpoint = 1050

jflyper, 2018-07-09:
@darksonny I thought you have a (truncated) bb log?

darksonny, 2018-07-09:
oh, here
[2506.runaway.zip](https://github.com/betaflight/betaflight/files/2176246/2506.runaway.zip)

etracer65, 2018-07-09:
It simply doesn't look like you flew enough to have the feature consider you in stable flight and deactivate.  In particular you were staying at very low throttle and using small stick inputs before the 100 deg/sec flip.  Read here about possibly adjusting some of the settings for your quad:

https://github.com/betaflight/betaflight/wiki/Runaway-Takeoff-Prevention

darksonny, 2018-07-09:
I'm flying on a dropped throttle curve coupled with 6S... That value might explain that I was possibly still being "watched" by the algorithm, not exceeding that throttle.
But the other question is, why would the code detect it as a problem, as I already lifted off and flew for a while... I'll turn that feature off in any future flash!

etracer65, 2018-07-09:
Read the documentation.  You were basically just hovering at a very low throttle with basically no stick inputs.  You weren't using enough control inputs to confirm that you were in stable flight.

I wouldn't recommend disabling the feature as you'll have no protection if you accidentally install props incorrectly or have some ESC/motor problem on takeoff.  Just lower the throttle limit and fly more (actually move the sticks) instead of simply hovering before a roll/flip.

etracer65, 2018-07-09:
The other aspect of this is that you never got the throttle high enough to activate air mode.  As a result the PID controller was not responding as aggressively as it would otherwise and thus your quad was not flying as well as it should.  Then when you did the flip since air mode wasn't active your PIDSUM values climbed to large values (because the quad wasn't flying well) and this triggered the runaway takeoff as the PID controller seemed to not be performing correctly.  So if you had even momentarily gotten above 32% throttle (the air mode activation point) then it's unlikely you would have triggered runaway takeoff.

darksonny, 2018-07-10:
I don't find it very protective to check the throttle only, I see what happens when I mess with the throttle curve, just imagine someone using a more cell battery or just playing around with the quad in a not standard way. I'm pretty confident seeing people fall out of the sky having the RUNAWAY message on the screen, not sure either hitting themselves or other people out of control...

etracer65, 2018-07-10:
It doesn’t check throttle only - see the wiki page I linked to.  If you don’t want to use the safety feature then that’s fine.  But it’s been in place for every user since 3.3 with no problems. You have an unusual setup and the wiki entry notes that it may be necessary to adjust the throttle check for high powered or light quads.

And again, the reason it triggered was because the PID controller thought your quad was out of control because you never activated air mode and when you went to zero throttle the PID controller shut off (as if you were still on the ground).

darksonny, 2018-07-10:
Airmode was on, it's always on in the features tab, I never turn it off.
But I didn't hit the 32% throttle, which is again a throttle dependent thing...

etracer65, 2018-07-10:
Exactly, you never activated air mode. This completely unrelated to Runaway Takeoff.  You have an unusual setup and that means the defaults may not work well for you.  Adjust appropriately for you setup.

ctzsnooze, 2018-07-10:
The log shows hitting a peak throttle of 48% before the flip, and being above 32% for about 230ms, whereas air mode needs it to be above threshold for more than a second.  So yes, that was the cause of this crash, the motors stopped because air mode did not keep them running.  Runaway protection had nothing at all to do with this crash, other than showing a message.  

There's a lot of quads now that hover well below 25% throttle.  

I'm wondering if we should be thinking about changing the defaults to 20% for runaway throttle and 25% for airmode activation, with airmode activation happening after 200ms above threshold, not waiting a whole second.

closed 2018-08-16 by stale[bot]