betaflight/betaflight #6322: BF 3.4 Matekf405 VTX Settings

localidiot, 2018-07-09:
**Describe the bug**
This might not be a bug now that ive typed this. It could be pit mode.

When using a matekf405 std and an akk 2 vtx using the akk build for the lastest betalfight 3.4 on boot vtx settings are not set. If i flash the same setup with bf 3.3.3 akk build this issue does not occur.

**To Reproduce**
Steps to reproduce the behavior:

I switched between betaflight 3.3.3 and 3.4. Powered off quad and powered on quad.
When on bf 3.4 changing any vtx setting and setting that value, all vtx settings are then applied.

**Expected behavior**
On boot betaflight takes the last configuration of the VTX. with bf 3.4 this does not happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - matek f405 std
 - akk fx2

**Additional context**
Now that I have thought about this, this could be pit mode.

localidiot, 2018-07-09:
Can not duplicate.

wind0r, 2018-07-09:
Betaflight forces having it set via betaflight. This need to be done  so low and high power mode was possible. Also bf now can be set to the needed channels without having the vtx power and on powerup the VTx will resieve the wanted settings

localidiot, 2018-07-09:
Is this 3.3 or 3.4 or both?

closed 2018-08-15 by stale[bot]