betaflight/betaflight #6323: GPS rescue does not stop motors after landing

Quickred, 2018-07-09:
I tried not to abort the GPS rescue process and let it land my copter. Does not work well :( 
The motors don't stop after touch down and one motor burnt and killed the ESC or other way round ;)

With BF 3.4.0 (final) I put GPS rescue on a stick and started the procedure after 200m distance from the home point. At 100m distance the quad starts to fast with descending and touched down 45m before home point. Motors don't stop. I disarmed with stick - too late.

Is there any detection of landing build in GPS rescue?

set gps_rescue_angle = 32
set gps_rescue_initial_alt = 30
set gps_rescue_descent_dist = 100
set gps_rescue_ground_speed = 750
set gps_rescue_throttle_p = 150
set gps_rescue_throttle_i = 20
set gps_rescue_throttle_d = 50
set gps_rescue_velocity_p = 80
set gps_rescue_velocity_i = 20
set gps_rescue_velocity_d = 15
set gps_rescue_yaw_p = 40
set gps_rescue_throttle_min = 1200
set gps_rescue_throttle_max = 1600
set gps_rescue_throttle_hover = 1280
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set gps_rescue_min_sats = 8

# Betaflight / KAKUTEF4V2 (KTV2) 3.4.0 Jul  8 2018 / 04:15:56 (cce8a59be) MSP API: 1.39

video here (if ok) https://youtu.be/5KKakHGue3w

Please make a warning not to land with GS rescue

s0up, 2018-07-11:
You need to turn sanity checks on if you plan on letting it hit the ground 👍

Quickred, 2018-07-11:
Thanks for the info. Love the GPS rescue feature. Still looking for more documentation 😉

closed 2018-07-11 by Quickred