betaflight/betaflight #6324: ESC telemetry input via TX pin with half-duplex mode (RevoltOSD target)

TheMrJahn, 2018-07-10:
With STM MCU, a transmission (TX) pin may be used as reception (RX) by configuring the corresponding U(S)ART unit to half-duplex mode. Betaflight does not support this mode for ESC telemetry inputs.

RevoltOSD uses TX6 pin for ESC telemetry input.

[image]

It would be great to add this feature in BF.

jflyper, 2018-07-10:
@TheMrJahn My apologies for incorrectly informing you that the half-duplex is not supported for ESC telemetry input. It is already supported. Please try CLI command `set esc_sensor_halfduplex = ON`.

closed 2018-07-10 by jflyper

TheMrJahn, 2018-07-10:
@jflyper Yes, it's finally works. Now fc gets telemetry (erpm, temp, mah) from esc!
Tnx for help!