betaflight/betaflight #6325: 3.4 with Ibus

MewKeeWoo, 2018-07-10:
Updating Radiance Pikoblx board to 3.4, but i couldnt get ibus to work.. receiver tab had no movement.. switching around in port setting does nothing.. confiq was set to iBus.. same goes with 3.3.3..

but had no issue with 3.2.4.. 

so board to old or other issue ?

jflyper, 2018-07-10:
@MewKeeWoo Do a full chip erase flash with 3.4 and immediately and manually configure receiver only and see if it works. If not, please attach CLI `diff` result.

MewKeeWoo, 2018-07-10:
i did full chip erase. even swtching sbus and ibus. 

noob question, whats the command for cli to show ?

jflyper, 2018-07-10:
There is a CLI tab...

MewKeeWoo, 2018-07-10:
ya, what to type to shot what i need to show

jflyper, 2018-07-10:
`diff`

MewKeeWoo, 2018-07-12:
I asked what is need to show and what is the command to type in CLI

mikeller, 2018-07-12:
`diff`

jflyper, 2018-07-12:
diff

DusanF, 2018-07-20:
Same here with SP F3: I-BUS works in 3.2.2, in 3.4.0 it doesn't.
 diff:

```
board_name SPRACINGF3
manufacturer_id 

feature -RX_PPM
feature RX_SERIAL
serial 2 64 115200 57600 0 115200
set serialrx_provider = IBUS
```

DusanF, 2018-07-20:
It is disabled in _src/main/target/common_fc_pre.h_
by commit `39bb6ffe04dc40e6cac720e829f20dbbcd8d79c3`

jflyper, 2018-07-20:
> It is disabled in src/main/target/common_fc_pre.h
> by commit 39bb6ffe04dc40e6cac720e829f20dbbcd8d79c3

The 39bb6fe disabled IBUS telemetry feature from all F3 targets, not the IBUS itself.

jflyper, 2018-07-20:
@DusanF 
A genuine SPRacing F3, flashed with v3.4.0 and loaded with your diff works with IBUS receiver FS-IA6B connected to UART3.

I suspect you have a build or hardware problem.

DusanF, 2018-07-20:
well, after enabling
`USE_TELEMETRY_IBUS
and
USE_TELEMETRY_IBUS_EXTENDED`, my receiver started working.

jflyper, 2018-07-20:
Those features are irrelevant to your original issue, but if you are satisfied with current state of your craft, you can close this.

DusanF, 2018-07-20:
My problem was that I have receiver connected only to TX pin of FC set as bidirectional. So without telemetry, FC was expecting data on RX. My mistake. But I'm not the creator of this Issue

jflyper, 2018-07-21:
@DusanF Let's see your current `diff`.

> But I'm not the creator of this Issue

Ah, okay :)

DrUlf, 2018-07-23:
I have the same issue:

FF_PIKOBLX (PIKO) 3.2.2 IBUS ok
FF_PIKOBLX (PIKO) 3.2.5 IBUS ok
FF_PIKOBLX (PIKO) 3.3.3 IBUS not ok
FF_PIKOBLX (PIKO) 3.4.0 IBUS not ok

I do not use IBUS telemetry but just IBUS RX.

# diff
# Betaflight / FF_PIKOBLX (PIKO) 3.2.2 Nov 11 2017 / 00:28:28 (77782e602) MSP API: 1.36
name X2MAX

feature AIRMODE
beeper -ON_USB
serial 0 1 115200 57600 0 115200
aux 0 0 1 1900 2100
aux 1 1 2 1900 2100
aux 2 13 0 1200 1800
aux 3 27 0 1900 2100
aux 4 36 3 1900 2100
rxfail 3 h
rxfail 6 s 2000
set min_check = 1010
set max_check = 1990
set serialrx_provider = IBUS
set failsafe_off_delay = 50
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set align_board_yaw = -45
set vbat_warning_cell_voltage = 34
set current_meter = NONE
set yaw_motors_reversed = ON
set small_angle = 180
profile 0

set vbat_pid_gain = ON
rateprofile 0

set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

---------------------------------------------------------------------------------------

# diff
# Betaflight / FF_PIKOBLX (PIKO) 3.4.0 Jul  8 2018 / 04:10:15 (cce8a59be) MSP API: 1.39

board_name FF_PIKOBLX
manufacturer_id 
name X2MAX

feature -TELEMETRY
feature AIRMODE
beeper -ON_USB
map AETR1234
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
aux 0 0 1 1900 2100 0
aux 1 1 2 1900 2100 0
aux 2 13 0 1200 1800 0
aux 3 27 0 1900 2100 0
aux 4 36 3 1900 2100 0
rxfail 3 h
rxfail 6 s 2000
set min_check = 1010
set max_check = 1990
set serialrx_provider = IBUS
set use_unsynced_pwm = OFF
set motor_pwm_protocol = ONESHOT125
set failsafe_off_delay = 50
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set align_board_yaw = -45
set vbat_warning_cell_voltage = 34
set current_meter = NONE
set yaw_motors_reversed = ON
set small_angle = 180
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
rateprofile 0

set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

jflyper, 2018-07-23:
@DrUlf 
Can we see output from `set serialrx` from both configs?

DrUlf, 2018-07-23:
When I saw the output of "set serialrx", I realized what the problem is. If I set serialrx_inverted = OFF, it works. When serial receiver is IBUS, the default should be OFF.

So this works with 3.4:

# set serialrx
serialrx_provider = IBUS
Allowed values: SPEK1024, SPEK2048, SBUS, SUMD, SUMH, XB-B, XB-B-RJ01, IBUS, JETIEXBUS, CRSF, SRXL, CUSTOM, FPORT
Default value: SBUS

serialrx_inverted = OFF
Allowed values: OFF, ON
Default value: ON

serialrx_halfduplex = OFF
Allowed values: OFF, ON

BTW: serialrx_inverted is not reported in diff values.

DrUlf, 2018-07-23:
After taking off the battery and reconnecting USB, serialrx_inverted is now in diff dump too. So forget about my last comment.

jflyper, 2018-07-24:
@MewKeeWoo 
Do you still have the issue?

I realized that `FF_PIKOBLX` target explicitly sets `serialrx_inverted` to `ON` to accommodate Furious FPV's non-standard non-inverting SBUS receiver as target default. This setting does persist even if `serialrx_provider` is set to `IBUS`, and must explicitly set to `OFF`.

This phenomenon started with v3.3.0, when old setting of `sbus_inversion` was changed (unified) to `serialrx_inverted`. The former only affected SBUS and not other protocols.

MewKeeWoo, 2018-07-25:
Oh wow it works now !!! 

Thanks a lot !!

mikeller, 2018-07-26:
@MewKeeWoo: In this case, would you mind closing this?

mikeller, 2018-07-26:
@DrUlf: You are correct. The point is that the whole serial port configuration is not very good as it is now, and the plan is to rewrite it so that the port parameters are stored per port, and not per function, and then add it to the configurator.

closed 2018-07-26 by MewKeeWoo