betaflight/betaflight #6327: GPS Position Hold - GPS Mode

fOmey, 2018-07-10:
I know this has been discussed in the past ( #4327 ), although considering GPS rescue was recently added I thought I would re-open the line of discussion on this subject.

I noticed in the following file it mentions GPS position hold: https://github.com/betaflight/betaflight/blob/master/docs/Modes.md

Does this mean GPS position hold is a planned feature? If not is it something that would be considered?

mikeller, 2018-07-10:
@fOmey: No, GPS hold is not currently a planned feature - unless somebody in the community is planning to re-introduce it. One of the things about GPS rescue is that it does away with the requirement for a (typically hard to get to work on a high performance craft) compass, and instead uses a 'fixed wing style' method to infer direction from its movement in forward flight. This means that the functions that are used by GPS rescue to control the craft's position will not work when the craft is not moving, and the only way for them to be used to 'hold position' would be to make the craft circle, like it is done for 'position hold' on fixed wing.

So unless somebody is planning to write GPS hold from scratch (which is unlikely), there is little chance that this will be added to Betaflight any time soon.

The documentation you refer to is woefully out of date, unfortunately.

fOmey, 2018-07-10:
Right, ah dam I have to admit I got a little excited when I read that documentation haha.. I often forget betaflight is also used outside the world of quads, tunnel vision at its finest.

Thank you for taking the time to reply..

PR #6328 (2018-07-12): Fixed documentation for modes.
Fixes #6327.
closed 2018-07-10 by fOmey