betaflight/betaflight #6330: Rx issue

Jedigreeny07, 2018-07-10:
I’m using configureator 10.3.1 and the  firmware 3.0.1 on a babyhawk

Jedigreeny07, 2018-07-10:
im Using flysky but every time I go to the configure page I put it on ibus and press save but when I reconnect the rx has been turned to  Spectrum1024 how can I stop this from happening

iRaven, 2018-07-10:
got simmilar problem, for me sometimes stuck on ppm instead of sbus which was selected. some reboots of fc shows select reciever.. also did full chip erase and flashback to 3.3.3 upload diff or so?

Jedigreeny07, 2018-07-10:
When I go to the receiver page betaflight can’t see my rx

wind0r, 2018-07-10:
Please update to a newer firmware or use a older configurator. 3.0.1 is very old and will not work with newest configurator.

MewKeeWoo, 2018-07-11:
I have posted in another post that i cant connect my flysky even when set to ibus after updating to 3.4. Same with 3.3.3. Had to flash 3.2.4 and it works.

jflyper, 2018-07-11:
@MewKeeWoo Our trouble shooting session is stalled. You have to provide `diff` output.

closed 2018-08-17 by stale[bot]