betaflight/betaflight #6331: Current sensor

wesk1, 2018-07-10:
Hello, Current sensor is not working i only get voltage meter, running latest bf on a kiss fc, I have set esc_sensor_halfduplex = ON, do i need to solder bat to pitch pad on fc to make this work?

mikeller, 2018-07-11:
Can you please provide a bit more of context, like you are asked when opening the issue?

wesk1, 2018-07-11:
in bf current meter is displaying 0 only voltage is working is this normal on kiss fc

jflyper, 2018-07-12:
@wesk1 Can we see your CLI `diff` output?

wesk1, 2018-07-14:
board_name KISSFC
manufacturer_id 
name RD

feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR
map TAER1234
serial 0 128 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 30 33 115200 57600 0 115200
led 0 5,15::AO:0
led 1 6,15::AO:0
led 2 7,15::AO:0
led 3 8,15::AO:0
led 4 9,15::AO:0
led 5 10,15::AO:0
mode_color 6 1 2
aux 0 0 0 1800 2000 0
aux 1 1 1 900 1600 0
aux 2 2 1 1700 2100 0
aux 3 13 2 1900 2100 0
aux 4 28 1 1400 2000 0
aux 5 35 5 1950 2050 0
rxfail 3 h
set gyro_sync_denom = 3
set acc_calibration = -46,1,-6
set min_check = 1070
set max_check = 2000
set rssi_channel = 11
set blackbox_p_ratio = 64
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_off_delay = 5
set failsafe_throttle = 1200
set failsafe_switch_mode = KILL
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1500
set current_meter = ESC
set battery_meter = ESC
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set vtx_band = 5
set vtx_channel = 7
set vtx_freq = 5880
set esc_sensor_halfduplex = ON
profile 0

set dterm_notch_cutoff = 0
set p_pitch = 58
set d_pitch = 35
set p_roll = 44
set i_roll = 40
set d_roll = 30
set p_yaw = 70
set d_yaw = 20
set p_level = 70
set d_level = 100
rateprofile 0

set thr_expo = 10
set roll_rc_rate = 82
set pitch_rc_rate = 82
set yaw_rc_rate = 86
set roll_expo = 2
set pitch_expo = 2
set yaw_expo = 2
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_rate = 25

TheNewKobra, 2018-07-17:
Hi, I don't know if my problem with the latest build is related to this one, but I think there are similarities. I use a SPRACING F3 flight controller and a SPRACING F3 OSD/PDB for the OSD and current sensing. I also just get a 0 for current und mah used on my OSD (which by the way works flawlessly). When I downgrade both controllers to 3.3.3 the current meter is working again. Building a custom OSD/PDB firmware for 3.4 doesn't let me establish a connection between OSD and flight controller. 

This is my diff:

# Betaflight / SPRACINGF3 (SRF3) 3.4.0 Jul 16 2018 / 23:57:27 () MSP API: 1.39

board_name SPRACINGF3
manufacturer_id

feature -RX_PPM
feature -RSSI_ADC
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature LED_STRIP
feature OSD
beeper -ON_USB
serial 1 2 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 1 19200 57600 0 115200
serial 31 8192 115200 57600 0 115200
aux 0 0 3 1900 2100 0
aux 1 1 1 1000 1200 0
aux 2 2 1 1400 1600 0
aux 3 3 2 1400 1600 0
aux 4 46 2 1900 2100 0
aux 5 13 0 1900 2100 0
rxfail 4 s 2000
set acc_calibration = 1,82,232
set mag_declination = 3000
set mag_calibration = -118,-93,34
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_throttle_low_delay = 50
set bat_capacity = 1550
set current_meter = MSP
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_ground_speed = 1500
set osd_cap_alarm = 1085
set osd_vbat_pos = 2089
set osd_tim_1_pos = 32
set osd_tim_2_pos = 2081
set osd_flymode_pos = 2380
set osd_vtx_channel_pos = 2104
set osd_ah_pos = 2094
set osd_current_pos = 2368
set osd_mah_drawn_pos = 2401
set osd_altitude_pos = 2422
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2095
set osd_disarmed_pos = 2282
set osd_nheading_pos = 2393
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
profile 0

rateprofile 0

closed 2018-08-24 by stale[bot]

hydra, 2018-08-24:
try latest cleanflight.