betaflight/betaflight #6338: Bad channel defaults on bootup (receiver off)

robert-hoffmann, 2018-07-12:
Not sure if this is on all builds, but if i turn transmitter off and boot the board via battery, defaults in channels have values that could be dangerous. See image attached.

When transmitter is turned ON, these values jump to what they are on the transmitter

If i would have had arming on AUX 3, i guess the quad qould want to jump off the table as soon as i plug the battery in ? ^^

[image]

mikeller, 2018-07-12:
What you are looking at are the failsafe settings, and AUX2 was probably set to be whatever it is by somebody changing the failsafe settings.

robert-hoffmann, 2018-07-12:
Don't understand what you mean, here is the failsafe screen

[image]

mikeller, 2018-07-12:
Ok, looks all normal to me. The throttle value of 885 means that without the RX on, the values will be recognised as failsafe, and not result in any mode changes.

If you'd provided the information you were asked to provide when opening an issue, you could have received some more help maybe.

closed 2018-07-12 by mikeller

ifgesha, 2020-04-14:
Hey.
I have the same problem.
If you turn on the FC and the receiver, but do not turn on the control equipment, then by default the receiver does not transmit anything.
This leads to the fact that almost all AUX are set to 1500. This is NOT a failsave receiver. The same thing happens if you turn on the FC without a receiver at all.
This is extremely inconvenient and not right.

In my version, on one of the AUXs, I have Beeper tied
and he beeps until I turn on the control equipment. And I should squeak only if I myself give 1500 from AUX

During initialization, it is logical to set all AUX to 1000
Why are they put in 1500 and how to fix it?

vmzhivetyev, 2023-12-10:
> In my version, on one of the AUXs, I have Beeper tied
and he beeps until I turn on the control equipment. And I should squeak only if I myself give 1500 from AUX
> 
> During initialization, it is logical to set all AUX to 1000
Why are they put in 1500 and how to fix it?

I have the same concern as @ifgesha.

Is there a way to set default values of AUX channel before the RX is connected to a TX? Some CLI variables?

TheMonomah, 2024-01-24:
This problem bothers me too! Has anyone found a solution?

haslinghuis, 2024-01-24:
There is no problem and nothing to worry about. Aux channels are initialized on center value for 8 years or longer.

After reset to defaults and adding AUX1 for ARM, AUX2 for FM, AUX3 for RTH and AUX4 for BEEPER

[image]

Without valid RC packets and after enabling GPS Rescue in failsafe tab throttle will be initialized with the failsafe throttle.

[image]

Adding Failsafe switch on AUX3 and make it active we finally get this as it would be the same as powering without transmitter:

[image]

Even when we force the quad to be armed (setting 1850 on set in failsafe 1 settings) we get

[image]