betaflight/betaflight #6340: Flip Over After Crash without Disarm/Arm

adamweld, 2018-07-12:
Suggested change of behavior for flip over after crash that would make it faster and less 'clunky' to use.

Current usage:

Crash --> Disarm --> Flip Over mode switch --> Arm --> Flip Over --> Disarm --> Flight mode switch --> Arm --> Fly

That's six switch changes to operate. Not horrible once you learn the sequence but in my opinion the UX could be much simpler (two switch changes):

Crash --> Flip Over mode switch --> Flip Over --> Flight mode switch --> Fly

My rationale is that we already have switches that can cause a crash if accidentally flipped in flight - Arm and Airmode specifically. If Flip Over mode is only activate with a low throttle value, then this change makes the mode no more dangerous than Airmode while greatly improving ease of usage. In fact I would be fine with no throttle min check as Flip Over operates without regard to the throttle value. 

Thoughts? I would be happy to make this change if it is deemed a good idea.

mikeller, 2018-07-12:
@adamweld:

I see a couple of issues with your proposal:

First and foremost, I think one important skill of any safe pilot should be to have 'intuitive' control of the arm / disarm switch through muscle memory, i.e. on hitting the ground, disarming should be immediate and instinctive - training yourself to hit the 'crash flip switch instead of disarming will mean picking up a bad habit, and force you to make a decision in a moment when your reaction should be immediate.

To a lesser extent this applies to arming: There should be one and only one way to put your craft into a condition where motors can spin (a lot of pilots also add a second momentary 'confirmation switch' to reduce the risk of accidentally arming when bumping the TX. Adding a second switch that will make it possible for motors to spin will lower the safety, and add risk.

evnngsky, 2018-07-12:
How dangerous!

closed 2018-08-18 by stale[bot]