betaflight/betaflight #6346: ACC Trim issue on Matek F722

lanman20, 2018-07-13:
Can't get any response when trying to trim ACC using ThrottleUp+(pitch/roll).  I AM able to reset ACC using ThrottleUpLeft+PitchDown without a problem.  Not sure what I am missing.
This is a Hexacopter with a Matek F722 flight controller using fport for both control and telemetry.  Currently running 3.4.0 stable release 7/8/2018 firmware.  No previous firmware loaded as it is a newly built copter.  Any help would be appreciated.

0crap, 2018-07-14:
You do use the "Save Setting" stick command?

https://github.com/martinbudden/betaflight/blob/master/docs/Controls.md

lanman20, 2018-07-14:
Yes, I am able and do use the Save Setting stick command.  All stick commands seem to function correctly except for the commands to trim the ACC forward, backward, left, and right.   When I try to trim the craft via the trim stick commands, nothing happens...no lights flash on the flight controller, no beeps sound from the beeper, leaving no change to be saved.  I have several other quads with F3 and F4 flight controllers and am able to apply and save trim changes without a problem.  However, this new F7 board just doesn't seem to recognize the trim stick positions.

mikeller, 2018-07-14:
@lanman20: Silly question: What mode are you in when trying to apply accelerometer trims? The functionality was changed so that these stick commands only do accelerometer trims when in ANGLE / HORIZON mode, and switch rateprofiles when in ACRO mode, see #4581.

lanman20, 2018-07-15:
@mikeller: Thank you for this info!  This was the problem.  The only quads I have updated the firmware on over the last year or so are quads I use for fpv, which are acro only for me and don't use the ACC.  However, I am flying this Hex line-of-sight in Angle mode.  I had simply configured the same switch I use to arm with for Angle mode.  Therefore, when unarmed, the hex was actually in acro mode and given the functionality change you mentioned above, it now makes sense why I couldn't trim the ACC using the sticks.  Once I configured the Modes section to keep the craft in Angle mode when disarmed, I was able to trim it without a problem.  Thanks again!

closed 2018-07-15 by lanman20