betaflight/betaflight #6347: [Glitch] Motor glitch after fast roll input

Jonitsvk, 2018-07-13:
Sometimes after fast roll or sharp turn quad gets out of control and go haywire. It happens sporadicaly, not very often though. It happened maybe 3-4 times on BF3.4 (50 packs flown totally) with my 2 inch quad and on my 5 inch quad (BF3.4 RC4) I have only flown 3 packs and it occured once and caused a crash (i don't have a log unfortunately), it happened after fast roll input (one axial roll). 

I was looking at the log and motor behaviour seems weird at the moment of glitch. 
Seems like the front left motor goes full power, but after a moment it looks like even though it is at full power, the motor is not producing any thrust because the quad (at the blackbox explorer) is rotating in the oposite direction as it should when that motor is at full power. Or maybe that is just my wrong perception, but anyway...

[Here is the screenshot from the BB explorer at the moment of the glitch.](https://imgur.com/a/aBhFisv)
-look at how the motors are exactly at 20%, 30%, 30% and 100% values

It is log 4/6 and it starts at 33s. 

[BTFL_BLACKBOX_LOG_Jonit_20180712_212756.zip](https://github.com/betaflight/betaflight/files/2191469/BTFL_BLACKBOX_LOG_Jonit_20180712_212756.zip)

NOTE: the problem started to occur after updating to BF3.4. Previously I was flying BF3.2.2 and it was without problem.

-------------------------------------------------------------------

```# diff

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.4.0 Jun 24 2018 / 10:00:20 (1c081ae8c) MSP API: 1.39

board_name OMNIBUSF4
manufacturer_id AIRB

# name
name Jonit

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature AIRMODE

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 1800 2000 0
aux 1 1 0 900 1150 0
aux 2 13 1 1625 2025 0
aux 3 35 1 1625 2025 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 5 h

# master
set gyro_lowpass_hz = 120
set acc_calibration = -158,10,-215
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_type = FILTER
set serialrx_provider = SBUS
set blackbox_p_ratio = 16
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 30
set align_board_yaw = -45
set yaw_deadband = 12
set pid_process_denom = 1
set report_cell_voltage = ON
set osd_tim2 = 1025
set osd_tim_1_pos = 2134
set osd_tim_2_pos = 2113
set osd_throttle_pos = 2081
set osd_vtx_channel_pos = 2103
set osd_crosshairs_pos = 2285
set osd_ah_pos = 199
set osd_debug_pos = 258
set osd_avg_cell_voltage_pos = 2124
set osd_disarmed_pos = 2219
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 3
set vtx_power = 4
set vtx_freq = 5705

# profile
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_threshold = 400
set anti_gravity_gain = 7000
set setpoint_relax_ratio = 15
set dterm_setpoint_weight = 100
set p_pitch = 57
set i_pitch = 63
set d_pitch = 61
set p_roll = 53
set i_roll = 63
set d_roll = 63

# rateprofile
rateprofile 0

set thr_mid = 35
set thr_expo = 30
set roll_rc_rate = 87
set pitch_rc_rate = 87
set yaw_rc_rate = 110
set roll_expo = 10
set pitch_expo = 10
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 65
```

**SETUP**
2 inch miniquad
FC: HGLRC MICRO F4 V2
ESC: HGLRC MICRO 13A 4IN1 BLHELI_S + 330uF capactior
MOTORS: BROTHERHOBBY RETURNER R3 1402 8100KV
RX: FrSky R-XSR

Any idea on what the problem might be?

Thanks!

Jonitsvk, 2018-07-13:
I always set idle value lower than default because I don't like props to spin that fast at idle. Especially on my 2 inch quad with those 8000KV motors it spins pretty fast at idle. 

Are there any negative consequences for having it set that low? 

I will definitely test it with default value of 450. But I might not be able now to fly for a couple of days though (full schedule). 

Thanks for your suggestion  Gim.

McGiverGim, 2018-07-13:
The only negative consequence having it low is that you have the risk to desync the motors, that is what it seems is happening to you. Maybe some of the new features of 3.4 make a faster change in motors than with earlier 3.3, and this is the reason why you only notice it now.

Let's see if I'm right in your next tests ;)

Jonitsvk, 2018-07-13:
Thanks for the clarification. Will test.

Zer0Exeption, 2018-07-15:
This is definitly not a desync issue. Have the same problem, occurs with any motor, but only on betaflight 3.4. Also the other motors go to 10%, 30%, 30%, 20%. Looks like it freezes but the gyro data is still captured and pid still calculated. 
Seems like it's related to pids, the lower they are, the less ,,fakedesyncs,, I get. I got desyncs on another quad before, that looked different because they just gave 100% to that one motors but the others wheren't that flat.

flint723, 2018-07-15:
It's not a fake desync, and the motors going to 100,30,30,20% is normal when one motor doesn't respond (motor mixer forces this if throttle stick is low or zero).  

I also wonder if putting `throttle_boost` to 0 or 1 would help.

Ozzy33, 2018-07-23:
To low idle can cause added response time. It takes longer for the motor to spool up and make thrust to start rotating the quad.

When blh-s came out, i could lower the idle beyond zero thrust without desync, you could actually see the bi-blades rotating. but r/p response was noticably slower. A start to stop roll is ~240ms (quarter sec) with my rates. With idle set lower than zero thrust, my quad does not complete the roll in that time.

My favorite way to set idle is, get momentum going straight away from you at an upward angle, then knife edge at zero thro, if it curves, its making thrust, lower idle till it goes straight, dont go lower than this. This is LOS and no wind.

Jonitsvk, 2018-07-24:
I have set iddle to 450 (up from 250) and so far so good... But only flown like 12 packs so far. Will see after a longer period / more packs, but it seems that that was the problem, as Gim pointed out.

McGiverGim, 2018-07-24:
Great! Thanks for your feedback.

closed 2018-08-30 by stale[bot]