betaflight/betaflight #6348: FC crash when downloading logs from built-in flash

wysiwyng, 2018-07-13:
When downloading blackbox log data from the built-in dataflash on an Omnibus Fireworks V2, the flight controller stops responding after a short time. This happens only if my RC transmitter is switched on while downloading the log. If it is switched off, I can download logs without problems. I am using a Spektrum DX10t with a SPM4649T receiver in SRXL + Telemetry mode connected to TX1 on the FC.

diff output:

```
# diff

# version
# Betaflight / OMNIBUSF4FW (OBFW) 3.4.0 Jul  8 2018 / 04:24:16 (cce8a59be) MSP API: 1.39

board_name OMNIBUSF4FW
manufacturer_id 

# name
name Fury 5X

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper

# beacon

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux
aux 0 0 4 900 1100 0
aux 1 1 0 1900 2100 0
aux 2 2 0 1400 1600 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW270
set align_acc = CW270
set mag_hardware = NONE
set serialrx_provider = SRXL
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set debug_mode = GYRO_RAW

# profile
profile 0

set dterm_notch_cutoff = 0
set p_pitch = 46
set i_pitch = 45
set d_pitch = 25

# rateprofile
rateprofile 0
```

When running BF configurator from a terminal, nothing suspicious is printed. I tried and verified this behavior on 2 different PCs with Windows and Linux and a whole bunch of different USB cables.

AndersHoglund, 2018-07-14:
OK, verified using BF3.4.0 on DYSF4PRO, SPM4649T and iX12. Telemetry and SRXL in use. The problem only appears when using the old GUI method to download BB-log, it does not happen when using the MSC method. 

Could we please have some more tests with other Serial RX setups, to see if this is specific to the Spektrum Rx code or if it is a general problem. I only have Spektrum stuff...

AndersHoglund, 2018-07-14:
Updated to current HEAD version (build no 987) and it looked as though problem was gone, 5 OK BB download attempts in a row. But then it failed on the next try. All-in-all 2 failures in 10 tries so far.
```
# version
# Betaflight / DYSF4PRO (DYS4) 3.5.0 Jul 14 2018 / 03:59:18 (2c084cbf2) MSP API: 1.40

```

wysiwyng, 2018-07-14:
> The problem only appears when using the old GUI method to download BB-log, it does not happen when using the MSC method.

Yes, that's what I tried to say. This behaviour occurred to me already in BF version 3.3.3, I ignored it at that point as I did not have the time to mess with it further. I cannot test anything at the moment, as soon as I get home again I could also try with a normal Spektrum satellite. Maybe also only telemetry causes this, I have not tried that yet.

wysiwyng, 2018-07-20:
Had another look into this, it seems that telemetry causes this issue. With it turned off, log download works fine while leaving the transmitter on.

haslinghuis, 2022-05-10:
Fixed in https://github.com/betaflight/betaflight/pull/6560

closed 2022-05-10 by haslinghuis