betaflight/betaflight #6350: F.Port on UART RX pin?

fredericdepuydt, 2018-07-13:
Dear,

I've got a BNF drone (MakerFire Armor 65) which I want to upgrade with telemetry. The drone came with a FrSky XM, but I'd like to install a FrSky R-XSR instead. The drone only has a PPM and a UART RX pin pad. Would I be able to connect the FrSky R-XSR on the UART RX pin (instead of the UART TX) if I made some changes to the Betaflight build?

So my first question, is it possible with this hardware (SP Racing F3)?
If so, can I make this work by only changing some parameters in the Betaflight CLI, or do I need to make changes to the Betaflight build?

Kind regards,
Frederic Depuydt

jflyper, 2018-07-13:
@fredericdepuydt 
Unfortunately, half-duplex mode of an UART use TX side pin.

closed 2018-07-13 by jflyper

DieHertz, 2018-07-13:
Couldn't we allow using fport with no telemetry, just never answering any requests?

jflyper, 2018-07-13:
Yes. That is a working configuration, but @fredericdepuydt sounded like he chose F.Port to get telemetry working over the half-duplex protocol.

fredericdepuydt, 2018-07-13:
@jflyper that's too bad, but thanks for the quick response, I'll have to stick with the FrSky XM for now until I have the courage to solder directly on the TX pin coming from the STM32F03.
@DieHertz wouldn't that be the same as using normal SBUS (only difference is the receiver firmware)

DieHertz, 2018-07-14:
Fport is 2.5ms per frame, sbus is 2.97ms, but you also get RSSI natively

fredericdepuydt, 2018-07-14:
@DieHertz that's only the length of 1 frame, the update cycle for F.Port is 9ms, and for SBUS it's 14ms (analog mode) or 7ms (highspeed mode). So SBUS could be more responsive than F.Port (No idea how to select analog of highspeed mode though).

DieHertz, 2018-07-14:
On FrSky X both SBUS and FPORT will have 110 Hz updates (that's 9ms), but FPORT will have have tiny bit less latency and RSSI